rstrouse/relayEquipmentManager #1: proper permissions to start REM with pm2?

johnny2678, 2021-02-23:
I can run REM as `pi` using `npm start` and everything works fine (i2c, etc)

When I run with pm2 (as `pi`), it gets caught in a loop:

```
3|relayEquipmentManager  | > relay-equipment-manager@0.0.0 start /home/pi/relayEquipmentManager
3|relayEquipmentManager  | > npm run build && node dist/app.js
3|relayEquipmentManager  | > relay-equipment-manager@0.0.0 build /home/pi/relayEquipmentManager
3|relayEquipmentManager  | > tsc
3|relayEquipmentManager  | linux
3|relayEquipmentManager  | info: Connecting mqtt to mqtt://192.168.5.140:1883
3|relayEquipmentManager  | info: Initializing GPIO Pins 1
3|relayEquipmentManager  | info: Configuring Pin #36 Gpio #16:out on Header 1.
3|relayEquipmentManager  | info: Pin #36 Gpio #16:out on Header 1 Configured.
3|relayEquipmentManager  | info: Initializing SPI Bus #0
3|relayEquipmentManager  | info: SPI Bus #0 Initialized
3|relayEquipmentManager  | info: Initializing SPI Bus #1
3|relayEquipmentManager  | info: SPI Bus #1 Initialized
3|relayEquipmentManager  | info: Initializing i2c Interface
3|relayEquipmentManager  | info: Detecting i2c Buses
3|relayEquipmentManager  | info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
3|relayEquipmentManager  | info: Initializing i2c Bus #1
3|relayEquipmentManager  | info: Server is now listening on 0.0.0.0:8085
3|relayEquipmentManager  | info: Connected to njspc-15620 : http://192.168.5.205:4200
3|relayEquipmentManager  | info: Scanning i2c Bus #1
3|relayEquipmentManager  | info: Found I2C device Atlas_EZO-ORP at address: 0x62
3|relayEquipmentManager  | info: Found I2C device Atlas_EZO-pH at address: 0x63
3|relayEquipmentManager  | warn: Atlas_EZO-ORP Suspend Polling 3
3|relayEquipmentManager  | info: Device Atlas EZO-ORP initialized for i2c bus #1 address 98
3|relayEquipmentManager  | info: New socket client connected RpewmNvFknuFdwhtAAAA -- 192.168.5.205
3|relayEquipmentManager  | warn: Atlas_EZO-pH - Device not ready re-trying the command L,? again: Retries 0.
3|relayEquipmentManager  | error: Atlas_EZO-pH 99 Slope,?. No I2c data to send
3|relayEquipmentManager  | info: Device Atlas EZO-pH initialized for i2c bus #1 address 99
3|relayEquipmentManager  | info: i2c Bus #1 Initialized
3|relayEquipmentManager  | info: New socket client connected 78V4RAsegw8D1nYYAAAB -- 192.168.5.138
**3|relayEquipmentManager  | Shutting down Relay Equipment Manager**
3|relayEquipmentManager  | info: Closing 2 devices.
3|relayEquipmentManager  | info: Stopped I2c Atlas_EZO-ORP
3|relayEquipmentManager  | info: Stopped I2c Atlas_EZO-pH
3|relayEquipmentManager  | info: Closed i2c Bus #1
3|relayEquipmentManager  | info: Stopping GPIO Controller
3|relayEquipmentManager  | > relay-equipment-manager@0.0.0 start /home/pi/relayEquipmentManager
3|relayEquipmentManager  | > npm run build && node dist/app.js
3|relayEquipmentManager  | > relay-equipment-manager@0.0.0 build /home/pi/relayEquipmentManager
3|relayEquipmentManager  | > tsc
```

Not a lot there to tell me what's going on.  Here's my pm2 ecosystem js file:
```
module.exports = {
 apps: [
         {
             "name": "relayEquipmentManager",
             "script": "npm",
             "args": [
                 "start"
             ],
             "cwd": "/home/pi/relayEquipmentManager",
             "autorestart": false,
             "restart_delay": 10000,
             "env": {
                 "NODE_ENV": ""
             },
             "watch": true,
             "ignore_watch": ["node_modules", "config.json", "src", "dist", ".cache"]
         }
     ],
   deploy : {
     production : {
       user : 'SSH_USERNAME',
       host : 'SSH_HOSTMACHINE',
       ref  : 'origin/master',
       repo : 'GIT_REPOSITORY',
       path : 'DESTINATION_PATH',
       'pre-deploy-local': '',
       'post-deploy' : 'npm install && pm2 reload ecosystem.config.js --env production',
       'pre-setup': ''
     }
   }
};
```

See anything obvious that would cause a loop?  Any way to increase logging on REM to give more details about why it's shutting down?

Not a big deal as I can run directly using `npm start` but would be better process mgmt if I could get it to work using pm2.

johnny2678, 2021-02-23:
also just a comment... on my pi, i had to run REM as root (`sudo su -`) when I set up the i2c connection.  

Otherwise it couldn't find the probes.  Once I set up the i2c connection and configured the probes, REM works just fine running as `pi`

a bit odd since I can query probe values via an i2c python script I wrote running as `pi` but not a big deal in the end.  Just need to get pm2 working and environment will be pretty steady.

I completely shut down my acid dosing nodered flow.  Here's a graph of REM controlling my acid dosing:

[image]

I've configured the dosing/mixing exactly like i had in nodered so there are no differences between today's graph and any other day prior to today, except REM is running the dosing automation.  Great job!

johnny2678, 2021-02-23:
omg... rooking mistake on my part.  I had `pm2` `watch` set to `true` so it kept restarting when the REM folder contents changed.  Sorry to bother!

closed 2021-02-23 by johnny2678

DrJeff, 2021-03-13:
@johnny2678 
How do you get the values into njspc via a script to mqtt I have sensors external but want to feed to njspc? Next item on list to do PH ORP with SWG IC60