rstrouse/relayEquipmentManager #2: FEATURE REQUEST: d18b20 temp sensor over 1wire

johnny2678, 2021-02-23:
Just logging this as a wish list item.  I wish I could help with the code but I do have these connected to my pi right now.  Currently using python to grab values.  

Happy to help test if you guys want to take a stab at adding support at some point.

DrJeff, 2021-03-09:
I want it but gotta install REM first :), I'm using d18b20 currently on my other setup

DrJeff, 2021-10-08:
any update on ds18B20 temp sensor working?

tagyoureit, 2021-10-09:
Not yet... thanks for the reminder.

tagyoureit, 2021-10-09:
I ordered a DS18b20 so I'll work on getting this hooked up and then verify by next week.

tagyoureit, 2021-10-21:
Pull down REM 7.5.0.  1-wire + DS18B20 added.

rstrouse, 2021-11-22:
This feature has been added.  Open another issue if you find any problems.

closed 2021-11-22 by rstrouse