rstrouse/relayEquipmentManager #3: Help setting up REM with ...

DrJeff, 2021-03-09:
Help setting up REM to complement Virtual Pump, and IC60. 
Ok my thoughts are to use this as my complete pool controller currently using nodejs-PoolController to operate the Pump and IC60 until a few days ago. Then I use an esp32 to control relays for actuators, heater, multiple pumps, blower, acid pump, Temperatures from 1 wire ds18b20, pressure from a transducer. Orp and PH from atlas. Just wondering what can be put into REM.

DrJeff, 2021-03-10:
Can REM be used for recieving MQTT messages from esp32 for temps, pressure, and valves?

rstrouse, 2021-03-11:
Try running REM with sudo.  Permissions are required to write to the io bus.  There is a socket feed for pH temps.

DrJeff, 2021-03-11:
Yes I'm running as root, still no joy. My system is virtual controller for intelliflo vs, and ic60 is that the issue and what can we do?
[image]

DrJeff, 2021-03-11:
This is the dialog box that pops up
[image]

```
MessageCannot read property 'setPinTriggerAsync' of undefined
TypeError: Cannot read property 'setPinTriggerAsync' of undefined
    at GpioPin.setPinTriggerAsync (/home/pi/REM/boards/Controller.ts:1100:24)
    at Gpio.setPinTriggerAsync (/home/pi/REM/boards/Controller.ts:790:66)
    at /home/pi/REM/web/services/Config.ts:254:44
    at Layer.handle [as handle_request] (/home/pi/REM/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/REM/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/REM/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/REM/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/REM/node_modules/express/lib/router/index.js:281:22
    at param (/home/pi/REM/node_modules/express/lib/router/index.js:354:14)
    at param (/home/pi/REM/node_modules/express/lib/router/index.js:365:14)
    at param (/home/pi/REM/node_modules/express/lib/router/index.js:365:14)
    at Function.process_params (/home/pi/REM/node_modules/express/lib/router/index.js:410:3)
    at next (/home/pi/REM/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/REM/web/Server.ts:245:17
    at Layer.handle [as handle_request] (/home/pi/REM/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/pi/REM/node_modules/express/lib/router/index.js:317:13)
```

DrJeff, 2021-03-11:
@rstrouse 
Any thoughts on how to get these pins to work with the virtual controller. The pins react to REM via the gui pin on and off. I just can't tie them to nodejspc or I get that traceback error. I assume it's because of the Virtual controller but have no idea.

rstrouse, 2021-03-11:
This has nothing to do with njspc at this point.  The GPIO pin is not being recognized  **Post your data/controllerConfig.json file here.**  

The other thing to do is reload the server and watch the console.  You can do this by going to the general tab and clicking reload server.

DrJeff, 2021-03-11:
```
root@rem:~/rem/data# cat controllerConfig.json
{
  "controllerType": "raspi",
  "gpio": {
    "pins": [
      {
        "id": 40,
        "headerId": 1,
        "isInverted": false,
        "direction": "output",
        "triggers": [],
        "feeds": [],
        "isActive": true,
        "name": "Pin #1-40",
        "state": "off"
      }
    ],
    "exported": [
      21
    ]
  },
  "spi0": {
    "isActive": false,
    "busNumber": 0,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "spi1": {
    "isActive": false,
    "busNumber": 1,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "i2c": {
    "isActive": false,
    "buses": [],
    "detected": [
      {
        "driver": "bcm2835 (i2c@7e804000)",
        "path": "/sys/class/i2c-dev/i2c-1",
        "name": "i2c-1",
        "busNumber": 1
      }
    ]
  },
  "connections": [
    {
      "id": 1,
      "isActive": true,
      "name": "nodejspc",
      "type": "njspc",
      "protocol": "http:",
      "ipAddress": "192.168.0.179",
      "port": 4200,
      "userName": "",
      "password": "",
      "sslKeyFile": "",
      "sslCertFile": "",
      "options": {}
    }
  ],
  "genericDevices": {
    "isActive": true,
    "buses": [],
    "detected": []
  },
  "lastUpdated": "3/11/2021, 10:51:25 AM",
  "configVersion": 1
```

rstrouse, 2021-03-11:
Well look at that it isn't initializing the trigger definitions for events.  I'll do some debugging to find out the path to this and post a fix.  It is fine when the pin had a previous trigger.

rstrouse, 2021-03-11:
Do a pull it should allow you to define the trigger on a new pin.

DrJeff, 2021-03-11:
Yes it works now one more thing if using mqtt where are the topics? event name? How do i formulate it?

rstrouse, 2021-03-11:
If you create an MQTT connection on the connections page, you can set whatever topic and payload you like as a feed.

DrJeff, 2021-03-11:
Ok the feed is the payload, The pin is working in the gui for setting the gpio on and off (led), but when the trigger is happening (njspc logic) only gui shows the change the actual gpio (led) is not changing.
```
1|REM  | verbose: 192.168.0.53 GET /config/options/pin/1/40?null {}
1|REM  | info: [12:25:53 PM] 192.168.0.53 GET /config/options/pin/1/40?null {}
1|REM  | verbose: 192.168.0.53 GET /config/options/general?null {}
1|REM  | info: [12:25:59 PM] 192.168.0.53 GET /config/options/general?null {}
1|REM  | verbose: 192.168.0.53 PUT /config/reset {"controllerType":"raspi","app":{"level":"silly","logToFile":false},"spi0":{"isActive":false},"spi1":{"isActive":false},"name":"mqtt","type":{"desc":"MQTT Client"}}
1|REM  | info: [12:26:01 PM] 192.168.0.53 PUT /config/reset {"controllerType":"raspi","app":{"level":"silly","logToFile":false},"spi0":{"isActive":false},"spi1":{"isActive":false},"name":"mqtt","type":{"desc":"MQTT Client"}}
1|REM  | info: Stopping GPIO Controller
1|REM  | info: Initializing GPIO Pins 1
1|REM  | info: Configuring Pin #40 Gpio #21:out on Header 1.
1|REM  | info: Pin #40 Gpio #21:out on Header 1 Configured.
1|REM  | info: Connecting mqtt to mqtt://192.168.0.4:1880
1|REM  | silly: Setting Dirty... true 8205
1|REM  | info: Initializing SPI Bus #0
1|REM  | info: SPI Bus #0 Initialized
1|REM  | info: Initializing SPI Bus #1
1|REM  | info: SPI Bus #1 Initialized
1|REM  | info: Initializing i2c Interface
1|REM  | info: Detecting i2c Buses
1|REM  | info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
1|REM  | silly: Setting Dirty... true 8218
1|REM  | info: Connected to nodejspc : http://192.168.0.179:4200
1|REM  | info: Binding circuit from nodejspc
1|REM  | debug: Persisting Configuration data...
1|REM  | verbose: 192.168.0.53 GET /config/options/gpio?null {}
1|REM  | info: [12:26:04 PM] 192.168.0.53 GET /config/options/gpio?null {}
1|REM  | verbose: 192.168.0.53 GET /config/options/pin/1/40?null {}
1|REM  | info: [12:26:07 PM] 192.168.0.53 GET /config/options/pin/1/40?null {}
1|REM  | verbose: 192.168.0.53 DELETE /config/pin/trigger/1/40/2 {}
1|REM  | info: [12:26:12 PM] 192.168.0.53 DELETE /config/pin/trigger/1/40/2 {}
1|REM  | silly: Setting Dirty... true 7359
1|REM  | silly: Setting Dirty... true 7361
1|REM  | verbose: 192.168.0.53 GET /config/options/general?null {}
1|REM  | info: [12:26:14 PM] 192.168.0.53 GET /config/options/general?null {}
1|REM  | debug: Persisting Configuration data...
1|REM  | verbose: 192.168.0.53 PUT /config/reset {"controllerType":"raspi","app":{"level":"silly","logToFile":false},"spi0":{"isActive":false},"spi1":{"isActive":false},"name":"mqtt","type":{"desc":"MQTT Client"}}
1|REM  | info: [12:26:15 PM] 192.168.0.53 PUT /config/reset {"controllerType":"raspi","app":{"level":"silly","logToFile":false},"spi0":{"isActive":false},"spi1":{"isActive":false},"name":"mqtt","type":{"desc":"MQTT Client"}}
1|REM  | info: Stopping GPIO Controller
1|REM  | info: Initializing GPIO Pins 1
1|REM  | info: Configuring Pin #40 Gpio #21:out on Header 1.
1|REM  | info: Pin #40 Gpio #21:out on Header 1 Configured.
1|REM  | info: Connecting mqtt to mqtt://192.168.0.4:1880
1|REM  | silly: Setting Dirty... true 672
1|REM  | info: Initializing SPI Bus #0
1|REM  | info: SPI Bus #0 Initialized
1|REM  | info: Initializing SPI Bus #1
1|REM  | info: SPI Bus #1 Initialized
1|REM  | info: Initializing i2c Interface
1|REM  | info: Detecting i2c Buses
1|REM  | info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
1|REM  | silly: Setting Dirty... true 681
1|REM  | info: Connected to nodejspc : http://192.168.0.179:4200
1|REM  | info: Binding circuit from nodejspc
1|REM  | verbose: 192.168.0.53 GET /config/options/gpio?null {}
1|REM  | info: [12:26:18 PM] 192.168.0.53 GET /config/options/gpio?null {}
1|REM  | debug: Persisting Configuration data...
1|REM  | verbose: 192.168.0.53 GET /config/options/pin/1/40?null {}
1|REM  | info: [12:26:19 PM] 192.168.0.53 GET /config/options/pin/1/40?null {}
1|REM  | silly: Setting Dirty... true 5413
1|REM  | debug: Persisting Configuration data...
1|REM  | verbose: 192.168.0.53 GET / {}
1|REM  | verbose: 192.168.0.53 GET /config/options/general?null {}
1|REM  | info: [12:26:30 PM] 192.168.0.53 GET /config/options/general?null {}
1|REM  | info: New socket client connected t-dxzVAxZD2qD6z5AAAF -- 192.168.0.53
1|REM  | verbose: 192.168.0.53 GET /config/options/i2c?null {}
1|REM  | info: [12:26:30 PM] 192.168.0.53 GET /config/options/i2c?null {}
1|REM  | verbose: 192.168.0.53 GET /config/options/gpio?null {}
1|REM  | info: [12:26:32 PM] 192.168.0.53 GET /config/options/gpio?null {}
1|REM  | verbose: 192.168.0.53 GET /config/options/pin/1/40?null {}
1|REM  | info: [12:26:34 PM] 192.168.0.53 GET /config/options/pin/1/40?null {}
```

rstrouse, 2021-03-11:
I am not sure I follow.  If you turn on the circuit id you specified on the trigger from njspc it should be setting the gpio on the pi.

DrJeff, 2021-03-11:
So the circuit is triggering the pi in REM's gui but not REM's actual  GPIO pin

rstrouse, 2021-03-11:
Hmmm.  I'm not sure how that can happen as the pin states are read back from the pi to ensure the setting is correct.  Is this attached to a relay?

DrJeff, 2021-03-11:
Just a LED

rstrouse, 2021-03-11:
Did you create a trigger for the on state as well as the off state in REM?

DrJeff, 2021-03-11:
> Did you create a trigger for the on state as well as the off state in REM?

Yes

DrJeff, 2021-03-11:
Maybe after the update something is hanging around. I will reset/remove the data folder contents this evening and try again

rstrouse, 2021-03-11:
Yeah that is odd.  If the pin state is changed in the ui, it is driven by the return from the GPIO on the pi.

rstrouse, 2021-03-11:
Do a pull on REM.  There was a socket upgrade that was causing the GPIO to wait.  You may have been seeing only the transition from the socket.

DrJeff, 2021-03-12:
> Do a pull on REM

Yes it is working! 
Thanks 

Now how to use mqtt. I see this page
[image]

[image]

I don't understand where you put the topic ```pool\state\pin\40``` and where the payload will go? Filtered Expression? how to capture an on is it just "on"?

rstrouse, 2021-03-12:
No this is the wrong wat to go about this.  If you want to trigger a circuit you must do it from njspc.  REM is along for the ride in your case, so a trigger from an MQTT broker is not applicable here.  If you would like to publish data from REM in this case for use by other apps then simply add a Feed not a Trigger.  For your use case, to trigger the activation of a circuit, to that from using the MQTT interface in njspc.

Now if this were something other than a controllable component of another system, then feel free to add MQTT triggers to control the pin because your REM device would be its master.  To do that you put in the statement
`return true;` to set the desired state or `return false;` to not trigger the state.  The closure -- the lines after => { to } -- is a full blown javascript closure in which you can put any logic you like.  Just return true when you want to set the state selected in the dropdown above.

**NOTE:** Not setting the state simply means that the Desired State value is not set.  It does not mean that it sets the opposite of it.

I get it that when you are referring to GPIO it is typically on/off however REM also does neat tricks with SPI and I2c devices that have a much greater data payload.  **Simple Example:** a pH sensor has a temperature compensation value, the current slope, probe voltage, and status values.  All of this is exposed through the REM trigger/feed interface.

DrJeff, 2021-03-12:
Awesome Sauce you are Amazing! I get it now now i'm going to ask @tagyoureit if there is a way to add more features to njspc I can not add virtual circuits (I'm not using a real controller). Features work fine but my pumps extra speeds (4) are being used as features. I need about 17 features for valves and  relays. I really want to get my PH and ORP working on this next. I need to order new carrier boards just running current setup on breadboard.

rstrouse, 2021-03-12:
There is a project that has been started called Nixie.  In fact when you boot njspc you will see a message saying Initializing Nixie controller.  What this is is a virtual controller that has the ability for manage bodies, circuits, features, groups, schedules, valves, heaters, remotes... etc.  

If you go to the config pages you can add virtual valves.  The way valves work is that they are triggered by circuits or features.  From there you can have it trigger either an I2c or GPIO relay.

If you are using atlas gear I would suggest the Whitebox tentacle.  It has 3 carriers for pH, ORP, and non-isolated temperature RTD.  There are also 2 ports that can be used for EZO-PRS or any other I2c devices.

DrJeff, 2021-03-12:
> project that has been started called Nixie

I saw that on startup and was like OK but will look more now :grinning: where do I get to that? I see it in the code https://github.com/tagyoureit/nodejs-poolController/tree/master/controller/nixie but how do I go about using it. 

> If you go to the config pages you can add virtual valves. The way valves work is that they are triggered by circuits or features. From there you can have it trigger either an I2c or GPIO relay.

Every time in dashPanel I get errors about ID this is for features but also Valves. Permissions? So to add features I used webClient. 
```
1|poolController  | [3/12/2021, 10:57:33 AM] info: [10:57:33 AM] 192.168.0.53 PUT /config/feature {"id":-1,"nameId":202,"type":0,"showInFeatures":true,"freeze":false,"dontStop":true,"eggTimer":720}
1|poolController  | [3/12/2021, 10:57:33 AM] error: Invalid feature id: -1
```
[image]

For atlas I have some isolated Gen1 boards and PH, ORP Embedded with cheap probes but the isolated boards were acting up so I breadboarded the embedded and worked fine for now. But Thinking of better setup with the Pi now and REM instead of esphome and a python script I stole. 
Can pressure be an analog via arduino? something like this https://www.electronics-lab.com/build-i2c-sensor/ 
with a sensor like this?
https://www.amazon.com/Pressure-Transducer-Sender-Sensor-Stainless/dp/B074QPD5RT/ref=sr_1_30?dchild=1&keywords=water+pressure+sensor+i2c&qid=1615576416&s=industrial&sr=1-30

Can flow sensor be the IC60 flow sensor or need an external to get at that? any suggestions?

rstrouse, 2021-03-12:
Your best bet is definitely getting the hat.  The problem with I2c is that any cable length makes it ugly and the hat eliminates all that.  In the next couple of weeks I will be testing an I2c extender that is supposed to make it square wave at 100+ feet.

Yes there is a pressure transducer device in REM under the Generic devices but why would you want to put that on an Arduino.  You can just go direct from the pi.  @tagyoureit  has this setup running on his system right now.

DrJeff, 2021-03-12:
> Yes there is a pressure transducer device in REM under the Generic devices but why would you want to put that on an Arduino. You can just go direct from the pi. @tagyoureit has this setup running on his system right now.

Analog on Pi?

> Your best bet is definitely getting the hat

👍🏽 

is the nixie just located @ the port set in the config file or am I missing something it is enabled

rstrouse, 2021-03-12:
I believe @tagyoureit used an ADSx chip (you can get them for a few cents on a breakout board).

You can disable the valve relays interface as this is for the IntelliValve when we finally get it decoded.  Our roadblock is that after you send it 4 or 5 million commands, we believe the flash memory wears out and it stops responding to the buttons on the front of the valve.

I'll have a look at the Virtual OCP.  It is probably not assigning equipment Ids for that controller as this was not originally the case with -webClient.

DrJeff, 2021-03-12:
> I believe @tagyoureit used an ADSx chip (you can get them for a few cents on a breakout board).

Duh, I wasn't thinking. Sometimes I can get really dumb :smile: Got those laying around I was stuck on I2c (mental block)

> I'll have a look at the Virtual OCP. It is probably not assigning equipment Ids for that controller as this was not originally the case with -webClient.

OK thanks again

> You can disable the valve relays interface as this is for the IntelliValve when we finally get it decoded. Our roadblock is that after you send it 4 or 5 million commands, we believe the flash memory wears out and it stops responding to the buttons on the front of the valve.

So thats different than nixie then?

rstrouse, 2021-03-12:
> So thats different than nixie then?

Yes Nixie is much more sophisticated than that.  Simply set up a trigger for the relay with the valve event.  It will take care of the diverted/non-diverted positions based upon the currently running circuits and features within the Nixie Control Panel (NCP).   Currently we don't have a way for you to define Intake and Return valves for shared systems but that is one of the first orders of business when I get back to it.  This will take into account Pool vs Spa vs Spillway modes and send the appropriate positioning information to REM to set the relays in their appropriate positions.

The idea with Nixie is to actually manage the pool/spa operation from within njspc while using REM to control the hardware bits.

DrJeff, 2021-03-12:
I want in!

DrJeff, 2021-03-15:
@rstrouse 
Got I2c connected to relay board and EZO, now how do I connect the I2c relays to njspc? To control them? Triggers?
[image]

tagyoureit, 2021-03-17:
This isn't implemented in the UI yet.

DrJeff, 2021-03-17:
> This isn't implemented in the UI yet.

Ok thanks

DrJeff, 2021-03-18:
I setup Generic Devices Pressure Transducer and Ads1115 how does the voltage/psi get sent to the Generic device?
[image]
[image]

To use solar heating in Virtual Controller how is that set.

How Do I set Spillway in Virtual Controller

rstrouse, 2021-03-19:
Create a feed from the ads channel to the generic device.

DrJeff, 2021-03-19:
That works, simple.

rstrouse, 2021-03-20:
If you pull the latest REM the triggers are enabled for i2c and GPIO over MQTT.

DrJeff, 2021-03-20:
Sweeeeeet! Thanks again

rstrouse, 2021-03-20:
I just checked in a javascript fix for when creating a new trigger.  It wasn't showing the relay # when using an i2c relay.

After you are done adding your triggers go to the general tab and reset server.  Right now it doesn't dynamically subscribe to MQTT topics or open/close new sockets.

DrJeff, 2021-03-22:
[image]
Ok got some issues with I2c not sure what it is but I have no control over relays see the problem below is there a replay in REM. Also I couldn't figure out how REM CHEM was supposed to get the PH and ORP into njspc but figured that out with the feeds. 😀 
[image]
But now I don't see how to use the Intellichlor60 for the ORP I check the box but what else do I need to do? Set the chlorine Generator to something?

Another thing to use Solar How do I do that with my setup I add it to Virtual Heaters but gives ID of -1. So how does njspc see it and use it?  How is that tied to a relay I assume I can use a feed to njspc with the "name" I set but how do I turn it on from the front end njspc?

```
1|REM  | silly: Setting Dirty... true 12489
1|REM  | debug: Persisting Configuration data...
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | silly: Setting Dirty... true 2396
1|REM  | silly: Setting Dirty... true 2434
1|REM  | silly: Sending on MQTT Channel {"eventName":"  pool/state/chem/voltage","value":1.01}
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Atlas_EZO-pH command R bytes written:1 result:8.090
1|REM  | silly: Setting Dirty... true 3126
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Atlas_EZO-pH command T,? bytes written:3 result:?T,13.33
1|REM  | silly: Setting Dirty... true 3435
1|REM  | silly: Sending on MQTT Channel {"eventName":"pool/state/chem/ph","value":8.09}
1|REM  | verbose: Emitting: /chemController : {"pHLevel":8.09,"id":50}
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: ORP command R bytes written:1 result:747.5
1|REM  | silly: Setting Dirty... true 5224
1|REM  | silly: Sending on MQTT Channel {"eventName":"pool/state/chem/orp","value":747.5}
1|REM  | verbose: Emitting: /chemController : {"orpLevel":747.5,"id":50}
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | silly: Setting Dirty... true 7434
1|REM  | silly: Sending on MQTT Channel {"eventName":"  pool/state/chem/voltage","value":1.01}
[... 33 lines cut ...]
1|REM  | setDeviceState: gpio:1:38 - {"state":false}
1|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: ENODEV: no such device, read
1|REM  | verbose: 192.168.0.179 PUT /state/device/gpio:1:36 {"state":false}
1|REM  | info: [11:01:57 PM] 192.168.0.179 PUT /state/device/gpio:1:36 {"state":false}
1|REM  | setDeviceState: gpio:1:36 - {"state":false}
1|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: ENODEV: no such device, read
1|REM  | debug: Atlas_EZO-pH command T,13.33 bytes written:7 result:
1|REM  | silly: Setting Dirty... true 10237
1|REM  | debug: Persisting Configuration data...
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
[... cut ...]
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
1|REM  | debug: ORP command Status bytes written:6 result:?STATUS,P,5.02
1|REM  | silly: Setting Dirty... true 1205
1|REM  | silly: Setting Dirty... true 1205
1|REM  | debug: Executed read command 0x12 byte read:0xff
1|REM  | debug: Executed read command 0x13 byte read:0xff
1|REM  | error: Multi-Relay Board Read Command: Invalid I2C command undefined
```

rstrouse, 2021-03-22:
Which relay board do you have?  I don't have all of these so the command list is from the documentation on the raw chip.  It is interesting that the 0x12 and 0x13 registers are returning 0xFF.  How many relays are on your board?

DrJeff, 2021-03-22:
The relays are connected to an expansion board this is one of them,
https://www.amazon.com/gp/product/B07JGSZJ73/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

There are 8 relays on on one chip on another address there are 16 relays but both chips are mcp23017

I can see and control both chips from the same pi via Node Red

rstrouse, 2021-03-22:
There should be only one chip with 2 8-bit registers on the mcp23017.  Register 0x12 is for the first 8 I/O and register 0x13 for the second 8.  Perhaps the I/O direction is not persisted on them when first set up.

DrJeff, 2021-03-22:
> There should be only one chip with 2 8-bit registers on the mcp23017. Register 0x12 is for the first 8 I/O and register 0x13 for the second 8. Perhaps the I/O direction is not persisted on them when first set up.

So re-setup only one even with different addresses?

rstrouse, 2021-03-22:
Oh so you have 2 mcp23017 boards on different addresses for a total of 32 I/Os?

DrJeff, 2021-03-22:
> Oh so you have 2 mcp23017 boards on different addresses for a total of 32 I/Os?

Si

DrJeff, 2021-03-22:
Even if I only have one MCP connected same behavior

rstrouse, 2021-03-22:
That should be fine so long as they have separate addresses.  Do both chips show in REM on separate addresses?

I am going to add some code to initialize these as outputs for each of the relays you have defined.   It should be reading 0x12 to get the current state of the first 8 (on each board) and 0x13 to get the state for the second 8.  

When you see a return of 0xFF
`1|REM  | debug: Executed read command 0x12 byte read:0xff`
It should indicate that the first 8 relays are triggered.

DrJeff, 2021-03-22:
> That should be fine so long as they have separate addresses. Do both chips show in REM on separate addresses?
> 
Yes They are seen separately
> I am going to add some code to initialize these as outputs for each of the relays you have defined. It should be reading 0x12 to get the current state of the first 8 (on each board) and 0x13 to get the state for the second 8.
> 
👍🏽

rstrouse, 2021-03-22:
Go ahead and pull REM.  It now initializes the I/O registers for the chip to outputs for every defined relay.  There is also a dump of all the registers on the chip (sans GPIO).  This lists the settings.  Mine are 0 because I do not have an mcp relay.  Please provide a screenshot so I can see the value of all these registers after you get set up.
[image]

DrJeff, 2021-03-23:
Ok was working but then setup other and lost all. Address 39 was the working toggle of relays ok. then setup address 38 and lost both.
[image]
[image]
[image]

rstrouse, 2021-03-23:
Alright, I think I figured this out.  Turns out the pin directions are exactly opposite of how I read the docs.
0 = output (what we want).
1 = input (what we don't want to control a relay).

In your case we were setting the IODIRA/B to 255 which meant add ports were inputs.

I also find it a bit weird that your IPOLA/B settings are set to reversed for some of the pins.  Is this intentional?

DrJeff, 2021-03-23:
> Alright, I think I figured this out. Turns out the pin directions are exactly opposite of how I read the docs. Pull REM.
> 0 = output (what we want).
> 1 = input (what we don't want to control a relay).
> 

Nice

> In your case we were setting the IODIRA/B to 255 which meant add ports were inputs.
> 

Makes since now

> I also find it a bit weird that your IPOLA/B settings are set to reversed for some of the pins. Is this intentional?
Not Sure maybe because I set them in Node Red?

DrJeff, 2021-03-23:
Yes, Thanks
Here goes both fully functional!
[image]
[image]

Ok now that the lighting relay board is in REM can nixie control Intellibrites with just the relays? Im currentlly doing it via Node Red
[image]

DrJeff, 2021-03-23:
Oops too excited too fast on the relays on address 39 1-6 not working?
[image]

rstrouse, 2021-03-23:
Hmmm.... that can't be.  There is a single register being set for the first 8 relays.  Is anything else hitting those relays?

rstrouse, 2021-03-23:
I wonder if the latch has to be written instead of the GPIO,   The docs couldn't be more unclear.  The GPIO docs say this.
[image]
But the OLAT docs say this.
[image]

rstrouse, 2021-03-23:
Ok so pull REM.  It is now writing to OLAT instead of directly to GPIO.  I can only imagine that this is the difference between writing instantaneous values vs writing a latched value.  The docs are pretty dumb.

[87 more replies]

closed 2021-05-01 by rstrouse