rstrouse/relayEquipmentManager #5: Virtual Light Relay?

theRealMCQwerty, 2021-04-16:
I have 8 individually wired lights. (5 microbrites, 1 globrite and 2 magicsteam laminars).
Each is individually wired back to my panel area but I only have 4 transformers and an IntelliCenter i8PS. (3 of the relays are used by other non-light equipment).

So currently I have multiple lights wired together into the 4 transformers and then the transformers back to 4 individual relays in the IC.
I would like (if possible) to create 'virtual' lights (aux circuits) within the IC, assign the light type (Globrite/Magicstream) and then  listen to these events using REM to control a low voltage relay via the pins of my RPi.

The idea being that I could switch the low voltage side each of the 8 lights using relays controlled by REM but still interact with these individual lights as if they were individually wired to transformers that were wired back to 8 different aux relays.

I hope that makes sense.

Is there a way of doing this now or is it something that could be implemented?

@rstrouse I hate the prehistoric 'switch on/off multiple times to change mode' thing too but I am stuck with it for the foreseeable. 😞 

Thanks!

rstrouse, 2021-04-19:
We are working on direct connectivity between njspc and REM to allow you to define additional relay circuits on njspc tied to REM.  These circuits are then exclusively controlled via REM.  Valves and Heaters are also getting this treatment.  This is how the REM Chem stuff works today which I am proud to say has been operating like a champ.
[image]
I included a pic of my expansion panel I created that will allow me to add all manner of ridiculous stuff to the pool and backyard.
[image]

theRealMCQwerty, 2021-04-19:
@rstrouse  That is going to be excellent. Controlling valve relays via Feature Circuits is working great for me but it was quite a bit of config work to get 14 of them tied in with REM 😄 
Looking forward to the update, do you have an ETA? I am currently running the 'next' branch of njspc.

Your expansion panel looks nice! I looked at the relay hat for my Pi but I needed so many relays I decided to go with a sperate module, no where near as neat though.
Is that fire control specifically for the box or is it part of a bigger setup?

rstrouse, 2021-04-19:
Fire control is to control the fire features on the pool.  The transformer you see there controls the gas valve and ignition sequencer.  

Next is no longer being updated.  We just found it too onerous to manage multiple versions and eventually everybody found themselves on next so at some point.  Switching repos became a support issue in and of itself.  I would suggest you switch back to master.   I can't give a definite ETA on when this all comes together as this is one of those things that I throw some code at it when I have time but I hope in the next couple of weeks to have it all coded.

rstrouse, 2021-04-24:
Ok so if you grab njsPC master, dashPanel, and REM you should now be able to add an AUX circuit and assign it to IntelliBrite or GloBrite.  If you do that, when you select a theme for it, it will send the appropriate number of on/offs to the relay to set the theme or color.  You can assign the circuit directly to REM within dashPanel.  This is also how the valves can work for you now and you can name the GPIO pins so you don't have to remember what is connected to what.

Does this help?

theRealMCQwerty, 2021-04-26:
@rstrouse This looks fantastic! I need to wire up the relays to fully test it out but watching REM flash when adjusting a virtual light is making me happy.
Would it be possible to add special features such as 'hold' and 'recall' to lights in DashPanel?
Additionally (sorry to ask for more...), but do you think you could add the special features for the Maigcstream Laminars; "Sync mode', 'Random mode', 'Party mode', 'Solenoid thumper' control and 'Reset'?

Thank you!

rstrouse, 2021-04-26:
Yes they are coming soon.

rstrouse, 2023-05-22:
Closing this issue for now as all of the options have been added to dashPanel.

closed 2023-05-22 by rstrouse