rstrouse/relayEquipmentManager #6: SyntaxError: Unexpected end of JSON input

theRealMCQwerty, 2021-05-05:
Running REM latest build as of 5/5/21

I am getting the following exception in the console that crashes the execution of REM:

```
/home/pi/relayEquipmentManager/config/Config.ts:29
                            this._cfg = fs.existsSync(this.cfgPath) ? JSON.parse(fs.readFileSync(this.cfgPath, "utf8")) : {};
                                                                           ^
SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at FSWatcher.<anonymous> (/home/pi/relayEquipmentManager/config/Config.ts:29:76)
    at FSWatcher.emit (events.js:315:20)
    at FSEvent.FSWatcher._handle.onchange (internal/fs/watchers.js:186:12)
```

I am assuming the config file in question is `controllerConfig.json` so I have attached it for reference but I put it though a json linter and it appears to be valid.

[controllerConfig.zip](https://github.com/rstrouse/relayEquipmentManager/files/6427514/controllerConfig.zip)

I am not sure what exactly I changed that caused this but the error. Sometimes it crashes shorlty after startup, sometimes it runs for a while and I can send it commands from NJSPC and then later it crashes. 
I don't think the crashes have anything to do with any magic sequence of input events.

I turned on Debug level logging, below is a trace of some output before a crash occured:

```
setDeviceState: gpio:1:21 - {"isOn":false}
debug: Writing Pin #1:21 -> GPIO #9 to 0
debug: Persisting Configuration data...
verbose: 192.168.1.80 PUT /state/device/gpio:1:19 {"isOn":true,"latch":10000}
info: [9:01:07 AM] 192.168.1.80 PUT /state/device/gpio:1:19 {"isOn":true,"latch":10000}
setDeviceState: gpio:1:19 - {"isOn":true,"latch":10000}
debug: Writing Pin #1:19 -> GPIO #10 to 1
verbose: 192.168.1.80 PUT /state/device/gpio:1:19 {"isOn":true,"latch":10000}
info: [9:01:07 AM] 192.168.1.80 PUT /state/device/gpio:1:19 {"isOn":true,"latch":10000}
setDeviceState: gpio:1:19 - {"isOn":true,"latch":10000}
debug: Writing Pin #1:19 -> GPIO #10 to 1
info: [9:01:08 AM] 192.168.1.88 GET /device {}
verbose: 192.168.1.80 PUT /state/device/gpio:1:19 {"isOn":false}
info: [9:01:08 AM] 192.168.1.80 PUT /state/device/gpio:1:19 {"isOn":false}
setDeviceState: gpio:1:19 - {"isOn":false}
debug: Writing Pin #1:19 -> GPIO #10 to 0
info: [9:01:08 AM] 192.168.1.53 GET /device {}
debug: Persisting Configuration data...
info: [9:01:18 AM] 192.168.1.88 GET /device {}
info: [9:01:18 AM] 192.168.1.53 GET /device {}
info: [9:01:28 AM] 192.168.1.88 GET /device {}
info: [9:01:28 AM] 192.168.1.53 GET /device {}
info: [9:01:38 AM] 192.168.1.88 GET /device {}
info: [9:01:38 AM] 192.168.1.53 GET /device {}
info: [9:01:48 AM] 192.168.1.88 GET /device {}
info: [9:01:48 AM] 192.168.1.53 GET /device {}
info: [9:01:58 AM] 192.168.1.88 GET /device {}
info: [9:01:58 AM] 192.168.1.53 GET /device {}
info: [9:02:08 AM] 192.168.1.88 GET /device {}
info: [9:02:08 AM] 192.168.1.53 GET /device {}
info: [9:02:18 AM] 192.168.1.88 GET /device {}
info: [9:02:18 AM] 192.168.1.53 GET /device {}
info: [9:02:28 AM] 192.168.1.88 GET /device {}
info: [9:02:28 AM] 192.168.1.53 GET /device {}
info: [9:02:38 AM] 192.168.1.88 GET /device {}
info: [9:02:38 AM] 192.168.1.53 GET /device {}

/home/pi/relayEquipmentManager/config/Config.ts:29
                            this._cfg = fs.existsSync(this.cfgPath) ? JSON.parse(fs.readFileSync(this.cfgPath, "utf8")) : {};
                                                                           ^
SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at FSWatcher.<anonymous> (/home/pi/relayEquipmentManager/config/Config.ts:29:76)
    at FSWatcher.emit (events.js:315:20)
    at FSEvent.FSWatcher._handle.onchange (internal/fs/watchers.js:186:12)

```

tagyoureit, 2021-05-05:
ControllerConfig holds your data. This is an error coming from 'config.json' in the root of the app. Can you inspect that file?

theRealMCQwerty, 2021-05-05:
Looks like valid json also (attached)

[config.zip](https://github.com/rstrouse/relayEquipmentManager/files/6429131/config.zip)

tagyoureit, 2021-05-05:
I don't see anything wrong with your config files either.  This section of code is watching the config.json for any file changes and reloading it if it sees the file change.  Is there any chance that another process is changing the file?  Can you check to make sure you didn't run out of disk space on the drive?

theRealMCQwerty, 2021-05-06:
So it's not a disk space issue:
[image]

I can't think what would be modifying the config file. I restarted REM and then modified the permissions on the config.json file to read-only. So far this seems to have solved the problem, (or at least validated your theory that the file was being modified),

Not a long-term solution though since REM crashes on boot if config.json is read-only, so when I loose power or otherwise need to reboot it will require manual intervention.

Hmmm... 😕

rstrouse, 2021-05-06:
I'll have a look.  I am surprised that this code is being called actually.  There should be nothing periodically writing that file from inside REM.  It appears whatever is doing it is locking the file so when REM tries to read it back it gets an empty string which is not valid.  

The reason why I suspect it is an empty string is because `Unexpected end` typically means there are no bytes on the read from the file.  When a file is locked that is what fswatcher returns.  By chance do you have this file open in an editor anywhere?

rstrouse, 2021-05-06:
Pull REM.  I added some code to detect the file busy state.  It still needs to be able to write to the file but it should not crash when something is modifying it.  It also now reports the when something may be writing the file and has locked the file but does not crash.

theRealMCQwerty, 2021-05-07:
I am confident I did not have any other file editors open.
I did have PM2 setup to _try_ to automate the start/restart of REM. I followed the instructions in the Wiki but I freely admit I have no idea what I am doing so I just copied the webclient entry and modified if for REM, this worked for dashpanel but didn't work for REM. see screenshot below:
[image]
Copy of `ecosystem.config.json` attached.
[ecosystem.config.zip](https://github.com/rstrouse/relayEquipmentManager/files/6438494/ecosystem.config.zip)

As I say, no idea what I am doing, but do you think PM2 could have been writing to the REM config file?

I did pull the latest version and changed the write permissions on `config.json` back to how they were. I started REM (still in a console window due to my lack of PM2 knowledge) and it started fine. It has been running for hours and no crashes.

I have been running with Debug logging (to console) and I have not seen any reports about it being unable to access the config file, but not sure what I am looking for there.

I have removed REM (and webclient) from my PM2 setup now so @rstrouse  I would be happy to test if you wanted to revert the change. Or I am happy to continue with current state since everything appears to be working but it may remain an unsolved mystery.

@tagyoureit may I request an update to the [Wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/Automatically-start-at-boot---PM2-&-Systemd) to include REM and DashPanel and exclude Webclient. For me and any other PM2-challenged knuckleheads 😄 

p.s.
..and thank you both for an excellent and getting-more-excellent-every-day system as well as the super attentive support.
I am loving this project!

p.p.s.
Yesterday, using REM (between crashes 😄) and some messy wiring I was able to set my pool lights to the Mexican flag colors for cinco de mayo...using only one transformer! 😉 

Gracias amigos!

theRealMCQwerty, 2021-05-07:
Thank you!
I was following the instructions in the Wiki and it says to use `.js` but 🤷‍♂️
Using your new file everything (including REM) has started up after a reboot and is working nicely.
I will let it run for a while but I am not anticipating any issues.

Do you think the PM2 mess was the source of the config file editing causing the issue with REM?

rstrouse, 2021-05-07:
Yes because every time REM wrote state to its data directory PM2 would kill REM before it completed.  Then at the same time it was force closing it, REM was forced into another close because it tried to persist and files that were currently open.  

In that instance REM was both trying to comply with the shutdown and clean itself up over and over again until it collided with the files being locked.

tagyoureit, 2021-05-07:
I just updated the [pm2 wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/Automatically-start-at-boot---PM2-&-Systemd) to only watch source file directories.  The whitelist approach should work better than the blacklist.

tagyoureit, 2021-05-07:
And yes... you can use either a [JS file](https://pm2.keymetrics.io/docs/usage/application-declaration/#generate-configuration) that uses a module.exports format or [.json](https://pm2.keymetrics.io/docs/usage/application-declaration/#multiple-json) with individual application startup files.

p.s. and p.p.s. Thank you for the kind words!

closed 2021-05-07 by theRealMCQwerty