rstrouse/relayEquipmentManager #7: SM relay8 mappings don't match official firmware

cmc0619, 2021-05-09:
running the 8relay command to enable a relay doesn't sync with what REM sees the relay as for all relays and vice-versa.  I made a table of how they're currently mapping. So for example, when I activate relay button 8 in REM it enables the relay SM considers 6. 1,4,7 are correct.

SM official/REM button
1/1
2/3
3/2
4/4
5/6
6/8 
7/7
8/6

rstrouse, 2021-05-10:
I’ll have a look.  They did some weirdness on the register mapping.  Unfortunately I had to cross reference the C code with the header file and probably got it mixed up.

rstrouse, 2021-05-10:
Ok after looking at this I now realize there are 2 different repos.

There is this one that has one mapping
https://github.com/SequentMicrosystems/8relay-rpi

And there is this one that has another mapping.
https://github.com/SequentMicrosystems/8relind-rpi

The boards look identical.  Which one do you have?

cmc0619, 2021-05-11:
Awesome! relay8 was the original, I have a couple of those on shelves in the basement, 8relay is the new one, 8relind was the old original Mega-IO Industrial board that was released between relay8 and 8relay.  Don't have one of those though.  
I think there might be a few more types of 4 relays boards too.  Though, I don't think you'll find them in the wild.  Some of the SM stuff had very brief runs.

The Mega-BAS does 1k thermistors on it's 0-10v boards for sure, selectable by jumper.  The Mega-IND doesn't have the selector jumpers so I'm assuming it's hardcoded to voltage ADC only so I'm back at voltage dividers. 

I'm not messing with the WDC or RTC either.  Like you said RTC is a meh feature for me and this is my 3rd season with a rPi controller on my pool and it's never frozen up or failed so WDC is meh too.  I'm not using the Ixian, that's Katodude.  my plan is to get a plain ole (industrial) probe and go direct without the middleman.

I will do a git pull; npm i and check it out and update.  Thank you!

cmc0619, 2021-05-11:
I have a v3.  I selected v3 in the pulldown and the relays match perfectly now