rstrouse/relayEquipmentManager #8: No way to configure 10k thermistor on ADS1115

jeffegg, 2021-05-14:
Trying to add a thermostat to my test system. I cannot see a way anymore to configure a connection to the ADS1115 to send njsPC anything but the voltage. I tried to setup a generic device and only option is Pressure Transducer, cannot find another option in the GUI for this. Also I seem to be allowed to send channel 3 only, not an issue, now. But may be down the road

Here is a snippet of output from my logs 
0|REM    | info: Stopped I2c ADS1115
0|REM    | info: Closed i2c Bus #1
0|REM    | info: Initializing i2c Interface
0|REM    | info: Detecting i2c Buses
0|REM    | info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
0|REM    | info: Initializing i2c Bus #1
0|REM    | info: Scanning i2c Bus #1
2|njsPC  | [5/14/2021, 12:42:10 AM] info: New socket client connected secretkey:) -- xxx.xxx.xxx.xxx
0|REM    | info: Connected to njsPC - automatic : http://xxx.xxx.xxx.xxx:4200
0|REM    | info: Binding circuit from njsPC - automatic to pin 1-37
0|REM    | debug: Created filter expression if(!(data.isOn == true)) return false; else return true;
0|REM    | debug: Created filter expression if(!(data.isOn == false)) return false; else return true;
0|REM    | debug: Created filter expression if(!(data.isOn == false)) return false; else return true;
0|REM    | debug: Created filter expression if(!(data.isOn == true)) return false; else return true;
0|REM    | debug: Created filter expression if(!(data.isOn == true)) return false; else return true;
0|REM    | debug: Created filter expression if(!(data.isOn == false)) return false; else return true;
0|REM    | [ 32, 60, 72 ]
0|REM    | info: Found I2C device Unknown at address: 32 - (0x20)
0|REM    | info: Found I2C device Unknown at address: 60 - (0x3c)
0|REM    | info: Found I2C device ADS1115 at address: 72 - (0x48)
0|REM    | info: Device ADS1x15 ADC initialized for i2c bus #1 address 72
0|REM    | info: i2c Bus #1 Initialized
0|REM        | verbose: Emitting: /temps : {"waterSensor1":1.63}

Setup in REM:
[image]
[image]

jeffegg, 2021-05-14:
And to add deleted and repulled njsPC, dashPanel, and REM from GIT about 3 hours ago, so should be on the latest greatest.

rstrouse, 2021-05-14:
I will take a look.  The Generic device used to have a 10k Thermistor output so maybe it got misclassified for SPI outputs only.  The way this works is you feed the voltage output to the 10k Thermistor device and it does all the curve fitting and resistance output for you.

rstrouse, 2021-05-15:
I added the profile for a 10k temperature probe.  You should feed the raw ADC value from the channel.  The voltage won't have the same resolution but it should also work.  When configuring the probe in the Generic devices section tell it what the input units are and set the bitness to the input value.

I also do not have this equipment set up so I was not able to check all the ranges with the ADC.

jeffegg, 2021-05-16:
I can make a 10k thermister generic object and route the raw adc value to it. Only issue is when trying to create the feed for new generic object I hit an error with no real message. 

[image]

jeffegg, 2021-05-16:
Issue seems to be in pages/GenericDevices.js
.on('additem', function (evt) {
                    $.getLocalService('/config/options/genericDevices/' + o.busNumber + '/' + o.address + '/feeds', null, function (feeds, status, xhr) {
                        feeds.feed = { isActive: true };
                        self._createFeedDialog('dlgAddI2cFeed', 'Add Feed to I2c Device', feeds);
                    });
In the debugger o.busNumber and o.address don't exist, and get set to undefined. Didn't dig further than this
   
0|REM  | debug: Persisting Configuration data... /home/pi/relayEquipmentManager/data/controllerConfig.json
0|REM  | verbose: xxx.xxx.xxx.xxx GET /config/options/genericDevices/undefined/undefined/feeds?null {}
0|REM  | info: [11:19:49 PM] xxx.xxx.xxx.xxx GET /config/options/genericDevices/undefined/undefined/feeds?null {}

rstrouse, 2021-05-17:
Ok so I don't have any generic devices at the moment but I reworked the code so that it does a better job at setting up the initial device including the feeds.  There are no triggers yet but we will get to them if they are fed from another interface.

Test this out with your probe.  I am pretty sure I replicated the accuracy of the SPI probe but let me know because there are some differences between the ADCs.

jeffegg, 2021-05-17:
Thanks. I'm now able to feed temperatures to njsPC! Is the expectation when using the raw ADC Value that we need to use the calibration to take into account the reference voltage and PGA we are using? 

As far as accuracy on both raw ADC and voltage, temperature seems to "step only +/- 2F.  I think the interpolation function is biasing the values too heavily to the value.low side of the calculation, causing this "step". I have some calculations below as an example. When I used a normal linear interpolation the values changes as I would expect.

I can file a new issue for this if you prefer to track separately as this issue is different than the one originally filed and isn't devastating, just a small delta in the temperature (and not really important :) ) 

In the example below I'm "heating" the thermistor (holding it), I have a offset set to -36F on the raw ADC value
in this snippet of the logs, I enabled the log output of the range for the interpolate. Looking at the data it always returns 80.91 as the temperature, but the adcValues are suggesting the number should be moving from 79 to 81 (or 115 to 116 degrees without the correction) .

info: Interpolating map thermistor10k {"low":{"input":4006,"value":117},"high":{"input":4182,"value":115}}
verbose: Emitting: /temps : {"waterSensor2":80.91}
verbose: Feed sending waterSensor2: 80.91
verbose: Feed sending adcValue: **19179** (I calculated input was **4137**)

info: Interpolating map thermistor10k {"low":{"input":4006,"value":117},"high":{"input":4182,"value":115}}
verbose: Emitting: /temps : {"waterSensor2":80.91}
verbose: Feed sending waterSensor2: 80.91
verbose: Feed sending adcValue: **19072** (I calculated input was **4104**)

info: Interpolating map thermistor10k {"low":{"input":4006,"value":117},"high":{"input":4182,"value":115}}
verbose: Emitting: /temps : {"waterSensor2":80.91}
verbose: Feed sending waterSensor2: 80.91
verbose: Feed sending adcValue: **19008** (I calculated input was **4085**)

info: Interpolating map thermistor10k {"low":{"input":4006,"value":117},"high":{"input":4182,"value":115}}
verbose: Emitting: /temps : {"waterSensor2":80.91}
verbose: Feed sending waterSensor2: 80.91
verbose: Feed sending adcValue: **18989** (I calculated input was **4079**)

diff = (4182-4006) / val
value  = 117+ ((115 - 117) * diff)
value  = 117+ ((115 - 117) * ((4182-4006) / val))

value  = 117+ ((115 - 117) * ((4182-4006) / **4137**)) = 116.9149
value  = 117+ ((115 - 117) * ((4182-4006) / **4104**)) = 116.9142
value  = 117+ ((115 - 117) * ((4182-4006) / **4085**)) = 116.9138
value  = 117+ ((115 - 117) * ((4182-4006) / **4079**)) = 116.9137

Using linear interpolation:
slope = (y2 - y1) / (x2 - x1)
y3 = y1+ slope x (x3 - x1)

info: Interpolating map thermistor10k {"low":{"input":4006,"value":117},"high":{"input":4182,"value":115}}
slope = (117-115)/(4006-4182) = -0.01136363636363636363636363636364
Y3 = 115 + -0.01136363636363636363636363636364 x (adc - 4182)

115 + (-0.01136363636363636363636363636364 * (**4137** - 4182)) = 115.5113
115 + (-0.01136363636363636363636363636364 * (**4104** - 4182)) = 115.8864
115 + (-0.01136363636363636363636363636364 * (**4085** - 4182)) = 116.1023
115 + (-0.01136363636363636363636363636364 * (**4079** - 4182)) = 116.1704

rstrouse, 2021-05-17:
Hmmm... there should not be a 36 degree difference here.  You are using an ads1115 not an ads1105 correct?  I'll have a look at my calculations to see what I am doing in there but the starting resistance from the adc value should not be that far off with a voltage divider.  Are you using a 10k resistor and a 10k probe on the input?

I assume the pga calculations should take care of themselves but if you are feeding voltage to the device then you must tell it what the reference voltage is.  So lets try to feed it the voltage from the adc.  You will need to change your feed from the adc to the probe then tell the probe the input is voltage and change the reference voltage to your input voltage.

I'll have a look at the interpolation code to see what is going on in the range mapping algorithm.

jeffegg, 2021-05-17:
Yes I just verified:
Room is 25C and reading near 10k on the thermistor and 10.1k across the resistor.
Using 5V (actually reading 5.3V) as the reference voltage and reading 2.65V on the analog input. 

This should be an ads1115, I can use all 4 inputs successfully and only the 16 bit values makes sense. But need to confirm it is not some knockoff under the microscope. Hard to move as currently using a 0805 sized resistor soldered between 2 wires while designing a proper board.

I actually setup 2 generic devices to test to get the offset.
1 using voltage feed and 1 using adc feed from the same sensor.
The voltage (and temperature) is reading properly (2.65V - 72.9F) when being feed the voltage. But the raw adc value is off.
Looking at the way you calculate voltage you are doing adcValue/Max * PGA, which makes sense. The way the hardware is implemented looks like all inputs are muxed and then sent through the PGA to the actual ADC. So the PGA is scaling the inputs before being sampled to make sure the inputs are able to use the full adc range. I currently set to 4.096 since unless there is a fault, I should never see above voltage on the inputs.

[image]
[image]
[image]

rstrouse, 2021-05-17:
Ok so this is where I am getting bound up with this.  The initial testing was done using an MCP3008 chip. This has a linear balance on the output values so the raw ADC values do not have a gain correction.  This is why the ADC value is correct for that chip and notsomuch for the ADS.  

For the ADS chips we are taking the pga into account when we calculate the voltage so voltage input is what you should be using.  I incorrectly assumed that the ADS was operating in a similar fashion to the MCP chips regarding their output in which case the resistance over the output values do not have pga values to contend with.

I am also changing the interpolation function to linear.  While the original calculation fit the overall curve, and did perform better than a straight Steinhart calculation, I think the assumptions over the temperature range is flawed.  It is like sorting the pepper out of a pile of fly poop.

rstrouse, 2021-05-17:
I checked in the changes.  Please pull REM and hard refresh your browser so that it gets all the script changes.  I have confirmed that the raw adc values would somehow need to be scaled to the gain if we want to use them.  After looking at it there is no need for this since the voltage values certainly do what we want.

jeffegg, 2021-05-18:
Looks good now seeing changes around 1/3 of a degree F. Using voltage is fine for this and seems maybe some additional rounding errors but nothing to care about. Looks like this issue can be closed

Curve fitting is one of those annoying topics. It depends on the range. If the points are sufficiently close and the graph is relatively smooth, linear usually makes sense.

rstrouse, 2021-05-18:
Awesome thanks for helping sort this out.

closed 2021-05-18 by rstrouse