rstrouse/relayEquipmentManager #10: Stack Trace Saving Pi

bsileo, 2021-05-24:
I am getting this stack trace when I try to save a configuration on Pin 35:

[image]

Error: Error saving pin 1-35
    at GpioPin.setPinAsync (/home/pi/relayEquipmentManager/boards/Controller.ts:1028:47)
    at Gpio.setPinAsync (/home/pi/relayEquipmentManager/boards/Controller.ts:824:66)
    at /home/pi/relayEquipmentManager/web/services/Config.ts:68:43
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:281:22
    at param (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:354:14)
    at param (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:365:14)
    at param (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:365:14)
    at Function.process_params (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:410:3)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/relayEquipmentManager/web/Server.ts:257:17
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:317:13)

bsileo, 2021-05-24:
I have configured a total of 8 relays on this board. All 7 others seem to have worked without incident. Perhaps I have some conflict on this GPIO pin?  I don't have anything else running on the device except njsPC, DashPanel and REM.

rstrouse, 2021-05-27:
Ok for the world of me I cannot replicate this.  That doesn't mean that it isn't happening, it just means that the pin initialization is not failing for me.  However, I did add a more descriptive error than the goofball Error saving pin message.  So if you could please pull REM and try again.  I would like to see that error.

bsileo, 2021-05-27:
Here is a little more information...this is making me think something else is using this GPIO pin on my device.

Error: Error saving pin 1-35: EBUSY: resource busy or locked, write
    at GpioPin.setPinAsync (/home/pi/relayEquipmentManager/boards/Controller.ts:1028:47)
    at Gpio.setPinAsync (/home/pi/relayEquipmentManager/boards/Controller.ts:824:66)
    at /home/pi/relayEquipmentManager/web/services/Config.ts:68:43
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:281:22
    at param (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:354:14)
    at param (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:365:14)
    at param (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:365:14)
    at Function.process_params (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:410:3)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/relayEquipmentManager/web/Server.ts:257:17
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:317:13)

bsileo, 2021-05-27:
Digging into this a bit, I found this output. I do not know GPIO subsystems that well, but this gives me the impression something is setting pin 35 for input

root@poolcontroller:~# cat /sys/kernel/debug/pinctrl/3f200000.gpio-pinctrl-bcm2835/pins
registered pins: 54
pin 0 (gpio0) 0:? function gpio_in in hi; irq 0 (none)
pin 1 (gpio1) 0:? function gpio_in in hi; irq 0 (none)
pin 2 (gpio2) 0:? function gpio_out in lo; irq 199 (none)
pin 3 (gpio3) 0:? function gpio_in in hi; irq 0 (none)
pin 4 (gpio4) 0:? function gpio_in in hi; irq 0 (none)
pin 5 (gpio5) 0:? function gpio_out in hi; irq 200 (none)
pin 6 (gpio6) 0:? function gpio_out in hi; irq 201 (none)
pin 7 (gpio7) 0:? function gpio_in in hi; irq 0 (none)
pin 8 (gpio8) 0:? function gpio_in in hi; irq 0 (none)
pin 9 (gpio9) 0:? function gpio_in in lo; irq 0 (none)
pin 10 (gpio10) 0:? function gpio_in in lo; irq 0 (none)
pin 11 (gpio11) 0:? function gpio_in in lo; irq 0 (none)
pin 12 (gpio12) 0:? function gpio_in in lo; irq 0 (none)
pin 13 (gpio13) 0:? function gpio_out in hi; irq 202 (none)
pin 14 (gpio14) 0:? function alt0 in hi; irq 0 (none)
pin 15 (gpio15) 0:? function alt0 in hi; irq 0 (none)
pin 16 (gpio16) 0:? function gpio_out in hi; irq 203 (none)
pin 17 (gpio17) 0:? function gpio_in in lo; irq 0 (none)
pin 18 (gpio18) 0:? function gpio_in in lo; irq 0 (none)
pin 19 (gpio19) 0:? function gpio_in in hi; irq 0 (none)
pin 20 (gpio20) 0:? function gpio_in in hi; irq 0 (none)
pin 21 (gpio21) 0:? function gpio_out in hi; irq 204 (none)
pin 22 (gpio22) 0:? function gpio_in in lo; irq 0 (none)
pin 23 (gpio23) 0:? function gpio_in in lo; irq 0 (none)
pin 24 (gpio24) 0:? function gpio_in in lo; irq 0 (none)
pin 25 (gpio25) 0:? function gpio_in in lo; irq 0 (none)
pin 26 (gpio26) 0:? function gpio_out in lo; irq 205 (none)
pin 27 (gpio27) 0:? function gpio_in in lo; irq 0 (none)
pin 28 (gpio28) 0:? function gpio_in in hi; irq 0 (none)
pin 29 (gpio29) 0:? function gpio_in in hi; irq 0 (none)
pin 30 (gpio30) 0:? function gpio_in in lo; irq 0 (none)
pin 31 (gpio31) 0:? function gpio_out in hi; irq 0 (none)
pin 32 (gpio32) 0:? function gpio_out in lo; irq 0 (none)
pin 33 (gpio33) 0:? function gpio_in in lo; irq 0 (none)
pin 34 (gpio34) 0:? function gpio_in in hi; irq 0 (none)
**pin 35 (gpio35) 0:? function gpio_in in hi; irq 0 (none)**
pin 36 (gpio36) 0:? function gpio_in in hi; irq 0 (none)
pin 37 (gpio37) 0:? function gpio_in in lo; irq 0 (none)
pin 38 (gpio38) 0:? function gpio_out in hi; irq 0 (none)
pin 39 (gpio39) 0:? function gpio_in in lo; irq 0 (none)
pin 40 (gpio40) 0:? function alt0 in lo; irq 0 (none)
pin 41 (gpio41) 0:? function gpio_out in lo; irq 0 (none)
pin 42 (gpio42) 0:? function gpio_out in hi; irq 0 (none)
pin 43 (gpio43) 0:? function gpio_in in hi; irq 0 (none)
pin 44 (gpio44) 0:? function alt0 in hi; irq 0 (none)
pin 45 (gpio45) 0:? function alt0 in lo; irq 0 (none)
pin 46 (gpio46) 0:? function gpio_in in hi; irq 0 (none)
pin 47 (gpio47) 0:? function gpio_out in lo; irq 0 (none)
pin 48 (gpio48) 0:? function alt0 in lo; irq 0 (none)
pin 49 (gpio49) 0:? function alt0 in hi; irq 0 (none)
pin 50 (gpio50) 0:? function alt0 in hi; irq 0 (none)
pin 51 (gpio51) 0:? function alt0 in hi; irq 0 (none)
pin 52 (gpio52) 0:? function alt0 in hi; irq 0 (none)
pin 53 (gpio53) 0:? function alt0 in hi; irq 0 (none)
root@poolcontroller:~#

bsileo, 2021-05-27:
Ok an more info I found this in a forum post:

**gpio 35 is part of the wifi/SDIO, the change you saw is just normal activity as it talks to the wifi chip**

So I am betting this pin is locked up for that purpose by the kernel and this board I bought is trying to use it to control a relay. Seems like a pretty lame board design. i will have to see about using a different GPIO pin for it.

rstrouse, 2021-05-28:
That would do it.

rstrouse, 2021-05-28:
I couldn’t find anything in their docs.  Is this low level triggered?  If it is then the relays will engage on power on.  Maybe preferred for some things but it may trigger for a couple of seconds on startup if it is low level trigger.   Manufacturers save a buck because they don’t need a latching power regulator that way.

bsileo, 2021-06-03:
So far I have not seen any problems with relay state on power up. I'll keep monitoring to see if that is an issue and may have to go to a higher end board. So far it looks like the pins used are hardwired o this board with the exception of running a separate wire and opening a jumper to change the GPIO pin used. I may resort to that but it certainly makes the install messy compared to using the built in 40 pin header. 

I am still finding time to see if I can disable use of GPIO 35 in my kernel to free it up for this board.

bsileo, 2021-06-27:
Just to close things out on this issue, I worked around the problem by removing the jumpers on this board for relays 5-8 and then manually wiring them to other unused GPIO pins. Certainly less than ideal. It's possible this would not be needed with a different model RPi, but with my older Pi2B I could not find a way to eliminate the conflicts with the kernel drivers to free REM up.  Working fine now with different GPIO pins.

bsileo, 2021-06-30:
I am reopening this as I am now seeing this may be a different problem. I rewired the relays to separate GPIO pins to avoid any system conflicts. I am still having a problem with what I will call lazy config of the pins.  Here is how this occurs:

1. Reboot the Pi
2. After startup, everything "looks" fine.
3. At this stage, the Pump controls work fine. I can set the pump to levels 1,2,3,4 and the correct corresponding relays get turned on/off.
4. I have two additional circuits configured - the slide (relay 7 - GPIO 27/pin 13) and the lights (Relay 8 - GPIO 22/pin 15)
5. At this stage - neither of these works in Nixie. When I click them, they show yellow briefly but do not change states in the UI or on the physical relays.  

6. I see these errors when I click one of them:

3|REM             | info: [22:17:12] 192.168.2.183 PUT /state/device/gpio:1:15 {"isOn":true,"latch":10000}
3|REM             | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-15. Could not find pin in controller
1|poolController  | [6/29/2021, 22:17:12] error: Nixie: Error setting circuit state 2-Lights to true

Here is the setup configuration for that circuit in NJSPC:
[image]

Showing it is connected to "Relay 8 - wire"

Sure enough, that is Pin 15 in REM. Here is the configuration of it there:
[image]

6. If I now hit "Save Pin" in REM, it will update the state of the pin and the relay.
7. At this point, if I go back to NJSPC I can operate that circuit without any other changes needed.
8. I can repeat this for the Slide circuit and then it works as well. But as soon as I reboot the device, I have to return to REM and "Save Pin" on both my Slide and Light Pins (Pin 13 / 15) to get them working again. 
9.

bsileo, 2021-07-01:
Moved the lazy/lasy comment item to a new issue - really different than what started this one.

closed 2021-07-01 by bsileo