rstrouse/relayEquipmentManager #11: REM went blank

DrJeff, 2021-06-07:
Lost rem configuration is it pulling from a new location? My setup is blank slate? tried a git pull but still blank, I knew that shouldn't make a difference but why not.
here is my config
```
pi@rem:~/rem/data $ cat controllerConfig.json
{
  "controllerType": "raspi",
  "gpio": {
    "pins": [],
    "exported": []
  },
  "spi0": {
    "isActive": false,
    "busNumber": 0,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "spi1": {
    "isActive": false,
    "busNumber": 1,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "i2c": {
    "isActive": false,
    "buses": [],
    "detected": [
      {
        "driver": "bcm2835 (i2c@7e804000)",
        "path": "/sys/class/i2c-dev/i2c-1",
        "name": "i2c-1",
        "busNumber": 1
      }
    ]
  },
  "connections": [
    {
      "id": 1,
      "isActive": true,
      "options": {},
      "name": "njsPC - automatic",
      "type": "njspc",
      "protocol": "http:",
      "ipAddress": "192.168.0.179",
      "port": 4200,
      "userName": "",
      "password": "",
      "sslKeyFile": "",
      "sslCertFile": ""
    }
  ],
  "genericDevices": {
    "isActive": true,
    "buses": [],
    "detected": [],
    "options": {},
    "devices": []
  },
  "lastUpdated": "6/7/2021, 12:41:09 AM",
  "configVersion": 1
```
But my setup is gone
[image]
[image]
[image]

rstrouse, 2021-06-08:
Add the I2c bus back in.

DrJeff, 2021-06-08:
? was never removed, just lost dhcp address and after reboot I got the above

rstrouse, 2021-06-08:
There isn’t a check in there for net connections that would remove the I2c bus definition.  It almost looks like there was a load error on startup.  Perhaps I need to force the persistence to off if there is something catastrophic on load.  I’ll harden that up so it won’t persist if the server doesn’t completely come up.

DrJeff, 2021-06-08:
Whats the best plan of action for me, just add all my stuff again?

rstrouse, 2021-06-08:
Yes then back up the file.  I will dig through this but the only path I can see for this to happen is that the file load on startup failed without throwing an error.  Maybe it ended up being locked or the file ownership changed.

closed 2021-06-17 by DrJeff