rstrouse/relayEquipmentManager #12: Reading raw analog sensors on BeagleBone Black.

zombielinux, 2021-06-11:
I've got a BeagleBone Black, with a very nice breakout cape. 

Connected to it, is a 10K NTC thermistor in a voltage divider configuration. 

I'm not seeing a way in the UI to configure the onboard ADCs to actually read the temperature. 

Is that configured in the json directly?

rstrouse, 2021-06-11:
Is this an MCP cape on SPI?

zombielinux, 2021-06-11:
> Is this an MCP cape on SPI?

No. Its directly attached to the ADC GPIO pins (specifically P9.36)

rstrouse, 2021-06-12:
Yeah we are not reading any of those ADC pins.  I looked at those originally and the but I don't have a Beagle.  They bark too much.  I2c is supported as well as SPI.

zombielinux, 2021-06-12:
Ah. I see. Would that be terribly difficult to implement? I can hack away a
little at it myself, but no one knows their code like the author.

I’d be glad to test implantations as they’re developed.

rstrouse, 2021-06-12:
I would bet that it is file system based so it probably has a file that it writes in devices somewhere.  Maybe sysfs.

rstrouse, 2021-06-12:
You do know those pins have some power limits.

zombielinux, 2021-06-12:
It definitely does have some sysfs interfaces. And yes. They have some
power limits, but there are several examples of measuring a 10k notch
thermistor without issue. Other devices may require some additional
circuitry.