rstrouse/relayEquipmentManager #13: EZO pH and ORP  Not reading

DrJeff, 2021-06-16:
The pH and ORP are not being read here is the error what can I try?
```
1|REM  | error: Atlas_EZO-pH Error getting device status: Cannot read property 'split' of undefined
1|REM  | error: Error retrieving device status: Cannot read property 'split' of undefined
1|REM  | (node:4945) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'split' of undefined
1|REM  |     at AtlasEZOpH.getStatus (/home/pi/rem/i2c-bus/AtlasEZO.ts:214:34)
1|REM  |     at AtlasEZOpH.getDeviceInformation (/home/pi/rem/i2c-bus/AtlasEZO.ts:271:13)
1|REM  | (node:4945) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 99649)
1|REM  | error: ORP Error getting device status: Cannot read property 'split' of undefined
1|REM  | error: Error retrieving device status: Cannot read property 'split' of undefined
1|REM  | (node:4945) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'split' of undefined
1|REM  |     at AtlasEZOorp.getStatus (/home/pi/rem/i2c-bus/AtlasEZO.ts:214:34)
1|REM  |     at AtlasEZOorp.getDeviceInformation (/home/pi/rem/i2c-bus/AtlasEZO.ts:271:13)
1|REM  | (node:4945) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 99653)
```

rstrouse, 2021-06-17:
Check i2cdetect.  It appears that it isn’t responding on the bus.  Also if you are running their app it locks up the i2c bus addresses.

DrJeff, 2021-06-17:
Not running the App but i2cdetect shows. I can deactivate the sensor in REM and the sensor does not try to read, but when activated the pH sensor flashes red on reading like it can't understand the command?

DrJeff, 2021-06-17:
I stand corrected no the sensors are not showing up in i2c! They are lost what can cause i2c to get lost I lost all of them a week ago and not sure what caused even my relays they are back but not the EZOs. Any thoughts on how to jump start them?

DrJeff, 2021-06-17:
OK i did what I was scared to do I pulled the sensors out while live and reset them and that worked! strange but effective.

closed 2021-06-17 by DrJeff