rstrouse/relayEquipmentManager #14: Lazy Connections/Devices after reboot

bsileo, 2021-07-01:
I am reopening this as I am now seeing this may be a different problem. I rewired the relays to separate GPIO pins to avoid any system conflicts. I am still having a problem with what I will call lazy config of the pins.  Here is how this occurs:

1. Reboot the Pi
2. After startup, everything "looks" fine.
3. At this stage, the Pump controls work fine. I can set the pump to levels 1,2,3,4 and the correct corresponding relays get turned on/off.
4. I have two additional circuits configured - the slide (relay 7 - GPIO 27/pin 13) and the lights (Relay 8 - GPIO 22/pin 15)
5. At this stage - neither of these works in Nixie. When I click them, they show yellow briefly but do not change states in the UI or on the physical relays.  

6. I see these errors when I click one of them:

3|REM             | info: [22:17:12] 192.168.2.183 PUT /state/device/gpio:1:15 {"isOn":true,"latch":10000}
3|REM             | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-15. Could not find pin in controller
1|poolController  | [6/29/2021, 22:17:12] error: Nixie: Error setting circuit state 2-Lights to true

Here is the setup configuration for that circuit in NJSPC:
[image]

Showing it is connected to "Relay 8 - wire"

Sure enough, that is Pin 15 in REM. Here is the configuration of it there:
[image]

6. If I now hit "Save Pin" in REM, it will update the state of the pin and the relay.
7. At this point, if I go back to NJSPC I can operate that circuit without any other changes needed.
8. I can repeat this for the Slide circuit and then it works as well. But as soon as I reboot the device, I have to return to REM and "Save Pin" on both my Slide and Light Pins (Pin 13 / 15) to get them working again. 
9.

_Originally posted by @bsileo in https://github.com/rstrouse/relayEquipmentManager/issues/10#issuecomment-871048842_

rstrouse, 2021-07-01:
Are these the only two pins you have inverted?

bsileo, 2021-07-02:
> Are these the only two pins you have inverted?

No I actually have all of the pins set as inverted. This is getting me to On==On for this relay board.

bsileo, 2021-07-02:
**3|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-13. Could not find pin in controller**
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
3|REM  | info: [08:44:04] 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
3|REM  | debug: Writing Pin #1:29 -> GPIO #5 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
3|REM  | info: [08:44:04] 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
3|REM  | debug: Writing Pin #1:31 -> GPIO #6 to 1
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
3|REM  | info: [08:44:04] 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
3|REM  | debug: Writing Pin #1:33 -> GPIO #13 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
3|REM  | info: [08:44:04] 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
3|REM  | debug: Writing Pin #1:36 -> GPIO #16 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | info: [08:44:06] 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-13. Could not find pin in controller
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
3|REM  | info: [08:44:06] 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
3|REM  | debug: Writing Pin #1:29 -> GPIO #5 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
3|REM  | info: [08:44:06] 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
3|REM  | debug: Writing Pin #1:31 -> GPIO #6 to 1
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
3|REM  | info: [08:44:06] 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
3|REM  | debug: Writing Pin #1:33 -> GPIO #13 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
3|REM  | info: [08:44:06] 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
3|REM  | debug: Writing Pin #1:36 -> GPIO #16 to 0

bsileo, 2021-07-02:
From the Console log at Silly:

[7/2/2021, 08:45:55] error: undefined
[7/2/2021, 08:45:57] verbose: 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
[7/2/2021, 08:45:57] info: [08:45:57] 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
[7/2/2021, 08:45:57] debug: Writing Pin #1:29 -> GPIO #5 to 0
[7/2/2021, 08:45:57] verbose: 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
[7/2/2021, 08:45:57] info: [08:45:57] 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
[7/2/2021, 08:45:57] debug: Writing Pin #1:31 -> GPIO #6 to 1
[7/2/2021, 08:45:57] verbose: 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
[7/2/2021, 08:45:57] info: [08:45:57] 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
[7/2/2021, 08:45:57] debug: Writing Pin #1:33 -> GPIO #13 to 0
[7/2/2021, 08:45:57] verbose: 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
[7/2/2021, 08:45:57] info: [08:45:57] 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
[7/2/2021, 08:45:57] debug: Writing Pin #1:36 -> GPIO #16 to 0
[7/2/2021, 08:45:58] verbose: 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
[7/2/2021, 08:45:58] info: [08:45:58] 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
[7/2/2021, 08:45:58] error: undefined
[7/2/2021, 08:45:59] verbose: 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
[7/2/2021, 08:45:59] info: [08:45:59] 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
[7/2/2021, 08:45:59] debug: Writing Pin #1:29 -> GPIO #5 to 0
[7/2/2021, 08:45:59] verbose: 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
[7/2/2021, 08:45:59] info: [08:45:59] 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
[7/2/2021, 08:45:59] debug: Writing Pin #1:31 -> GPIO #6 to 1
[7/2/2021, 08:45:59] verbose: 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
[7/2/2021, 08:45:59] info: [08:45:59] 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
[7/2/2021, 08:45:59] debug: Writing Pin #1:33 -> GPIO #13 to 0
[7/2/2021, 08:45:59] verbose: 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
[7/2/2021, 08:45:59] info: [08:45:59] 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
[7/2/2021, 08:45:59] debug: Writing Pin #1:36 -> GPIO #16 to 0
[7/2/2021, 08:46:01] verbose: 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
[7/2/2021, 08:46:01] info: [08:46:01] 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
[7/2/2021, 08:46:01] error: undefined
[7/2/2021, 08:46:01] verbose: 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
[7/2/2021, 08:46:01] info: [08:46:01] 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
[7/2/2021, 08:46:01] debug: Writing Pin #1:29 -> GPIO #5 to 0
[7/2/2021, 08:46:01] verbose: 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
[7/2/2021, 08:46:01] info: [08:46:01] 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
[7/2/2021, 08:46:01] debug: Writing Pin #1:31 -> GPIO #6 to 1
[7/2/2021, 08:46:01] verbose: 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
[7/2/2021, 08:46:01] info: [08:46:01] 192.168.2.183 PUT /state/device/gpio:1:33 {"isOn":0}
[7/2/2021, 08:46:01] debug: Writing Pin #1:33 -> GPIO #13 to 0
[7/2/2021, 08:46:01] verbose: 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
[7/2/2021, 08:46:01] info: [08:46:01] 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
[7/2/2021, 08:46:01] debug: Writing Pin #1:36 -> GPIO #16 to 0

rstrouse, 2021-07-02:
I want to see these messages as REM is starting up.  For some reason the message you are getting is one that you would get if REM failed to take control of a given pin.  The controllerConfig.json file when it is in the bad state should also help.

bsileo, 2021-07-02:
ControllerConfig.json

{
  "controllerType": "raspi",
  "gpio": {
    "pins": [
      {
        "id": 3,
        "headerId": 1,
        "isInverted": false,
        "direction": "output",
        "triggers": [],
        "name": "Pin #1-3",
        "feeds": [],
        "isActive": true,
        "state": {
          "desc": "On"
        },
        "debounceTimeout": 0
      },
      {
        "id": 29,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 1",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 31,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 2",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 33,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 3",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 36,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 4",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 35,
        "headerId": 1,
        "isInverted": false,
        "direction": "output",
        "triggers": [],
        "name": "Relay 5 - Old",
        "feeds": [],
        "isActive": false,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 38,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 6 - Old",
        "feeds": [],
        "isActive": false,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 40,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 7 - Old",
        "feeds": [],
        "isActive": false,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 37,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 8 - Old",
        "feeds": [],
        "isActive": false,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 13,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 7 - Wire",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "on"
      },
      {
        "id": 11,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "GPIO 17",
        "feeds": [],
        "isActive": false,
        "debounceTimeout": 0,
        "state": "on"
      },
      {
        "id": 15,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 8 - Wire",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 16,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 5 - Wire",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "off"
      },
      {
        "id": 18,
        "headerId": 1,
        "isInverted": true,
        "direction": "output",
        "triggers": [],
        "name": "Relay 6 - Wire",
        "feeds": [],
        "isActive": true,
        "debounceTimeout": 0,
        "state": "off"
      }
    ],
    "exported": [
      2,
      5,
      6,
      13,
      16,
      35,
      27,
      22,
      23,
      24
    ]
  },
  "spi0": {
    "isActive": false,
    "busNumber": 0,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "spi1": {
    "isActive": false,
    "busNumber": 1,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "i2c": {
    "isActive": false,
    "buses": [],
    "detected": []
  },
  "connections": [
    {
      "id": 1,
      "isActive": true,
      "options": {},
      "name": "Jimmy's Pool",
      "type": "njspc",
      "protocol": "http:",
      "ipAddress": "192.168.2.183",
      "port": 4200,
      "userName": "",
      "password": "",
      "sslKeyFile": "",
      "sslCertFile": ""
    }
  ],
  "genericDevices": {
    "isActive": true,
    "buses": [],
    "detected": [],
    "options": {},
    "devices": []
  },
  "lastUpdated": "7/2/2021, 11:13:34",
  "configVersion": 1

bsileo, 2021-07-02:
Startup log with Silly Enabled:

Looks like this bindings file issue needs resolved. 

3|REM      | > relay-equipment-manager@7.0.0 start
3|REM      | > npm run build && node dist/app.js
3|REM      |
3|REM      |
3|REM      | > relay-equipment-manager@7.0.0 build
3|REM      | > tsc
3|REM      |

3|REM  | linux
**3|REM  | Error: Could not locate the bindings file. Tried:**

I just ran a fresh git pull and npm install to verify I have everything. No luck with that.  Maybe I need to reinstall?

Here is the full startup log:

3|REM      | > relay-equipment-manager@7.0.0 start
3|REM      | > npm run build && node dist/app.js
3|REM      |
3|REM      |
3|REM      | > relay-equipment-manager@7.0.0 build
3|REM      | > tsc
3|REM      |

3|REM  | linux
3|REM  | Error: Could not locate the bindings file. Tried:
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/build/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/build/Debug/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/build/Release/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/out/Debug/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/Debug/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/out/Release/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/Release/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/build/default/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/compiled/12.9.1/linux/arm/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/addon-build/release/install-root/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/addon-build/debug/install-root/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/addon-build/default/install-root/i2c.node
3|REM  |  → /home/pi/relayEquipmentManager/node_modules/i2c-bus/lib/binding/node-v72-linux-arm/i2c.node
3|REM  |     at bindings (/home/pi/relayEquipmentManager/node_modules/bindings/bindings.js:126:9)
3|REM  |     at Object.<anonymous> (/home/pi/relayEquipmentManager/node_modules/i2c-bus/i2c-bus.js:4:32)
3|REM  |     at Module._compile (internal/modules/cjs/loader.js:936:30)
3|REM  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:947:10)
3|REM  |     at Module.load (internal/modules/cjs/loader.js:790:32)
3|REM  |     at Function.Module._load (internal/modules/cjs/loader.js:703:12)
3|REM  |     at Module.require (internal/modules/cjs/loader.js:830:19)
3|REM  |     at require (internal/modules/cjs/helpers.js:68:18)
3|REM  |     at new i2cController (/home/pi/relayEquipmentManager/dist/i2c-bus/I2cBus.js:20:35)
3|REM  |     at Object.<anonymous> (/home/pi/relayEquipmentManager/dist/i2c-bus/I2cBus.js:680:15)
3|REM  |     at Module._compile (internal/modules/cjs/loader.js:936:30)
3|REM  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:947:10)
3|REM  |     at Module.load (internal/modules/cjs/loader.js:790:32)
3|REM  |     at Function.Module._load (internal/modules/cjs/loader.js:703:12)
3|REM  |     at Module.require (internal/modules/cjs/loader.js:830:19)
3|REM  |     at require (internal/modules/cjs/helpers.js:68:18) {
3|REM  |   tries: [
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/build/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/build/Debug/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/build/Release/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/out/Debug/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/Debug/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/out/Release/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/Release/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/build/default/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/compiled/12.9.1/linux/arm/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/addon-build/release/install-root/i2c.node',
3|REM  |     '/home/pi/relayEquipmentManager/node_modules/i2c-bus/addon-build/debug/install-root/i2c.node',
[... 146 lines cut ...]
3|REM  | debug: Writing Pin #1:33 -> GPIO #13 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
3|REM  | info: [11:24:09] 192.168.2.183 PUT /state/device/gpio:1:36 {"isOn":0}
3|REM  | debug: Writing Pin #1:36 -> GPIO #16 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | info: [11:24:09] 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-13. Could not find pin in controller
3|REM  | debug: Persisting Configuration data... /home/pi/relayEquipmentManager/data/controllerConfig.json
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
3|REM  | info: [11:24:11] 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
3|REM  | debug: Writing Pin #1:29 -> GPIO #5 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
3|REM  | info: [11:24:11] 192.168.2.183 PUT /state/device/gpio:1:31 {"isOn":1,"latch":5000}
3|REM  | debug: Writing Pin #1:31 -> GPIO #6 to 1
[... cut ...]
3|REM  | debug: Writing Pin #1:36 -> GPIO #16 to 0
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | info: [11:24:13] 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-13. Could not find pin in controller
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}
3|REM  | info: [11:24:13] 192.168.2.183 PUT /state/device/gpio:1:29 {"isOn":0}

bsileo, 2021-07-02:
In the broken state I consistenly get these when I try to change circuits from njsPC:

3|REM  | info: [11:27:03] 192.168.2.183 PUT /state/device/gpio:1:15 {"isOn":true,"latch":10000}
3|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-15. Could not find pin in controller
3|REM  | verbose: 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | info: [11:27:03] 192.168.2.183 PUT /state/device/gpio:1:13 {"isOn":true,"latch":10000}
3|REM  | error: Could not set gpio state: setDeviceState: Error setting pin state: Error: Invalid Pin #1-13. Could not find pin in controller

rstrouse, 2021-07-02:
This appears to be related to one of the node dependencies.

What version of node are you running?

rstrouse, 2021-07-02:
I traced down into the node module that is failing after REM starts.  It appears that the pins that were disabled from the original problem need to be unexported.  I added some code to REM to do just that and ignore the error reported by the node_module.

bsileo, 2021-07-02:
> This appears to be related to one of the node dependencies.
> 
> What version of node are you running?

12.9.1

bsileo, 2021-07-02:
That looks like it fixed things. i will test a few more retstarts and then close

rstrouse, 2021-07-02:
Yeah I simply ignore the pin release errors for pins that are no longer active for REM.  The binding errors at the startup might be why you cannot get the multi-purpose pins to release for GP.

bsileo, 2021-07-02:
Alright all seems to be working now through a few reboots. Thanks for the help!

closed 2021-07-02 by bsileo