rstrouse/relayEquipmentManager #18: MQTT Feeds

truvec, 2021-07-16:
I have a standalone setup with pump and controller. I would like to create a schedule for the SWG. Could I setup a feature tied to REM that sent a Feed back to PC with the salt %. I don't understand the Javascript payload entry.  Are there any example feed settings?

Thanks

rstrouse, 2021-07-16:
Interesting.  I posted the picture below on gitter.
[image]

I have been thinking that the schedules in njsPC could be used to manipulate equipment items outside of circuits/features.  Although the chlorinator was not one of the things that I have given any thought about given the goofiness of how these things work with constant current power.  Depending on your chlorinator model there is a goofy calculation that has the cell on or off it is not changing the power.  For instance, a setting of 25% indicates that the cell will be on 25% of the time over a 15 minute period.  It does not mean that it will increase or decrease the power on the cell to the percentage you supply.

truvec, 2021-07-16:
This works great. Thanks

closed 2021-07-16 by truvec