rstrouse/relayEquipmentManager #19: Intellibrite or Microbrite Lights not changing

DrJeff, 2021-08-06:
The relay is not switching on|off for color changes, it will change on|off for power mode. The actual lights are the MicroBrite lights maybe thats the problem?
[image]

rstrouse, 2021-08-07:
How are you triggering the relay.  Is this through njsPC where the circuit is declared as IntelliBrite or are you doing something more low level?

DrJeff, 2021-08-08:
It works low level just not through njsPC, I tried intellibrite and globrite but neither worked.

rstrouse, 2021-08-08:
Go ahead and pull njsPC.  It looks like we were jumping ahead in the code and not processing the event properly.

DrJeff, 2021-08-08:
Ok its working but the number of switching is off by one from the on position to change of color and the off position to color selection. So for RED (10) I get WHITE(10+1) 
[image]

DrJeff, 2021-08-09:
Ok I found the problem and edited the code minus 1 and removed my delay in the relay's config! Nice :)
[image]

DrJeff, 2021-08-09:
How is this controlled via MQTT to change value of lightTheme?

rstrouse, 2021-08-09:
Yeah that is not quite right.  The sequence number should match the number of cycles in the docs.  I'll have a look but judging on your results I'll bet we are shutting it off first or are not waiting long enough after the initial turn on to start firing off sequences.

To set the theme with MQTT set a topic `<root>/circuit/{circuitId}/setTheme` with the payload
```
{
  id: <id of the circuit>
  theme: [0-11 id from the list above]
}
```

At some point I will go in and allow you to sent the theme name or the id.

DrJeff, 2021-08-09:
I'll wait for the proper fix, but hacking together what I have I'm good at that. ;) The theme selection is perfect just wasn't sure what to send.

Thanks

rstrouse, 2021-08-15:
I am trying to figure out how this is not correct.  According to the docs the number of times it is transitioned from on to off is what makes the difference.

Do you get different results if the lights start from the on position vs the off position?

DrJeff, 2021-08-15:
No the same results either way. I did have some delays on the pin/relay but I removed all those 0ms for on and off same results. For some strange reason the code I clipped above works flawlessly. Why I have no clue 

As a side note the setting of theme via mqtt isn't working either?
```
virtual-controller/circuit/2/setTheme

{"id":2,"theme":1}
```
I also tried using ``` virtual-controller/circuits/2/setTheme ``` not sure

rstrouse, 2021-08-15:
Funny I just watched a video where the guy started with off not on.  If you read the docs it explicitly says to turn it on then start the sequence and end with on.  The docs are wrong.

I posted up a fix to njsPC that operates the lights just like the guy in the video.  This should take care of the issue since the count is the number of on/off sequences not off/on sequences.  No matter how you read it the documentation is wrong according to your testing and the video I just watched.
[image]

DrJeff, 2021-08-15:
Ok I thought you were asking for the On|Off inside of REM the solution I posted worked if it started on or off.

I looked at my Node Red and I did turn it Off first
[image]
So let me pull and test now...
Ok if the circuit is OFF and you chose a color from the color wheel it will turn On the circuit but the color will not be correct it is -1. So this works correctly  from the light being ON first only.

DrJeff, 2021-08-15:
But it does have to start on first!  I stand corrected look up at my flow I make sure its off first then turn it ON then start the sequences of OFF|ON 
https://www.youtube.com/watch?v=dugotHIW0tA 

then just the ONs count

rstrouse, 2021-08-15:
Yeah that is the video I watched.  It should work as expected if the video is correct and the documentation is not.  I rewound that thing on each of theme selections to make sure I counted the number of times it hit the on and off position.  It still switches at 100ms per cycle so let me know if it hits them appropriately.  If it is too fast then add some time in REM until it is reliable.  If you get it too long then you are needlessly waiting in the dark.  IntelliTouch was really slow but IntelliCenter is much faster.

rstrouse, 2021-08-28:
Is this now working for you.

rstrouse, 2021-08-29:
I am going to assume that this is all working for you so I am closing the issue.

closed 2021-08-29 by rstrouse

DrJeff, 2021-09-06:
Can you code this so that if the status of the light is ON then turn OFF the light then set color? currently if the light is ON this will not change properly.