rstrouse/relayEquipmentManager #21: Issue setting up valves

Exit2Studios, 2021-08-30:
I have njspc and rem working on an Rpi4 combined with a Sequent Microsystems BAS and 8-relay IND board..  I'm using a single standard valve for testing and have it assigned as my intake valve on a shared body nixie controller.  However, when I switch from pool/spa the valve does not activate.  I can control it via the Sequent Microsystems command line prompt.

rstrouse, 2021-08-30:
So it sounds like you have the valve wired properly.  Shoot me a screenshot of the I2c page.  It should looks something like this when you select the Multi-Relay Board.
[image]

Exit2Studios, 2021-08-30:
Here are the screenshots:

rstrouse, 2021-08-30:
The screenshots didn't come through

Exit2Studios, 2021-08-30:
[image]

rstrouse, 2021-08-30:
Ok so delete that trigger.  You do not want that trigger set up for a Nixie controller.  The connection is more direct that when using Nixie and it does some work to make sure relays reflect the desired state of the pool.  This is done through relay latching from njsPC.  To configure an Intake valve simply navigate to the valves tab in dashPanel and set it up like you have above.
[image]

Show me a screenshot just like the first one I posted.

rstrouse, 2021-08-30:
Ok so I just now saw the screenshot.  You do not have the correct address for the Relay board.  It is not at address 3 (0x03) it is at address 63 (0x3F) if you did not change any of the jumpers on the card.

rstrouse, 2021-08-30:
Delete the Multi-Relay Board from REM and if it does not automatically detect the relay, then hit the + and use 63 as the address.  You can hit the Re-scan button to have it scan for addresses but it is possible that the address might not show up on the scan.  Especially if there is other software hitting the device.  Such is the nature of I2c.

Exit2Studios, 2021-08-30:
That seems to have done the trick.  I knew it had to be user error.  Thank you!

rstrouse, 2021-08-30:
Yeah for the world of me I couldn't see a path where it wouldn't be working.  Sometimes a picture is worth a thousand words.

closed 2021-08-30 by rstrouse