rstrouse/relayEquipmentManager #22: Lost i2c again

DrJeff, 2021-09-25:
Not sure of why I lost comms to 12c so I deleted it and re added but it will not see the i2c
```
1|REM  | verbose: 127.0.0.1 PUT /state/device/i2c:1:6:9 {"isOn":true,"latch":10000}
1|REM  | info: [10:58:31 AM] 127.0.0.1 PUT /state/device/i2c:1:6:9 {"isOn":true,"latch":10000}
1|REM  | error: setDeviceState: i2c bus not found 1 - i2c:1:6:9
1|REM  | verbose: 127.0.0.1 PUT /state/device/i2c:1:6:4 {"isOn":false}
1|REM  | info: [10:58:31 AM] 127.0.0.1 PUT /state/device/i2c:1:6:4 {"isOn":false}
1|REM  | error: setDeviceState: i2c bus not found 1 - i2c:1:6:4
```
The pi sees it though 
```
pi@pool:~/rem $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- 26 27 -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- 62 63 -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

rstrouse, 2021-09-25:
There are a couple of things going on here.  First if this is from a period when the server is starting up, the Raspberry Pi can be tardy responding to the requests as it starts up.  This is normal because it does some wacky stuff as it puts the fssys stuff together.  It was a real head scratcher.  REM is persistent though and will eventually pick it up.  So if the messages are during startup you should ignore them until the pi has stabilized.  Also, turn off Bluetooth and Serial Console as these delay the whole thing if you have them on.

i2cdetect does some broadcom stuff when it fails fssys and while you can detect the device address it will not communicate with it using a buffer.  So it is possible to get a detection of addresses that cannot be communicated with.

As for backups.  Under the burger menu in dashPanel there is now a Backups tab that will auto backup the REM server on an interval.  This backs up to the njsPC backups directory and will keep as many backups as you have selected.  The restore function while it works for njsPC it should not yet be trusted until we have had some time to test it in every environment.  It also does not restore REM servers yet.  The settings below will keep nearly 4 months of backups.  If you manually back up, these will stick around until you delete them from the restore option.
[image]

And now a note on SD cards.  Let me start off by saying I hate them.  First you should never use an SD card that has a slow write speed.  U3 or better.   I also tend to buy the larger ones and 32Gb seems to be the minimum (for me 128Gb EVO Select).  The higher endurance cards seem to use some trim to make sure erroneous writes due to power fade get trimmed from the fs.

DrJeff, 2021-09-25:
I just rebuilt REM from scratch I had some stragglers that I didn't like anyways and it is working now.
I have ran Pi s' on ssd but did't want to for the pool equipment.
I really like the backup option from the menu 👍🏽

DrJeff, 2021-09-27:
@rstrouse Need your help again...
Ok, I tried re installing or doing the REM setup but my board isn't acting right. Errors and no control to the relays here are the logs from REM and the Capture data from njsPC
[consoleLog(2021-09-26_22-09-43).log.zip](https://github.com/rstrouse/relayEquipmentManager/files/7233557/consoleLog.2021-09-26_22-09-43.log.zip)
[replay.zip](https://github.com/rstrouse/relayEquipmentManager/files/7233558/replay.zip)

rstrouse, 2021-09-27:
Ok so all the devices appear to read normally on the I2c bus with the exception of one or more relays.  I2c isn't failing as much as the relays are not communicating anymore.  REM is being persistent and asks repeatedly for response.

So this leads me to believe that there is one of 3 things going on here.
1, You have a wiring issue where the two wire comms (SDA/SCL) to the relay controller is having issues.  This can be due to poor connections or the wire length.  A remedy sometimes is to install an active terminator such as LCT4311.
2. The relay controller is attempting to use clock stretching.   Raspberry Pi cannot do this in hardware and you must either use an i2C bus overlay or reduce the clock speed of the hardware bus.
3. The relay controller is failing.  Remote I/O error is an error that occurs when the endpoint is no longer there.  This is not like suddenly REM decided to stop talking to the device.  Communication is still going on.  The relay and more importantly the i2c controller on the relay decided not to respond.

DrJeff, 2021-09-27:
> Ok so all the devices appear to read normally on the I2c bus with the exception of one or more relays. I2c isn't failing as much as the relays are not communicating anymore. REM is being persistent and asks repeatedly for response.

Yes you are correct like always. It was a loose wire on the I2c side, guess I need to clean that part up! 
Thanks again! :grin

closed 2021-09-27 by DrJeff

rstrouse, 2021-09-27:
While I am often wrong I am rarely incorrect.  Sometimes though I am just mistaken.  Glad you got it working.