rstrouse/relayEquipmentManager #23: Under Voltage?

DrJeff, 2021-10-04:
I think I have an issue with under voltage?  My pi is running via POE and it is serving the i2C buses power. When I get problems reading the PH and ORP also randomly restarts on the i2c bus. Before I switch my power source what do you think?
[replayUnderVoltage.zip](https://github.com/rstrouse/relayEquipmentManager/files/7274836/replayUnderVoltage.zip)

I did try powering the i2C bus the other way from 5V source and only connecting the Ground with SDA SCL to the pi but was not working. Unless I fried something? I don't think so because it still works and I didn't see any magic smoke.

DrJeff, 2021-10-04:
Ok I added i2c on bus 4 using `dtoverlay=i2c-gpio,bus=4,i2c_gpio_delay_us=1,i2c_gpio_sda=23,i2c_gpio_scl=24` 

It works and the chips are seen but when I add them to REM they get added then not active and when checking the box it says they are already added but can't access.
[image]
[image]
[image]

 It looks like the bus 4 can't be identified by REM??

Somehow I nuke my install njsPC is blank as well

tagyoureit, 2021-10-04:
> Ok I added i2c on bus 4 ...

Wait a minute... the Pi only has 2 busses.  How are you adding a first and a fourth bus?

DrJeff, 2021-10-04:
I read somewhere not to use the 0-2 so I used the above is there a better way, I just been having issues with those three devices on 1 with the tentacle board different addresses but still

DrJeff, 2021-10-04:
from the above
>Note on the Bus Numbering and order:

Never use bus 0 and 2, it is use for other things in the board like eprom on hats etc

For the April 2019 raspbian release:

You should always start with the highest bus (Bus 4 in this case) in your config.txt and work through to the lowest bus (bus 3).

The lowest bus must always be bus 3

If you need 5 extra busses, the busses must bi in the order of 7,6, 5, 4, 3

This issue on bus order was not there when this Instructable was originally written. It seems like changes was made to the kernel.

tagyoureit, 2021-10-04:
I'm reading that instructables like you should only create additional busses (using GPIO ports) if you have conflicts.  If you don't have i2c peripherals with conflicting addresses, you are just making your life hard.  

Do you have 2 multi-relay boards both with address 38?  If so, which board(s) is it?  There could be something with the code that isn't looking at unique busses (rstrouse or I will check it out), but most boards you can configure the address.

tagyoureit, 2021-10-04:
I have a similar issue... haven't dug into it by my Sequent Microsystems 4 relay hat keeps disappearing from REM, but I can see it on the i2cdetect.  It's on my list to investigate.

DrJeff, 2021-10-04:
crazy thing before was working fine ... something crept in. My Atlas sensors 98 and 99 work fine but when I add Relay boards 26, 27 or ADS 48 they will flip flop missing and there

DrJeff, 2021-10-05:
Ok maybe I found the answer to my woes

I have been running my i2C bus with 5V for the Input Voltage and it's been working but after cutting cables trying other i2C to no avail I wired the VDD on the i2C to 3V3 and it is all working again!

tagyoureit, 2021-10-05:
Oh, yes, incorrect voltage can really be shocking in what it interferes with.  Great to hear you have found the issue!  I'll close this for now but re-open if you find anything else that should be current knowledge.  I'll put a resistor on any more bad puns...

closed 2021-10-05 by tagyoureit

DrJeff, 2021-10-05:
I thought some thing was wrong with my existing relay board or MCP23017, I'm getting flickering or better yet random on off of a few relays. So I added another MCP23017 and another 16 relay board on the bench and it's doing the same thing especially with relay 1 and 13. Attached is my replay it happens for reference but it happens about every 5 mins. or so.

rstrouse, 2021-10-05:
Ok so open a new issue.   I just happened across this and I will summarize some of what I am reading for you.  If the flickering is occurring during the startup, this is because the some of these 16 relay boards use a scheme that holds power for the off position of the relay.   The problem with that is they typically get power to the board before startup finishes so you have no sig but have power and the relay goes closed until the I2c interface chip finishes booting up.  Normally open is what happens when you have power and it is closed when you do not.   The theory is that it doesn't require as much power to hold the relay open.  I guess there may be benefit with this depending on your application but it isn't the case here.

If the flickering is occasional it could be that the 5v for the coil voltage may need to be supplemented with another power source.  I am not sure what you are powering the pi with but if you a tapping 2 16 relay boards off the 5v on the pi it may not be enough.

You do have an option to install a pull-up and switch the bias of the relay board on some of them.  So normally open is open when there is no power to the board.

VCC for I2c on the Raspberry Pi is 3v3 I am not sure how much it can tolerate a 5v VCC.  VCC on the other side of an isoloated circuit can be whatever you want because this is done with an isolation circuit.  For instance, the Atlas RTD is non-isolated and the PH is isolated.  Note the differences in VCC voltage reported by the controller.
[image]
[image]

Yes you can assign GPIO to another bus but the SDA/SCL lines need to be isolated and bus 0 doesn't exist in the overlay.  Bus 2 can be used but it is often occupied by one of the serial interfaces.  It might be part of the bluetooth stuff.   However, the thing you likely read was that addresses (not bus numbers) 0x00 to 0x02 are typically reserved to make an interrupt that does not actually exist on the pi.  This is related to eeprom devices.  Rumors are that the pi5 is going to have hardware interrupts of some fashion.  Say it isn't so!