rstrouse/relayEquipmentManager #25: REM won't start after adding 1W device

jslabotsky, 2021-11-30:
After creating a 1-wire temperature sensor, REM won't launch successfully. Log attached.
[REM-error.log](https://github.com/rstrouse/relayEquipmentManager/files/7626623/REM-error.log)

rstrouse, 2021-11-30:
Please upload your controllerConfig.json file from the REM data directory.  You will have to rename it to controllerConfig.txt to get it to upload.

jslabotsky, 2021-11-30:
Here you go.
[controllerConfig.txt](https://github.com/rstrouse/relayEquipmentManager/files/7628004/controllerConfig.txt)

jslabotsky, 2021-11-30:
Since I now have a backup copy of controllerConfig.json, I manually removed the configs for my devices from the file and restarted REM. Now I can get into the web page. I was able to add the 1-wire bus and the correct directory is shown. However, the 1-wire bus does not detect my temp sensor. The ID is 28-000000040545. At the command line, if I go to the device directory and cat w1_slave I get the expected output, and the raw temperature changes in response to what I'm doing to the probe.  If I try to manually add the device in REM it says bus 1 is not initialized. Did I skip something in the configuration?

rstrouse, 2021-11-30:
The issue was a generic device feed that REM did not remove when you deleted the generic temp sensor attached to the ADS1x.

Try resetting the server by pressing the Reset Server button on the general page.

jslabotsky, 2021-11-30:
I've done that. No love.

jslabotsky, 2021-11-30:
I rebooted the Pi again and now the 1-wire sensor is detected. We can close this. Thanks as always for the help. It is greatly appreciated.

rstrouse, 2021-11-30:
Thanks I will add some code to remove latent triggers and feeds when a device is deleted.  This all stemmed from deleting a Generic Temp Probe that was part of an internal feed.  For external feeds this wouldn't be a factor because it isn't out of the realm of possibility that an MQTT server or API service was down,  The code for Internal Devices however assume that the references are always valid.

When the Generic Temp Sensor went away it should destroy the internal feed from the ADS1x device.

closed 2021-11-30 by jslabotsky

jslabotsky, 2021-12-01:
Thanks for all the help with my dumb questions on the temperature probes. Does REM currently support 1-wire pressure transducers? Or any desire to do that in the future?

tagyoureit, 2021-12-02:
The only 1-wire device currently programmed is the DS18B20. We could certainly add one, but since I don’t have one it might be a little bit of back-and-forth to have your help to make sure that it works correctly. If you want to open up a new issue for that we can work on it.