rstrouse/relayEquipmentManager #26: FEATURE REQUEST: Add ability to set GPIO startup pin states

johnny2678, 2021-12-01:
[image]

Not a high priority issue.  Sometimes my 1-wire connection needs to be reset so rather than using the dedicated 3.3v pin I use GPIO, which supplies 3.3v when on.  That way I can toggle through code, REM, etc.

Only problem is when the REM pi reboots the GPIO PIN I've assigned starts in the off state and I don't get any temp readings from my ds18b20's.  Would be helpful for forgetful people (like me) to have to option to default the GPIO state so I don't lose data if I forget to flip the pin state after a reboot/power cycle.

low priority - thanks!

rstrouse, 2021-12-01:
Hmmm.  I'll see what the pin export is doing.  There may be a small delay to this as the pin is constructed on startup but I am thinking on, off, and last state.  There is some goofiness with pulling the pin high on startup if it is inverted but it may be a logic issue on my end.  It is a double negative.

rstrouse, 2021-12-01:
Well I looked into setting these states.  Are you aware that the default state of the pin is the last state.  So if the pin was high when REM was shut down it initializes the pin high (on).  Is something shutting this output off before REM is shut down?

rstrouse, 2021-12-01:
Ok so now you should be able to select On, Off, or Last State as the initial startup state for the pin.  Go ahead and pull REM.

johnny2678, 2021-12-01:
> Is something shutting this output off before REM is shut down?

hmm... I don't think so.  Don't recall doing anything on the pi that would cause the pin to change state but it's been in service for almost 4 years now.  Who knows what old logic I have left on there. 😂🤷‍♂️

> Ok so now you should be able to select On, Off, or Last State as the initial startup state for the pin. Go ahead and pull REM.

Great, will check this out ASAP and post back.

johnny2678, 2021-12-02:
works!

[image]

confirmed with a restart and pin was still hot.  Thanks!

closed 2021-12-02 by johnny2678