rstrouse/relayEquipmentManager #27: Relays not actuated and cannot find device definition after restart

Fumado, 2021-12-19:
Environment: Raspberry Pi 4, Ubuntu server 21.10, REM 7.5

Equipment: EP-0099 Multi-Relay board by 52pi.  "Using Docker Pi 4" as the relay controller.

Steps to reproduce:
Install EP-0099, verify that it's being seen by the I2C bus at 0x10 with 'i2cdetect -y 1'
Verify that any of the relays turn on with 'i2cset -y 1 0x10 0x01 0x01'
Add the board as a multi-relay board device on the I2C bus at address 16 (0x10).  Choose 'Docker Pi 4' as the controller. Save the device.
Configure one of the relays, and enable and save then save the device  
You can also enable the device, but it makes no difference.
When you click on the relay the green virtual LED turns on, but the relay doesn't actuate.

If you restart REM, and navigate to that device and click the relay you enabled to see if the reboot makes the relay actuate, you get the error: Bus #1 Address 16 cannot be found.

If you configure a second relay and save the device, you no longer get the error on any of the relays, and the green virtual LED lights, but the relays still do not actuate.

Attached controllerConfig.json
[controllerConfig.json.txt](https://github.com/rstrouse/relayEquipmentManager/files/7741641/controllerConfig.json.txt)

rstrouse, 2021-12-20:
It appears that the i2c bus is not opening.  So lets do some steps to figure out while.

1. First shut down REM.  If you are using pm2 then type `pm2 stop REM`
2. Go to the directory where REM is installed `cd ~/relayEquipmentManager`
3. Ensure the supporting libraries are installed properly `npm i`
4. Start REM and watch the logs.  We are looking for any errors spit out during initialization.  There will be a log entry that says `Initializing i2c Bus #1` then I suspect we will see an error.

Fumado, 2021-12-20:
I turned on logging before I shutdown REM, and it listed a bunch of 'permission denied' on /dev/i2c-1.  The permissions were cleared on world, so I did a chmod '666' and once I restarted REM it initialized the I2c bus and found the multi-relay board, and I was able to operate the relay.

Sorry for opening an issue which was user error.

rstrouse, 2021-12-20:
No worries glad we got you sorted.

closed 2021-12-20 by rstrouse