rstrouse/relayEquipmentManager #29: duplicate entries for Atlas PH/ORP feeds

johnny2678, 2022-02-09:
[image]

I cleared out a bunch (10+) of dupes a week ago and they are starting to creep back in.  Wish I could provide more info on how to repeat but not sure when they show up.  Guessing on app restart?

Let me know what other info I can provide.

renamed to .txt
[controllerConfig.txt](https://github.com/rstrouse/relayEquipmentManager/files/8029060/controllerConfig.txt)

rstrouse, 2022-02-13:
Can I get a copy of your poolConfig.json from njsPC.  It appears that there is an invalid feed id in there.

johnny2678, 2022-02-13:
sure thing.  Let me know if I can provide anything else.
[poolConfig.txt](https://github.com/rstrouse/relayEquipmentManager/files/8056113/poolConfig.txt)

rstrouse, 2022-02-15:
Go ahead and pull njsPC and REM.  You will need to clean up your duplicates but from then on it should not re-create them.   

Part of the issue is that there were some invalid identifiers for the feeds that would fool njsPC.  When REM starts it will remove feeds with invalid ids but it really shouldn't be cleaning up the duplicates.  You will need to go into REM and delete the duplicate feeds.

johnny2678, 2022-02-15:
Sure thing. Thanks for looking into this.

rstrouse, 2022-03-31:
If you are still receiving duplicates after the fix open another issue.

closed 2022-03-31 by rstrouse