rstrouse/relayEquipmentManager #30: Cannot feed 10k BAS Temp Probe to Generic Device

ddyer8, 2022-04-01:
Copy-paste from Gitter discussion:

I'm in the process of initial configuration of a DIY-Standalone-Nixie-Pool-Controller with REM, and I cannot get the 10k BAS Temp Probe to feed the Generic Device. I have nodejs-poolController, dash Panel, and REM up and running on a pi4 with Sequent MEGA-BAS. I have 10k temp probe hooked up to AI 1. I have followed the instructions on the wiki a few times (https://github.com/rstrouse/relayEquipmentManager/wiki/Sequent-Microsystems-Boards#sequent-building-automation-hat-mega-bas) and the generic device will not retain the settings to be fed by the BAS. Usually, the Generic Device configuration screen lacks the Save and Delete Device buttons. Have you seen this behavior before? Recommended troubleshooting steps?

[image]

rstrouse, 2022-04-01:
Change the send value on the MEGA-BAS feed from the `0-10v in #1` to `0-10v in #1 value`.  You are currently feeding the entire object reference to the generic device.   The temperature probe device only knows how to process the raw numeric value from the input.
[image]

ddyer8, 2022-04-01:
Shoot.  I see now the instructions clearly say to pick the one with "value." Works fine for me now.

Any way for the Generic Object to generate a warning or error when fed the object and expecting a value?  If so, perhaps this would be a good enhancement? Either way, I'm fine closing the issue.

rstrouse, 2022-04-01:
I just need to do some work on the UI but I have other priorities right now.

closed 2023-03-12 by rstrouse