rstrouse/relayEquipmentManager #31: error with ADS1115

magimat, 2022-05-06:
Hi,

I'm able to configure my ADS1115 and it gets detected, but as soon as I try to enable a channel on it, REM crashes and I get this error:

error: Cannot read properties of undefined (reading 'pgaMask')

/home/pi/relayEquipmentManager/i2c-bus/ads1x15.ts:392
                c.pgaMask = this.pga.get(c.pga).pgaMask;
                                               ^
TypeError: Cannot read properties of undefined (reading 'pgaMask')
    at ads1x15.setOptions (/home/pi/relayEquipmentManager/i2c-bus/ads1x15.ts:392:48)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at I2cBus.setDevice (/home/pi/relayEquipmentManager/boards/Controller.ts:2493:61)
    at I2cController.setDevice (/home/pi/relayEquipmentManager/boards/Controller.ts:2100:26)
    at /home/pi/relayEquipmentManager/web/services/Config.ts:262:27

any idea?

rstrouse, 2022-05-26:
I can't believe I missed this.  Sometimes git misses a notify.

When you enable the channel are you selecting a Power Gain?  You also have to save the device before this takes effect.  This has to do with some weirdness related to setup registers for ADS.  So configure the channel then press the Save Device button.

rstrouse, 2023-03-12:
Closing this since I assume you got the PGA set correctly now.

closed 2023-03-12 by rstrouse