rstrouse/relayEquipmentManager #34: REM stopped responding upon startup

spifftek70, 2022-08-21:
I was configuring the REM per directions when it stopped responding.  Now when I start or restart REM, I get an `Error: General Error` and nothing responds.

```
TypeError: Cannot read property 'type' of undefined
    at Controller.getDeviceByBinding (/home/pi/relayEquipmentManager/boards/Controller.ts:553:22)
    at DeviceFeed.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:903:32)
    at I2cDevice.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:2796:57)
    at I2cBus.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:2346:59)
    at I2cBusCollection.toExtendedArray (/home/pi/relayEquipmentManager/boards/Controller.ts:205:85)
    at I2cController.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:1990:30)
    at Controller.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:529:26)
    at /home/pi/relayEquipmentManager/web/services/Config.ts:129:34
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:341:12)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/relayEquipmentManager/web/Server.ts:281:17
```

tagyoureit, 2022-08-21:
You can upload your `./data/controllerConfig.json` here and we can take a look.

spifftek70, 2022-08-21:
'''
{
  "controllerType": "raspi",
  "gpio": {
    "pins": [],
    "exported": []
  },
  "spi0": {
    "isActive": false,
    "busNumber": 0,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "spi1": {
    "isActive": false,
    "busNumber": 1,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "i2c": {
    "isActive": false,
    "buses": [
      {
        "id": 1,
        "isActive": true,
        "devices": [
          {
            "id": 1,
            "values": {
              "inputs": {
                "in0_10": [
                  {
                    "id": 1,
                    "name": "Solar",
                    "type": "T10k",
                    "enabled": true,
                    "units": "kohm",
                    "ioType": "analog",
                    "value": 8.947
                  },
                  {
                    "id": 2,
                    "name": "Pool",
                    "type": "T10k",
                    "enabled": true,
                    "units": "kohm",
                    "ioType": "analog",
                    "value": 9.312
                  },
                  {
                    "id": 3,
                    "name": "Air",
                    "type": "T10k",
                    "enabled": true,
                    "units": "kohm",
                    "ioType": "analog",
                    "value": 8.999
                  },
                  {
                    "id": 4,
                    "name": "Spa",
                    "type": "T10k",
                    "enabled": true,
                    "units": "kohm",
                    "ioType": "analog",
                    "value": 8.191
                  },
                  {
[... 474 lines cut ...]
    ]
  },
  "oneWire": {
    "isActive": false,
    "buses": [
      {
        "id": 1,
        "isActive": true,
        "devices": [],
        "busNumber": 1,
        "addresses": []
      }
    ],
    "detected": [
      {
        "path": "/sys/devices/w1_bus_master1",
        "name": "w1_bus_master1",
        "busNumber": 1
      }
    ]
  },
  "lastUpdated": "8/21/2022, 07:22:14",
  "configVersion": 1
}
```

spifftek70, 2022-08-21:
2nd Error:

`TypeError: Cannot read property 'type' of undefined
    at Controller.getDeviceByBinding (/home/pi/relayEquipmentManager/boards/Controller.ts:553:22)
    at DeviceFeed.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:903:32)
    at I2cDevice.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:2796:57)
    at I2cBus.getExtended (/home/pi/relayEquipmentManager/boards/Controller.ts:2346:59)
    at I2cBusCollection.toExtendedArray (/home/pi/relayEquipmentManager/boards/Controller.ts:205:85)
    at /home/pi/relayEquipmentManager/web/services/Config.ts:204:39
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/pi/relayEquipmentManager/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at /home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:341:12)
    at next (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:275:10)
    at /home/pi/relayEquipmentManager/web/Server.ts:281:17
    at Layer.handle [as handle_request] (/home/pi/relayEquipmentManager/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/pi/relayEquipmentManager/node_modules/express/lib/router/index.js:323:13)`

tagyoureit, 2022-08-21:
Rename the attached to end with .json.  You had a feed in the Mega-Bas that was not complete.  This was likely caused by your feeds on the Solar (generic device).  One of them was feeding back to itself creating a loop.  The other feeds on that temp sensor went to other temp sensors which also doesn't make sense.  You only need to feed from the Mega-Bas -> Temp sensor (and then from the Temp sensor to njsPC assuming you want the temps to appear there).

[controllerConfig.txt](https://github.com/rstrouse/relayEquipmentManager/files/9389485/controllerConfig.txt)

spifftek70, 2022-08-21:
Thanks...  Now in the terminal for REM it keeps scrolling:
```
info: Starting up SSDP server
info: Initializing GPIO Pins 0
info: Initializing SPI Bus #0
info: SPI Bus #0 Initialized
info: Initializing SPI Bus #1
info: SPI Bus #1 Initialized
info: Initializing i2c Interface
info: Detecting i2c Buses
info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
info: Detecting i2c Buses /sys/class/i2c-dev/i2c-20
info: Detecting i2c Buses /sys/class/i2c-dev/i2c-21
info: Initializing 1-Wire Interface
info: Detecting 1-Wire Buses
info: oneWireInterface: w1_gpio                16384  0
wire                   36864  1 w1_gpio

info: Detecting 1-Wire Buses at /sys/bus/w1/devices
info: Initializing Generic Devices
info: Initializing i2c Bus #1
info: Initializing 1-Wire Bus #1
info: Scanning 1-Wire Bus #1
info: Device 10k Temperature Probe initialized for generic device 1 - Solar
info: Device 10k Temperature Probe initialized for generic device 2 - Pool
info: Device 10k Temperature Probe initialized for generic device 3 - Air
info: Device 10k Temperature Probe initialized for generic device 4 - Spa
info: Generic Devices Initialized
info: Server is now listening on 0.0.0.0:8080
info: Configuration file /home/pi/relayEquipmentManager/config.json changed.
info: Reloading app config: config.json
error: Error connecting to Pool Controller http://autopool.lan:4200: xhr poll error
info: Scanning i2c Bus #1
[ '00-800000000000', '00-400000000000' ]
info: 1-Wire Bus #1 Initialized
[ 72 ]
info: Found I2C device Sequent MEGA-BAS at address: 72 - (0x48)
info: Adding I2C device that could not be scanned 3 - (0x3)
info: Adding I2C device that could not be scanned 48 - (0x30)
info: Device Sequent MEGA-BAS initialized for i2c bus #1 address 72
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 1,0: EREMOTEIO: remote I/O error, write
info: Device Multi-Relay Board initialized for i2c bus #1 address 3
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: Sequent Watchdog error getting firmware version: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: Sequent Watchdog error getting period: , Remote I/O error
error: 3 1,0: EREMOTEIO: remote I/O error, write
error: Sequent Watchdog error getting defaultPeriod: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: Sequent Watchdog error getting off interval: , Remote I/O error
error: Sequent Watchdog error setting reload: , Remote I/O error
error: Sequent Watchdog error getting fan temperature: , Remote I/O error
error: Sequent Watchdog error getting input source voltage: , Remote I/O error
error: Sequent Watchdog error getting output/Raspberry Pi voltage: , Remote I/O error
error: Sequent Watchdog error getting battery voltage: , Remote I/O error
error: Sequent Watchdog error getting charge status: , Remote I/O error
error: Sequent Watchdog error getting reset count: , Remote I/O error
info: i2c Bus #1 Initialized
```

tagyoureit, 2022-08-21:
First try a reboot.

spifftek70, 2022-08-21:
From `nodeJS-poolController`:

`[8/21/2022, 16:37:43] error: Error sending setDriveState for Pool Pump : Comms port 0 is not open. Message aborted: 165,0,96,33,6,1,10,1,55 
[8/21/2022, 16:37:43] warn: Comms port 0 is not open. Message aborted: 165,0,96,33,10,4,2,196,8,152,2,154 
[8/21/2022, 16:37:43] warn: Message aborted after 0 attempt(s): 165,0,96,33,10,4,2,196,8,152,2,154 
[8/21/2022, 16:37:43] error: Error sending setPumpRPMAsync for Pool Pump: Comms port 0 is not open. Message aborted: 165,0,96,33,10,4,2,196,8,152,2,154 
[8/21/2022, 16:37:44] warn: Comms port 0 is not open. Message aborted: 165,0,96,33,7,0,1,45 
[8/21/2022, 16:37:44] warn: Message aborted after 0 attempt(s): 165,0,96,33,7,0,1,45 
[8/21/2022, 16:37:44] error: Error sending requestPumpStatus for Pool Pump: Comms port 0 is not open. Message aborted: 165,0,96,33,7,0,1,45 
[8/21/2022, 16:37:44] warn: Comms port 0 is not open. Message aborted: 165,0,96,33,4,1,255,2,42 
[8/21/2022, 16:37:44] warn: Message aborted after 0 attempt(s): 165,0,96,33,4,1,255,2,42 
[8/21/2022, 16:37:44] error: Error sending setPumpToRemoteControl for Pool Pump: Comms port 0 is not open. Message aborted: 165,0,96,33,4,1,255,2,42 
[8/21/2022, 16:37:46] warn: Comms port 0 is not open. Message aborted: 165,0,96,33,6,1,10,1,55 
[8/21/2022, 16:37:46] warn: Message aborted after 0 attempt(s): 165,0,96,33,6,1,10,1,55 
[8/21/2022, 16:37:46] error: Error sending setDriveState for Pool Pump : Comms port 0 is not open. Message aborted: 165,0,96,33,6,1,10,1,55 
[8/21/2022, 16:37:46] warn: Comms port 0 is not open. Message aborted: 165,0,96,33,10,4,2,196,8,152,2,154 
[8/21/2022, 16:37:46] warn: Message aborted after 0 attempt(s): 165,0,96,33,10,4,2,196,8,152,2,154 
[8/21/2022, 16:37:46] error: Error sending setPumpRPMAsync for Pool Pump: Comms port 0 is not open. Message aborted: 165,0,96,33,10,4,2,196,8,152,2,15`

and from `relayEquipmentManager`

`error: Sequent Watchdog error getting battery voltage: , Remote I/O error
error: Sequent Watchdog error getting charge status: , Remote I/O error
error: Sequent Watchdog error getting reset count: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: 3 1,0: EREMOTEIO: remote I/O error, write
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: 3 1,0: EREMOTEIO: remote I/O error, write
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: 3 1,0: EREMOTEIO: remote I/O error, write
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: 3 1,0: EREMOTEIO: remote I/O error, write
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: 3 1,0: EREMOTEIO: remote I/O error, write
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: 3 3,0: EREMOTEIO: remote I/O error, write
error: 3 1,0: EREMOTEIO: remote I/O error, write
error: 3 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
error: Sequent Watchdog error setting reload: , Remote I/O error
error: Sequent Watchdog error getting fan temperature: , Remote I/O error
error: Sequent Watchdog error getting input source voltage: , Remote I/O error
error: Sequent Watchdog error getting output/Raspberry Pi voltage: , Remote I/O `

tagyoureit, 2022-08-22:
You have 2 completely separate things going on here:
1) njsPC - `Comms port 0 is not open`.  What RS485 port do you have?  You can check out [this wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details).

2) REM - I can't tell much from this log snippet other than REM isn't communicating with your equipment.  Did you try a hard power cycle?  Sometimes that will kick the i2c bus back into action.  What is the result of `i2cdetect -y 1`?  You can check [this wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/DIY-Standalone-Nixie-Pool-Controller#install-relayequipmentmanager) for some tips.

spifftek70, 2022-08-22:
I'm using the `Stack the Sequent Microsystems Building Automation HAT` for RS485.  I've pretty much modeled my entire system from the first diagram on the DIY Standalone [Nixie Pool Controller](https://github.com/tagyoureit/nodejs-poolController/wiki/DIY-Standalone-Nixie-Pool-Controller#install-relayequipmentmanager).

I set the jumpers per instructions and plugging into the first (left) of the two ports for RS485 on that board.
[image]

i2cdetect -y 1 returns:
```     
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- 27 -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```

tagyoureit, 2022-08-22:
Let's start with the rs485 port, then.  Are you able to see the /ttyS0 port in the command line?
[image]

And do you have that port set in njsPC?
[image]

spifftek70, 2022-08-22:
[image]

[image]

tagyoureit, 2022-08-22:
Have you set the Megabas for RS485 passthrough?
[image]
And you can confirm via the command line, as well.
[image]

spifftek70, 2022-08-22:
[image]

[image]

spifftek70, 2022-08-22:
[image]

tagyoureit, 2022-08-22:
Can you try to make a backup of your njsPC `./data/pool*.json` files and let the app recreate them from scratch?  Let's try with a set of fresh config files.

spifftek70, 2022-08-22:
Done, but nothing has changed.  I'm entirely frustrated....

spifftek70, 2022-08-22:
Here is a close up of my hardware.

[image]
[image]
[image]
[image]

rstrouse, 2022-08-22:
First lets verify the configuration for your pi.

Not to butt in but please make sure the following settings are correct for your pi.  If the Serial Port is not on or the Serial Console is on the RS485 port will be identified but will not work.  If the I2c interface is not enabled it will spit out these remote I/O errors.

* I2C = On
* Serial Port = On
* Serial Console = Off
* 1-Wire = On
* Remote GPIO = Off

Next lets double check the MEGA-BAS firmware version.
`cd ~/megabas-rpi`
`megabas 0 board`

You should be running fw v1.09+.  To upgrade the firmware pull the latest repo from the megabas-rpi and do the update procedure.  When you do this you must shut down REM if you do not it will fail the update.

Once we get here we will proceed to the next steps.

spifftek70, 2022-08-22:
I had to change these:

- Serial Console 
- 1-Wire
- Remote GPIO

$ megabas 0 board
Firmware ver 01.07, CPU temperature 45 C, Power source 21.10 V, Raspberry 5.16 V

Got this error when trying to upgrade:
`read cmd NACK ret 0x00 !!!`
`Cpu id read fail!!`

rstrouse, 2022-08-22:
Ok you did not shut down REM before trying to update.  You need to restart the pi and make sure you shut down REM before you attempt to update the BAS firmware.  If you do not the megabas will fail to update as soon as something tries to talk to it.

The Serial Console was affecting the RS485 and the remote GPIO messes with how the GPIO work.

spifftek70, 2022-08-22:
[image]
[image]

rstrouse, 2022-08-22:
Perfect.  Now power down the pi and go back in.  Now we should be updated with the latest BAS firmware.

spifftek70, 2022-08-22:
I did... the 2nd image is running `./update 0` and it said it did it... but the system still doesn't see the board.

spifftek70, 2022-08-22:
got it. `Firmware ver 01.09, CPU temperature 46 C, Power source 21.09 V, Raspberry 5.16 V`

rstrouse, 2022-08-22:
Perfect,  I assume you now have njsPC, REM, and dashPanel running.  Go into the serial port configuration and lets see if the port is opened.

spifftek70, 2022-08-22:
yes, all 3 are running again `pm2 start` - love that.
remind me where to check the serial port config?

spifftek70, 2022-08-22:
remind me where/how to check the serial port config?

spifftek70, 2022-08-22:
[image]

rstrouse, 2022-08-22:
In dashPanel go to the gears menu.  It will be on the Controller => RS485 Port tab.   I also want to review the devices you have on your pi stack.  From your pictures I do not see 2 of the devices you have defined.

spifftek70, 2022-08-22:
[image]

rstrouse, 2022-08-22:
Alright now lets go back into REM.  You do not have a Super Watchdog on your hardware stack and the Multi-Relay is not on address 3.

You will need to delete both of these and add the Sequent 8 relay at address 38 (0x27) from your i2c output.

tagyoureit, 2022-08-22:
If you aren't getting any errors opening the port like before, you may have the wires backwards.  
[image]

rstrouse, 2022-08-22:
@tagyoureit good catch.  That certainly looks like the bits are inverted.

spifftek70, 2022-08-22:
[image]

spifftek70, 2022-08-22:
okay.. let me changes the wires on the RS485

rstrouse, 2022-08-22:
Perfect.  Now unplug the connector with the Green/Yellow wires that is going to the MEGA-BAS and swap them around.  Do not short them together.

spifftek70, 2022-08-22:
Failure Rate is dropping... and we have Successful's

[image]

rstrouse, 2022-08-22:
Yep it is now talking to equipment.  Since we changed the addresses you will likely have to go in and set the REM connection dropdowns for njsPC.  You will also likely see the wattage and RPM from your pump showing in the dashPanel by now.

spifftek70, 2022-08-22:
No yet..  I think I need to set up `Circuits` first?  [image]

rstrouse, 2022-08-22:
You certainly do.  But it does look like we are getting closer.  It can be frustrating when that stupid serial console is turned on.  If you look at your pump you should see "Display not Active"

spifftek70, 2022-08-22:
Correct..  I can now control the Intelliflo pump via the dashboard.  But NOT the single-speed.... yet.

rstrouse, 2022-08-22:
The SS pump must be attached to a 3hp relay connected to your Sequent 8.  This will turn on whenever any of the defined circuits are engaged.

rstrouse, 2022-08-22:
You assign this relay on the Relays Tab for the pump.

spifftek70, 2022-08-22:
okay... what's next?  The temperature sensors are still not providing information.

[image]

[image]

rstrouse, 2022-08-22:
Ok navigate to the Generic Devices and lets get these set up.  The thermistors look like they are giving good values.

rstrouse, 2022-08-22:
Create 1 device for each of your temperature probes per this wiki.
https://github.com/rstrouse/relayEquipmentManager/wiki/Generic-Devices#10k-temperature-probes

spifftek70, 2022-08-22:
Done
[image]

rstrouse, 2022-08-22:
Now lets go into the MEGA-BAS and feed these devices.  To do this go to the I2c Bus 1 tab and click on the Feeds tab.

1. Press the + button on the Device Value Feeds tab
2. Choose Internal Devices in the _Connection_ dropdown
3. Scroll down to the `in0_101.value` for the _Send Value_ and select it.  Make sure you choose the one that ends in `.value`
4. In the _to Device_ dropdown select `Solar Temp`
5. In the _Input_ dropdown Choose `adcValue`

This will take the raw resistance values and connect them to the probe.  You should add a feed for each of the inputs.
`in0_102.value` will be sent to _to Device_ `Pool Temp`
`in0_103.value` will be sent to _to Device_ `Air Temp`
`in0_104.value` will be sent to _to Device_ `Spa Temp`

Don't forget to set Input to `adcValue` for each of them.

Then when you have finished doing that you should start seeing temperature values on your new temp probes you created.  Click the edit icon on each of these and add a feed to njsPC.  Choose `All Values` as the send value.  Use `temps` as the Socket Event and choos the appropriate sensor as the property.

`waterSensor1` will be the pool temp
`waterSensor2` will be the spa temp
`airSensor` will be the air temp
`solarSensor1` will be the solar temp

spifftek70, 2022-08-22:
[image]

rstrouse, 2022-08-22:
Ok those look good.  Go back to the Generic tab that has your probes that you set up and check to see if it has temperatures.  If it does not, verify you have the Input Type set to `Resistance` and the units are `kohm` since that is what the units from the BAS will be in.

rstrouse, 2022-08-22:
NOTE:  You have to click the little edit icon to get the display to come up on the probes.

spifftek70, 2022-08-22:
Success
[image]

rstrouse, 2022-08-22:
Now all that is left to do is create feeds for each of these to njsPC.  This is the second half of the above post.

spifftek70, 2022-08-22:
More success

[image]

rstrouse, 2022-08-22:
Alright now what other equipment do you have?

spifftek70, 2022-08-22:
UltraTemp connect via RS485
Single Speed Pump for Spa
2x Valve Actuators to divert flow of water to UltraTemp from Pool & Spa

rstrouse, 2022-08-22:
Is this an UltraTemp Hybrid or a heat pump?

spifftek70, 2022-08-22:
Electric.   Hybrid?... as in it can cool and heat water

rstrouse, 2022-08-22:
Nope.  Does your UltraTemp also have a gas line connected?

spifftek70, 2022-08-22:
No gas.  Only electric

[45 more replies]

closed 2022-08-23 by spifftek70