rstrouse/relayEquipmentManager #38: Sequent 4 not working

arieltools, 2023-01-13:
Got REM working on a Raspberry Pi 3 Model B Plus Rev 1.3 with a [Four Relays four HV Inputs
8-Layer Stackable HAT](https://sequentmicrosystems.com/collections/all-io-cards/products/four-relays-four-inputs-for-raspberry-pi) I got a similar problem where I don't get the relays activated when clicking on the blue Multi-Relay Board Buttons after being enabled.

[image]

I can control and test these relays on the same PI using the CLI tools offered by the vendor: 4rel4in.

Followed:
- https://github.com/rstrouse/relayEquipmentManager/issues/21
- https://github.com/rstrouse/relayEquipmentManager/wiki/I2c-Multi-Relays

tagyoureit, 2023-01-13:
Just to double check, did you click the "Save Device" button for the board (hidden in your screenshot)?  There is a save button on the relay but it won't save the entire device until you click the other save button.

arieltools, 2023-01-14:
> Just to double check, did you click the "Save Device" button for the board (hidden in your screenshot)? There is a save button on the relay but it won't save the entire device until you click the other save button.

Thanks for the reply. Yes, it's saved. Tried several combinations but I guess I'm missing one part of the puzzle.

arieltools, 2023-01-14:
The "CFG" Name shown on the screenshot above only appears on clicking "Save Device".

Deleted everything and restarted. Same result.

Changed the log level to debug. 

I'm running `npm run start:cached` directly to remove pm2 from the equation. Tried using the direct Sequent I2c device Definition as shown below:
[image]

Getting this on console:

debug: 14 - Sequent 4 relays 4 inputs Executed read command 0x00 byte read:0x00
verbose: 10.10.10.222 PUT /config/i2c/1/14/deviceCommand/setRelayState {"id":1,"state":true}
info: [11:24:31 AM] 10.10.10.222 PUT /config/i2c/1/14/deviceCommand/setRelayState {"id":1,"state":true}
debug: 14 - Sequent 4 relays 4 inputs Executed read command 0x00 byte read:0x00
warn: Retry #1 setting relay states Sequent 4 relays 4 inputs expected 128 but got 0

> Thanks for the reply. Yes, it's saved. Tried several combinations but I guess I'm missing one part of the puzzle.

arieltools, 2023-01-14:
Update, using Controller "Sequent 8 v2.x" it turns on a couple of the relays (not all) but doesn't turn them off. I'm glad the thing half-work at least.

tagyoureit, 2023-01-15:
Are you able to successfully control the card from the [Command Line tools](https://github.com/SequentMicrosystems/4rel4in-rpi) from Sequent?

arieltools, 2023-01-15:
> Are you able to successfully control the card from the [Command Line tools](https://github.com/SequentMicrosystems/4rel4in-rpi) from Sequent?

Yep. It works fine.

The tests as well.

Even the Python library allows me for interactively toggle relays.

arieltools, 2023-01-15:
I don't know much about the hex values but was tinkering a bit and found some code on the Python library for Set and Clear.

arieltools, 2023-01-15:
it would have to do with the bytes and maps, or something like that, because this works (I2c-Relay.ts - around line 830):
```
                        if (1 === byte){
                            await this.sendCommand([0x01, 1]);
                        }else{
                            await this.sendCommand([0x02, 1]);
                        }
```

(I'm just tinkering)

tagyoureit, 2023-01-15:
Can you upload a set of your configuration files and logs for when you are trying to change the relays?  And also output of the command line showing the firmware/version info?

arieltools, 2023-01-15:
How do you get firmware/version info?
 
software: 4rel4in v1.0.1

arieltools, 2023-01-15:
/data/controllerConfig.json

```
{
  "controllerType": "raspi",
  "gpio": {
    "pins": [],
    "exported": []
  },
  "spi0": {
    "isActive": false,
    "busNumber": 0,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "spi1": {
    "isActive": false,
    "busNumber": 1,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "i2c": {
    "isActive": false,
    "buses": [
      {
        "id": 1,
        "isActive": true,
        "devices": [
          {
            "id": 1,
            "values": {
              "relays": [
                {
                  "id": 1,
                  "name": "Relay #1",
                  "enabled": true,
                  "state": false,
                  "initState": "",
                  "invert": false,
                  "sequenceOnDelay": 0,
                  "sequenceOffDelay": 0,
                  "tripTime": 1673805579229
                },
                {
                  "id": 2,
                  "name": "Relay #2",
                  "enabled": true,
                  "state": false,
                  "initState": "",
                  "invert": false,
                  "sequenceOnDelay": 0,
                  "sequenceOffDelay": 0,
                  "tripTime": 1673806378517
                },
                {
                  "id": 3,
                  "name": "Relay #3",
                  "enabled": true,
                  "state": false,
                  "initState": "",
                  "invert": false,
                  "sequenceOnDelay": 0,
                  "sequenceOffDelay": 0,
                  "tripTime": 1673805576000
                },
                {
                  "id": 4,
                  "name": "Relay #4",
                  "enabled": true,
                  "state": false,
                  "initState": "",
                  "invert": false,
                  "sequenceOnDelay": 0,
                  "sequenceOffDelay": 0,
                  "tripTime": 1673805567410
                }
              ]
            },
            "options": {
              "name": "Multi-Relay Board",
              "idType": "sequent4",
              "controllerType": "sequent4",
              "maxRelays": 4
            },
            "info": {
              "registers": [
                {
                  "name": "CFG",
                  "register": 3,
                  "desc": "Configuration register for the relay",
                  "value": 0
                }
              ]
            },
            "address": 14,
            "typeId": 300,
            "sampling": 1,
            "isActive": true,
            "feeds": [],
            "triggers": [],
            "name": "Multi-Relay Board"
          }
        ],
        "busNumber": 1,
        "functions": {
          "i2c": true,
          "tenBitAddr": false,
          "protocolMangling": false,
          "smbusPec": true,
          "smbusBlockProcCall": false,
          "smbusQuick": true,
          "smbusReceiveByte": true,
          "smbusSendByte": true,
          "smbusReadByte": true,
          "smbusWriteByte": true,
          "smbusReadWord": true,
          "smbusWriteWord": true,
          "smbusProcCall": true,
          "smbusReadBlock": false,
          "smbusWriteBlock": true,
          "smbusReadI2cBlock": true,
          "smbusWriteI2cBlock": true
        },
        "addresses": [
          {
            "address": 14,
            "manufacturer": 0,
            "product": 0,
            "name": "Multi-Relay Board"
          }
        ]
      }
    ],
    "detected": [
      {
        "driver": "bcm2835 (i2c@7e804000)",
        "path": "/sys/class/i2c-dev/i2c-1",
        "name": "i2c-1",
        "busNumber": 1
      },
      {
        "driver": "bcm2835 (i2c@7e805000)",
        "path": "/sys/class/i2c-dev/i2c-2",
        "name": "i2c-2",
        "busNumber": 2
      }
    ]
  },
  "connections": [],
  "genericDevices": {
    "isActive": true,
    "buses": [],
    "detected": [],
    "options": {},
    "devices": []
  },
  "oneWire": {
    "isActive": false,
    "buses": [],
    "detected": [
      {
        "path": "/sys/devices/w1_bus_master1",
        "name": "w1_bus_master1",
        "busNumber": 1
      }
    ]
  },
  "lastUpdated": "1/15/2023, 10:22:06 AM",
  "configVersion": 1
}
```

arieltools, 2023-01-15:
Logs when changing the relay:

```
debug: 14 - Multi-Relay Board Executed read command 0x00 byte read:0x00
verbose: 10.10.10.222 PUT /config/i2c/1/14/deviceCommand/setRelayState {"id":1,"state":true}
info: [2:38:19 PM] 10.10.10.222 PUT /config/i2c/1/14/deviceCommand/setRelayState {"id":1,"state":true}
debug: 14 - Multi-Relay Board Executed read command 0x03 byte read:0x00
debug: 14 - Multi-Relay Board Executed read command 0x00 byte read:0x00
```

arieltools, 2023-01-20:
8relind - [Eight Relays 8-Layer Stackable HAT](https://sequentmicrosystems.com/collections/industrial-automation/products/8-relays-stackable-card-for-raspberry-pi) works out-of-the-box :)

brentk7, 2023-02-18:
I have a Sequent 4 relay 4 input module V1.1 which works perfectly with REM.
I have just received a new 4 relay 4 HV input v4 module which does not work with REM. (Sequent service is excellent - they sent me a brand new card free of charge when I discovered the RS485 port was not working, even though I purchased the card over a year ago.)
Looking at the Sequent documentation - there are new libraries available for the v4 card which give access to additional features of these cards. From my testing - the original command line tool works on both versions of the cards to drive the relays.
I see @arieltools mentioned his card was also the "HV input" which I assume is the v4 card - which I'm guessing why it does not work.
Is it possible to get the newer cards supported in REM?

rstrouse, 2023-02-19:
Uggh.  Sequent has this really dumb habit of remapping the registers between cards.  I don't know what they are thinking doing this since it makes for a ton of different versions of software to control their cards.  The hardware is nice but the software is a moving target.  I suppose if you bought the card for a specific project and were starting from scratch it probably wouldn't matter.

I spit out a crapload of F-Bombs when I read your post.  Bear in mind the 4-relay card is a completely different beast than the 4hv.  Do the relays work if you choose the 8-relay ind?  I will have a look at their repo but they also freaking map everything to sequential defines so you have to count the number of bits to get to the correct mapping.  Gets a bit hard when you don't have the hardware.  Even more annoying is that the bits are often not in order.</end rant>

Just to be sure is it this one?
https://github.com/SequentMicrosystems/4rel4in-rpi/blob/main/README.md

brentk7, 2023-02-20:
@rstrouse I totally agree with you - hardware and service wise the Sequent cards are great, but they make way too many changes with no consistency across the software versions and card differences. Would be great if the same functionality (i.e. relays) just worked the same way on all cards.

Both cards I  have are 4 relay 4 input, the original is v1.1 and the newer one is v4.0. As far as I can tell they changed the name to include HV with the newer ones as the Inputs can now handle up to 240VAC (and it also includes standalone Modbus capability). Neither of these are plain 4-relay cards. [Here is a link](https://sequentmicrosystems.com/pages/four-relays-four-inputs-downloads) to the downloads section of Sequents website which includes links for both versions of the "same" card.

The link in your post above is the correct link for the new V4.0 version of the card which is the one which does not work with REM. 

I tried all the different variants of 4 or 8 relay cards in REM under the 'Multi Relay Boards' - but none of them worked correctly (one of them did switch some of the incorrect relays on but not off). (As a side note I also tried all the versions of the 'Multi relay Boards' on one of their [Home Automation Cards](https://sequentmicrosystems.com/collections/home-automation/products/raspberry-pi-home-automation-card) which has 8 relays and these also did not work correctly)

rstrouse, 2023-02-20:
Alright I added a 4rel4in v4 card for you to try.  Bear in mind I do not have any of this hardware so I must simply decode the .c files in their repo.  Fortunately, they are not overly complex but the mapping is always goofy where I have to start with the first bit and work my way through each register.  Give it a shot and let me know.

brentk7, 2023-02-21:
Thanks @rstrouse I tested both the relays and the digital inputs with your new commit and it works perfectly. Thank you very much.

How difficult or time consuming is it to add support for additional cards? I have a [Home Automation Card](https://sequentmicrosystems.com/collections/home-automation/products/raspberry-pi-home-automation-card) which I was hoping to use for pressure sensors. Is it something I (with very rudimentary JavaScript or C skills) could easily achieve?

rstrouse, 2023-02-21:
The biggest problem is that you need to go through the CLI for sequent and figure out which registers map to which i2c commands/registers.  If they were all the same we would have been able to simply add/remove features for the specific card.

You are welcome to clone the code and take a crack at it.  I don't have much love for that card, given its cost and the relays are only NO.  A single common would have been fine split into banks of 4 but they cannot be used to move a valve.  Folks are much better off and have better options with the MEGA-BAS and an 8 relay card.

rstrouse, 2023-02-21:
Also an ADS1015 breakout is less than $5 last I checked.

rstrouse, 2023-05-22:
@brentk7 I added a commit for the Home Automation card.  I took a stab at it.  It supports the relays as well as the inputs and outputs.

closed 2023-05-22 by rstrouse