rstrouse/relayEquipmentManager #39: UART Support for EZO?

camaro4life18, 2023-02-22:
Have you considered adding UART support for the EZO devices? In my custom app I created last year, I used UART for all of the EZO device communication. I used a the GPIO compatible UART pins, plus a couple of USB to UART devices for communicating with the Probes and Pumps. I can share the class file I wrote if you would be interested in adding UART support.

The only issue I ran into was USB devices changing ports after a reboot. I was able to get around this by assigning a device name to the EZO device, then scanning for all serial comm ports, reading the device name from each and comparing to the configured device name.

I do see you support I2C, which EZO also supports. Are there any advantages over either comm. protocol? Should I just reconfigure all of my devices to use I2C?

tagyoureit, 2023-02-22:
I2C allows multiple devices to be installed on the same bus and to be discovered and identified (with a few exceptions).  UART requires a separate connection for each device.  We will not be adding UART support as I2C allows communications for not just Atlas Scientific but Sequent Microsystems and many of the other different types of boards that we already support.

If you want to use REM, you should switch to I2C.

camaro4life18, 2023-02-22:
Ok thanks

closed 2023-02-22 by camaro4life18