rstrouse/relayEquipmentManager #40: sudo command in the code

camaro4life18, 2023-02-22:
In the class OneWireBus.ts
Line: 527
The sudo command is issued from within the code. I discovered this because I come from an enterprise systems background where we create users to run applications and give them permissions to run only what they need. For REM I created a user called "poolRelay", which does not have sudo access, and it does not have a password assigned to it which also prevents ssh access through that user.

So when you make changes to a w1 device and click save in REM, you get this error:

> [22/02/2023, 15:26:39] error: Pool Return Temp error getting conversion time: Error: Command failed: echo 41 127 | sudo tee -a /sys/bus/w1/devices/w1_bus_master1/28-3c01f09638f9/alarms
> 
> We trust you have received the usual lecture from the local System
> Administrator. It usually boils down to these three things:
> 
>     #1) Respect the privacy of others.
>     #2) Think before you type.
>     #3) With great power comes great responsibility.
> 
> sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper
> sudo: a password is required

Since REM also needs access to the GPIO, I added poolRelay to the gpio user group.
So my suggestion would be to add these commands to the user installation instructions and remove the "sudo" from the code.

sudo adduser poolRelay gpio
sudo chown root:gpio /sys/bus/w1/devices/w1_bus_master1/28-3c01f09638f9/alarms
sudo chmod 664 /sys/bus/w1/devices/w1_bus_master1/28-3c01f09638f9/alarms

Of course the user would have to change the permissions for each W1 device, but its a 1 time change at install.

rstrouse, 2023-02-24:
I pulled the sudo reference out for now.  We will be refactoring that code to use the fs.  The original code for that was for troubleshooting alarm data from the device.  Since we didn't have the equipment we needed to see the response from stdout.

rstrouse, 2023-03-12:
I am closing this for now.  If you have further issues let me know.

closed 2023-03-12 by rstrouse