rstrouse/relayEquipmentManager #41: Sequent Smart Fan v4+ Observations

subcritical71, 2023-03-11:
Just a couple of observations on the smart fan integration, which is working fine for me (thanks!) but does have some issues.

- Blinky light option no longer exists
- For the Fan Curve, Exponential and logarithmic seem to be swapped. When my CPU is running at 90F, exponential power is ~88%, while logarithmic is ~22. It seems the curves are reversed, or my thinking is incorrect.
- When you switch to F or K, the labels change but the units are still in C on the Fan Curve page (the General page shows the correct units).
- On the General page, once F or K is selected, any subsequent selection of C does not change the units back to C. Also, and I hope I can explain this correctly, when you select F or K and then Save Device the units dropdown shows Celsius, even though the units and label are either F or K.
- Docs are behind, I'll contribute when I get a chance this weekend. [https://github.com/rstrouse/relayEquipmentManager/wiki/Sequent-Microsystems-Boards#sequent-building-automation-hat-mega-bas](url)

rstrouse, 2023-03-12:
I got my hands on a v4 unit.  Most of the underlying functions were removed from it and it has this really weird thing that it needs a GPIO to enable it.  If you look in the GPIO section you will find that pin 32 is automatically assigned as an input and the value is pulled high when the fan is running, otherwise it is pulled low.  The other issues I have fixed.

* On the version 4+ hardware of the SmartFan the blinky light has been removed and replaced with a GPIO reference.
* This stems from the units change issue below
* It was setting the units on the device then recalculating the units with the old values.  You may need to change the units to K then back to c or f to get the curves in sync.
* Thanks for the help.

subcritical71, 2023-03-12:
Just installed this morning. All appears to be working properly. Thanks for your time!

closed 2023-03-12 by subcritical71