rstrouse/relayEquipmentManager #42: error with bindings on start

thrystan, 2023-04-20:
I'm hoping that this is a correct place to ask this. I just installed REM and am trying to start it and I get the following:
pi@PoolController:~/relayEquipmentManager $ npm start

> relay-equipment-manager@8.0.0 start
> npm run build && node dist/app.js

> relay-equipment-manager@8.0.0 build
> tsc

/home/pi/relayEquipmentManager/node_modules/bindings/bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → /home/pi/relayEquipmentManager/node_modules/epoll/build/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/build/Debug/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/build/Release/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/out/Debug/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/Debug/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/out/Release/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/Release/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/build/default/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/compiled/18.16.0/linux/arm/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/addon-build/release/install-root/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/addon-build/debug/install-root/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/addon-build/default/install-root/epoll.node
 → /home/pi/relayEquipmentManager/node_modules/epoll/lib/binding/node-v108-linux-arm/epoll.node
    at bindings (/home/pi/relayEquipmentManager/node_modules/bindings/bindings.js:126:9)
    at /home/pi/relayEquipmentManager/node_modules/epoll/epoll.js:7:31
    at Object.<anonymous> (/home/pi/relayEquipmentManager/node_modules/epoll/epoll.js:15:3)
    at Module._compile (node:internal/modules/cjs/loader:1254:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1308:10)
    at Module.load (node:internal/modules/cjs/loader:1117:32)
    at Module._load (node:internal/modules/cjs/loader:958:12)
    at Module.require (node:internal/modules/cjs/loader:1141:19)
    at require (node:internal/modules/cjs/helpers:110:18)
    at Object.<anonymous> (/home/pi/relayEquipmentManager/node_modules/onoff/onoff.js:5:15)
    at Module._compile (node:internal/modules/cjs/loader:1254:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1308:10)
    at Module.load (node:internal/modules/cjs/loader:1117:32)
    at Module._load (node:internal/modules/cjs/loader:958:12)
    at Module.require (node:internal/modules/cjs/loader:1141:19)
    at require (node:internal/modules/cjs/helpers:110:18) {
  tries: [
    '/home/pi/relayEquipmentManager/node_modules/epoll/build/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/build/Debug/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/build/Release/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/out/Debug/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/Debug/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/out/Release/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/Release/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/build/default/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/compiled/18.16.0/linux/arm/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/addon-build/release/install-root/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/addon-build/debug/install-root/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/addon-build/default/install-root/epoll.node',
    '/home/pi/relayEquipmentManager/node_modules/epoll/lib/binding/node-v108-linux-arm/epoll.node'
  ]
}

Node.js v18.16.0

Help please?

rstrouse, 2023-04-21:
Did you run npm i first?

thrystan, 2023-04-21:
Yes, and it appeared to complete successfully.

rstrouse, 2023-04-21:
Please run `npm rebuild` in the relayEquipmentManager folder.  If this does not work delete the relayEquipmentManager/node_modules directory and run `npm i` again.  It looks like npm did not pull the correct versions for the gpio library.  I am assuming this is a Raspberry Pi.

thrystan, 2023-04-21:
Yes, this is a Raspberry Pi Zero 2. I removed the whole folder before I saw our reply and re-cloned it and then ran npm i, and it has been sitting here for at least 30 minutes.
pi@PoolController:~/relayEquipmentManager $ npm i
npm WARN deprecated eslint-plugin-standard@5.0.0: standard 16.0.0 and eslint-config-standard 16.0.0 no longer re
quire the eslint-plugin-standard package. You can remove it from your dependencies with 'npm rm eslint-plugin-st
andard'. More info here: https://github.com/standard/standard/issues/1316
[##################] | reify:jquery-ui: timing reifyNode:node_modules/@fortawesome/fontawesome-free Completed in 71971ms

rstrouse, 2023-04-22:
Yeah the Pi Zero is pretty slow.  I don't have one because they suddenly became as expensive as a Pi4 when I decided I would get one to play with.  That warning should not cause any issues but I suspect the Pi Zero 2 needs something from the dependencies that a pi3 or 4 does not require.

I will remove the eslint package at some point now that we are on node 16+.

thrystan, 2023-04-22:
So, I swapped the card into a pi4 and was able to run the npm i command successfully and then swapped it back to the zero 2 and it is now running fine.

Thanks for your help.

rstrouse, 2023-04-22:
Clever

thrystan, 2023-04-22:
One other question, and this could probably be another issue. I don't see any instructions anywhere for how to setup a flow sensor. I've setup the REM Chem for my atlas scientific orp and ph sensors, and I added a flow sensor, I tried both flow switch and rate sensor, I have an inline flow meter connected to a gpio pin, and I set the device to the gpio pin I defined, but I still see a warning message and am not getting any readings. Can you point me to the instructions I missed on how to set that up?

rstrouse, 2023-04-23:
What is the device that you connected to GPIO?  Do you mean a flow switch?

thrystan, 2023-04-23:
I have this sensor connected. https://www.amazon.com/gp/aw/d/B07MY6JHNR. I tried the flow switch, and I also tried the rate sensor options and neither seemed to work.

rstrouse, 2023-04-23:
That is a pulse count sensor designed to measure flow.  I have not written any code to decode those pulses.  I am also not so sure that it can keep up with the flow rates either.  If that is inline with the pressure side of your plumbing, you may end up with pieces of impeller in your plumbing.  On the suction side it is definitely too small.

thrystan, 2023-04-23:
It's been working with an interrupt based python script for a couple of years now. I'll have to take a look at how I can add it to this system. My current "system" is all small python scripts that log data to a database which I then display in home assistant. But this is much nicer software.

rstrouse, 2023-04-24:
Yes you can use that pin and simply set it on or off.  This will trigger the flow switch.  Enable it as an input in REM and it will show up as an option for a flow switch.

closed 2023-10-25 by thrystan

mhmatttu, 2023-12-30:
@rstrouse Good morning.  I am getting this same error message in the origional post.  I have tried all the suggestions above including removing REM altogether with re-install.  My node.js version is 18.9.0.  I have multiple other applications on this rPi model 2b including njsPC and dashPanel all running without issue.   REM was working fine previuosly.  I had to restore my SD card from a backup image due to an update issue on another application.; however, REM was working at the time of the bakcup.

tagyoureit, 2023-12-30:
Can you try stopping all other applications on the Pi and then run 'npm i' or 'npm rebuild'?  Maybe it's a memory/resource issue?

mhmatttu, 2023-12-31:
I gave that a shot but same result.  However, I think you're on to something.  I noticed that with REM stopped, CPU load is around 20% but when I start REM, load increases to about 70%.  Still getting same errors and page will not load.

mhmatttu, 2024-01-01:
it is working now. i chaned node to 18.16.1 with npm version 9.5.1.  Previously node was 18.19.0 with npm of 10.2.3.  i'm not sure why or if this is what caused it to work, but that is all that i changed.  cpu load is also back to normal at 18% with all applications running.