rstrouse/relayEquipmentManager #44: MCP3008 ADC question

thrystan, 2023-04-28:
My MCP3008 chip is not being recognized, how to I specify which GPIO pin is connected to the cs pin on the chip?

rstrouse, 2023-04-28:
Are you using SPI?  If it were me I would abandon that and go with an I2c chip like ADS10x5.  There is so much better support on the I2c side.  SPI wile you can get it to talk it can be a pain to work through all the connections.

thrystan, 2023-04-28:
Yes I'm using the MCP3008 SPI chip, it has been working great with the code I've been using, but I can't get it to work with REM, and after I started using REM it quit responding to my code that had been working. My thought is that REM is doing something to block the GPIO pin that is used for the chip select pin on the MCP3008.

rstrouse, 2023-04-28:
Disable the SPI bus by unchecking it on the General tab then reload the server.

thrystan, 2023-04-29:
Ok, thank you.

rstrouse, 2023-05-23:
Pull REM.  I improved the SPI interface to allow devices to be better defined.

closed 2023-10-25 by thrystan