rstrouse/relayEquipmentManager #45: Design options & deployment help request

FlaMike, 2023-04-30:
Good  afternoon, my automate the world friend!  I'm considering seriously leveraging off my existing pool infrastructure to add ORP, pH, & pressure sensors and the ability to control the salt cell & stenner pump as apropos.

As you may recall, I have an EasyTouch 2 4 installed & am running nodejsPoolController, dashPanel, & Crewski's HA integration.  From previous conversations with you, I understand REM is part of the existing PoolController package already in place.

I'm thinking about getting the Atlas Scientific sensors: industrial pH probe, industrial ORP probe, & A-100 analog  pressure sensor.  If my understanding is correct, these are all i2c devices.  If that's the case, can I connect them to an ESP32 & then forward the data to the poolController possibly by using MQTT (& HA??)?  Do I need the EZO boards in this scenario or will the ESP32 suffice?  

I've read the DIY REM Chem Controller wiki & understand I will need to get a flow cell for the pH & ORP probes.  Do you know of any that don't require drilling & tapping for the Altas sensors?  I don't have those skills & would prefer not to ruin an flow cell or three.  Alternatively, are there adapters that may be used with the Atlas sensors?

Assuming I can get the sensor data into REM, can I leverage off the poolController package & the existing EasyTouch hardware, perhaps along with Home Assistant to control the salt cell & stenner pump?

As you can tell. I'm just beginning to think about how to put this all together.  Any & all suggestions & tips are welcome.

Thanks so much for letting me suck up more of your generosity & time!

rstrouse, 2023-05-01:
You can easily extend EasyTouch with a REM Chem implementation.  This repo provides the hardware interface to include probe calibrations, watchdog relay triggers, and dosing relays.

Also, the probes are not I2c.  The EZO circuits are and yes you will need EZO circuits.  Analog pressure sensors do not need the amplification that are required for chem probes so you can run them from an ADC and serve them up with MQTT.  But why would you do that?

FlaMike, 2023-05-01:
So, I'm having a stupid Monday...

My instance of njsPoolController et al are running on a docker box inside the house connected over an Ethernet cable with  RS485 terminators.  It's not near the EasyTouch.  To add the Ph & ORP probes do I need a Pi connected to the EasyTouch, perhaps by another RS485 comm line or over WiFi to the poolController box??  Still trying to get the "big picture layout" in my little head.

I apologize for my ignorance but it's taking me over a little more every day, like a bad sci fi flick from the 1950s.

rstrouse, 2023-05-01:
This is where you must have a pi with REM out at the pad.  The reason is because i2c or any other probe interface for that matter is very short range.  7 feet max for an un-unboosted i2c interface and just a bit more than that for most other sensors.  You can run REM on a pi all by itself out at the pad if you want.  I did that for a while then just ran the whole shebang out on the pad on a single pi.

FlaMike, 2023-05-01:
After reading the various wikis multiple times, I thought that might be the case...  Thank you for confirming.  Gotta decide how I'm going to put it all together.  Do you find the SD card in the Pi poops out frequently?  I know there are cards that supposedly can withstand a relatively high number of read/write cycles.

Also wondering if I can put everything in the EasyTouch.  I could always get a POE hat for the Pi since I have Ethernet available.  Lots of logistics to noodle.

As always, I appreciate your help & patience!

rstrouse, 2023-05-01:
> Do you find the SD card in the Pi poops out frequently?

REM, njsPC, and dashPanel only write in batch.  I have never had a failure on the SD card.  However, I never buy them from the drugstore or supermarket checkout line.  I typically uses SanDisk Ultra/Extreme or Samsung EVO.  In fact, I have only seen one or two instances over the years and the one that sticks out was a cheapo (they are all cheap) that had been used for a long time prior to installing njsPC on it.

If you want to be clever another user tapped the control transformer on the EasyTouch.  I don't know what voltage it is but if it is 24v then putting a MEGA-BAS on your stack can power the pi and all the peripherals using that 24v ac or dc.

FlaMike, 2023-05-02:
Thanks for the additional info & suggestions!  I have used Samsung EVO for anything of value & not had problems.  I've not run a Pi for extended periods in years--once I moved HA to a "real" PC.

I'll close out this thread & try not to bother you.  Time to roll up my sleeves & dig in!

Have a super day!

closed 2023-05-02 by FlaMike