rstrouse/relayEquipmentManager #46: Sequent Home Automation V4 8 relay

dakowals, 2023-05-03:
Any chance you can update for the 8 relay version of the home automation board?  I can use the 4rel4in to run the 8rel8in, but only the first 4 relays work.

rstrouse, 2023-05-04:
I can't seem to find that board.  Are you possibly confusing this with the 8 relay hat?  If you are then choose Multi-Relay Board and choose the Sequent8 IND board for the controller.  If there is a new board can you point me to the spec sheet for it?

dakowals, 2023-05-04:
It is here:
https://sequentmicrosystems.com/products/raspberry-pi-home-automation-card
It is not the 8 relay card.  It is an automation card which acts as I/O or relay.  If you select Multi-Relay, relays contacts 1 and 4 turn on, but will not turn off.  If I select the 4rel4in, all of the relays work properly.

rstrouse, 2023-05-04:
Oh that is the Home Automation card not a variant of the 4rel4in card.  The bummer about that card is that the contacts for the relays all share a common and they are only normally open.  When I get a chance I will add a profile for it hopefully in the next week or two.

dakowals, 2023-05-04:
Great.  Thank you.

rstrouse, 2023-05-20:
I took my first stab at the Building Automation Card.  Pull REM.  It should support the relays, digital inputs, analog inputs, DAC outputs, and open drain.
[image]

dakowals, 2023-05-22:
Awesome.  I will give it try.  Thank you!!

sonapsent, 2023-08-01:
First off thanks for you work on this project. I hobbled together a DIY system with a MCP relay board, ADS1115 ADC, and Atlas EZO sensors 3 or 4 years back. I saw that the $70 Sequent Home Automation board was now an option in REM and thought it would simplify my setup. I am not sure if @dakowals is having problems with this board in REM, but I am. I have tested the relays and ADC using Sequent's ioplus program and everything seems to work.

REM Relays
After setting up the relays, I can only turn on ch1 no other channels. Also I can not turn it back off.

REM Inputs Analog
I can enable each channel, but REM does not seem to report a value or give me any options to edit these particular inputs into a feed.

Let me know if I should provide more detail or screenshots. I can help test from my end or send/buy you one of these boards if that would help.

Thanks,
Nick

tagyoureit, 2023-08-02:
Can you start by changing the logging in REM to silly and log to file?  And then toggle the relays and upload the file in the `/logs` directory here?

[image]

sonapsent, 2023-08-03:
Seems like a good lace to start. After starting the log I went to channels 1-5 and attempted to turn 1 on, turn 1 off, turn 2 on, etc. CH1 turned on, then do not turn off, hit CH2 but CH3 came on, then nothing would really register. The web display seems to read the status well when I manually turned off CH1 and CH3 with "ioplus 0 relwr 1 off".

[consoleLog(2023-08-02_21-01-10).log](https://github.com/rstrouse/relayEquipmentManager/files/12246175/consoleLog.2023-08-02_21-01-10.log)

Hect0rG, 2024-07-25:
I just purchased one of these Sequent Home Automation V4 8 Relay boards thinking it was the BAS. My current goal was to replace my current setup with an ADC that keeps failing to read temperatures every once in a while to something a little more production ready.  I tried using the analog inputs and it seems they are not working as mentioned above. "REM Inputs Analog
I can enable each channel, but REM does not seem to report a value or give me any options to edit these particular inputs into a feed." @rstrouse can I possibly help you debug this? I could probably return this board and get the BAS which seems like may be more capable but my return window is going to be short here.  I don't mean to hijack the thread but is there a reason that you would recommend the BAS over this? My intent in future is to control pool lights, pumps to automate dosing, as well as sensors to read levels in pool. This is all future and not something I am going to do right away.  At the moment all I want is temp readings for pool and ambient temp.

rstrouse, 2024-07-25:
The inputs have not been defined.  I will set it up this evening.

Hect0rG, 2024-07-25:
Excellent Thank you! I appreciate it.

rstrouse, 2024-07-26:
Update relayEquipmentManager the inputs have been added.

Hect0rG, 2024-07-26:
@rstrouse somehow my i2c stopped working so I am chasing that down. Probably shouldn't have updated my pi at the same time that I updated REM. I was able to download the new code last night but never got to test because my i2c isn't finding my sequent board on i2c.  I will report back when I get back online.

Hect0rG, 2024-08-11:
Hi @rstrouse it turns out my board was faulty but I have gotten it replaced and I am back at it. I have added two analog inputs and I enabled them but I do not see the values updating on REM. 

Just to make sure I am not missing any step, this is how I did it. 

I added the sequent board to I2C. 
[image]

Made sure board is active and saved.
[image]

Enabled the two channels One for Pool Temp and the other for Ambient Temp and saved on both the channel edit screen and the !2C device definition screen.
[image]

At this point I am expecting the values to start coming up on this screen but nothing is populating here. I assume my feed won't do anything till that data is coming through. 

[image]

For my feed, what should I select, I believe the pool controller can handle any but is one better than the other?  I have these three options but I am not 100% on which would be the correct or the best one. 
- inAnalog1 / Analog in # 1
- inAnalog1.Value / Analog in # 1 voltage
- inAnalog1.raw / Analog in # 1 raw ADC
 

On another note, I also added a relay to test and I am not able to manually toggle it after I enabled and saved.  This is how I configured it. Based on the documentation, I should be able to click on the box for the relay to toggle it manually but nothing seems to happen. I do see a little box pop up that says setting relay state when I click it  but nothing happens.

[image]

rstrouse, 2024-08-11:
On the Raspberry Pi check the REM logs.  It is odd that there is no temperature or board voltage reported on your first screenshot.

Hect0rG, 2024-08-12:
This is what shows up in the logs. 

[11/08/2024, 21:57:13] info: Starting up SSDP server
[11/08/2024, 21:57:14] info: Initializing GPIO Pins 0
[11/08/2024, 21:57:14] info: Initializing SPI Bus #0
[11/08/2024, 21:57:14] info: SPI Bus #0 Initialized
[11/08/2024, 21:57:14] info: Initializing SPI Bus #1
[11/08/2024, 21:57:14] info: SPI Bus #1 Initialized
[11/08/2024, 21:57:14] info: Initializing i2c Interface
[11/08/2024, 21:57:14] info: Detecting i2c Buses
[11/08/2024, 21:57:14] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
[11/08/2024, 21:57:14] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-2
[11/08/2024, 21:57:14] info: Initializing 1-Wire Interface
[11/08/2024, 21:57:14] info: Detecting 1-Wire Buses
[11/08/2024, 21:57:14] info: Initializing Generic Devices
[11/08/2024, 21:57:15] info: Initializing i2c Bus #1
[11/08/2024, 21:57:15] error: 1-Wire interface not loaded on os.  Please enable through OS config. Error: Command failed: lsmod | grep w1_gpio
[11/08/2024, 21:57:15] info: Generic Devices Initialized
[11/08/2024, 21:57:15] info: Server is now listening on 0.0.0.0:8080
[11/08/2024, 21:57:15] error: Error connecting to Pool Controller http://10.10.1.15:4200: xhr poll error
[11/08/2024, 21:57:15] info: Scanning i2c Bus #1
[11/08/2024, 21:57:15] info: Found I2C device Sequent Home Automation v4 at address: 40 - (0x28)
[11/08/2024, 21:57:15] error: Sequent Home Automation v4 Error initializing Sequent Home Automation v4 registers Invalid buffer length 64
[11/08/2024, 21:57:15] info: Device Sequent Home Automation v4 initialized for i2c bus #1 address 40
[11/08/2024, 21:57:15] info: i2c Bus #1 Initialized
[11/08/2024, 21:57:16] error: Error connecting to Pool Controller http://10.10.1.15:4200: xhr poll error
[11/08/2024, 21:57:18] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:20] info: New socket client connected LdkWhS2A22ijSo5WAAAB -- 10.10.1.8
[11/08/2024, 21:57:20] info: [21:57:20] 10.10.1.8 GET /devices/state?null {}
[11/08/2024, 21:57:21] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:21] error: Error connecting to Pool Controller http://10.10.1.15:4200: xhr poll error
[11/08/2024, 21:57:24] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:27] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:30] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:31] error: Error connecting to Pool Controller http://10.10.1.15:4200: xhr poll error
[11/08/2024, 21:57:33] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:36] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:39] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:42] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:45] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:48] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:51] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:52] error: Error connecting to Pool Controller http://10.10.1.15:4200: xhr poll error
[11/08/2024, 21:57:54] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:57:57] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:00] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:03] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:06] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:07] info: New socket client connected 1V8D6CU85DGkvEajAAAD -- 10.10.1.15
[11/08/2024, 21:58:09] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:11] info: [21:58:11] 10.10.1.15 PUT /config/checkconnection/ {"type":"njspc","isActive":true,"id":null,"name":"njsPC - automatic","protocol":"http:","ipAddress":"10.10.1.15","port":4200,"userName":"","password":"","sslKeyFile":"","sslCertFile":"","hostnames":["pi.example.com"]}
[11/08/2024, 21:58:12] info: Connected to Pool Controller : http://10.10.1.15:4200
[11/08/2024, 21:58:12] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:15] info: [21:58:15] 10.10.1.15 PUT /config/checkemit {"eventName":"checkemit","property":"result","value":"success","connectionId":1}
[11/08/2024, 21:58:15] info: checkEmit: id 1
[11/08/2024, 21:58:15] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:18] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:21] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:24] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:27] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:30] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:33] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.
[11/08/2024, 21:58:36] error: Sequent Home Automation v4 error setting I/O relayState channel e out of range.

Hect0rG, 2024-08-13:
Does it look like I selected the correct board? I am pretty sure I did but maybe there are two that look the same? 

This is the board I have. https://sequentmicrosystems.com/products/raspberry-pi-home-automation-card?srsltid=AfmBOoqQdaaRy8p64KEiE9no63IEI4f1RPnU3hrQShrequxVGYdbaJVd

marthoc, 2024-08-14:
I am also using this Home Automation v4 card; I can turn a relay "on", but when I try to toggle off I receive this message in the logs:
`1|REM  | [14/08/2024, 09:09:57] warn: Retry #1 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #2 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #3 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #4 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #5 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #6 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #7 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #8 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] warn: Retry #9 setting relay states Sequent Home Automation v4 expected 0 but got 0
1|REM  | [14/08/2024, 09:09:57] error: Error setting relay states Sequent Home Automation v4 register did not echo 0 <> 0
`

marthoc, 2024-08-14:
And, I also see the above issue where the ADC does not appear to be reading a voltage from a temperature sensor.

marthoc, 2024-08-15:
@sonapsent @Hect0rG @rstrouse I have tracked down the problem with the relays and opened a PR to fix. Next I'm going to try to track down why the Analog and other values aren't reporting correctly, unless @rstrouse gets to it first!

marthoc, 2024-08-15:
I've tracked down the analog inputs issue as well, and added a commit to the same PR.

marthoc, 2024-08-16:
@Hect0rG my PR now fixes both the analog sensor and relays so this card should work well in REM now once @rstrouse merges it. 

As for what feed to use, I would use the raw ADC value as that comes straight from the card - the ADCs are 12bit so select that from the dropdown. (The output voltage is calculated with reference to the supply voltage on the card, which should be 3.3v, to calculate temperature but there may be some variance there, so in theory the raw value could be more precise. Either way we are talking about a very minor difference between them.)

Hect0rG, 2024-08-16:
Thanks Mark for your help. I am looking forward to setting it up.

marthoc, 2024-08-16:
@rstrouse Thought I had this solved but one final issue is that once a relay is enabled in REM, I would expect it to show up in njsPC in e.g. the list of devices available for selection for a Heater, but it does not, unless I am missing a step.

rstrouse, 2024-08-16:
It needs to include the relays from the getDeviceDescriptions() method.

marthoc, 2024-08-20:
@rstrouse - I've added this to the getDeviceDescriptions() method, modelling after the other entries in that method:
```
        category = 'Relays'
        for (let i = 0; i < this.relays.length; i++) {
            let relay = this.relays[i];
            desc.push({ type: 'i2c', isActive: this.device.isActive, name: relay.name, binding: `i2c:${this.i2c.busId}:${this.device.id}:relay.${relay.id}`, category: category });
        }
```
But, when njsPC tries to set a relay state it doesn't change. Looking at some of the other subclasses in SequentIO.ts, it appears that the relay state is set in the setDeviceState() method, which is not present in the SequentHomeAuto class... thought I would float this out there for pointers from you before I start digging too deep trying to guess at the REM architecture.

Hect0rG, 2024-08-31:
Hi @marthoc and @rstrouse I don't necessarily need the relays just yet. Would I already be able to pull to get the updates for my temp sensors?

marthoc, 2024-08-31:
@Hect0rG yes @rstrouse merged my work so far. The analog inputs will work now.

Hect0rG, 2024-09-06:
Awesome thanks! I will give it a try this weekend for sure!

Hect0rG, 2024-09-07:
I have pulled the latest code and while it does seem better my pool seems very hot as shown in the picture 🥵 😁.

It doesn't look like I have it set up correctly. I tried following the sequent guide in the wiki but the options seem different.  For example in the wiki you can select between 1k and 10k thermistor but I don't have the option. Maybe it's hardcover? I use 10k for whatever that's worth. Is there  a guide for this specific board yet?

[image]
[image]
[image]

Hect0rG, 2024-09-17:
@marthoc I went ahead and did that but my temperatures are still really high. For example my ambient temperature is telling me its 97 when the actual temp is 78. Is it expected to be this far off? I see there is a place to calibrate in REM but I want to make sure I am not masking an issue by adjusting so much. 

[image]
[image]

marthoc, 2024-09-18:
@Hect0rG not sure if it will make a difference but can you try feeding the raw value to that 10k Temperature probe and then change the input type (the inputs on this card are 12 bit).

Hect0rG, 2024-09-18:
Unfortunately it is still really hot. I imagine you have a similar setup. On yours are the temperatures that far off? I wonder if its the sensor itself but it doesn't make sense that both are reporting really high temps. Other sequent cards also seem to have a way to select 1k vs 10k themistors but this one doesnt seem to have that setting. Maybe internally its different? 

[image]

marthoc, 2025-07-22:
I just wanted to point out that the Home Automation V4 card is now working as a result of #95. The temperature issues above are related more to calibration and the calculation of temperature from resistance in the code than the Home Automation card. This issue could be closed.