rstrouse/relayEquipmentManager #47: Sequent Relay Addition - 3 Relay Industrial

subcritical71, 2023-05-12:
## 💥 Proposal

### What feature you'd like to see

Add support for the Sequent "Three 40A/240V Relays RS485 Daisy-channable HAT for Raspberry Pi" HAT [Link](https://sequentmicrosystems.com/collections/industrial-automation/products/three-40a-240v-relays-rs485-for-raspberry-pi).

## Motivation

I recently purchased the 3 relay HAT with the intention to use it for pool lighting control as well as chlorinator isolation. The relay is installed and partially functioning. Initially it seemed the unit would work using the 'Sequent 4rel4in v4' device. While it did work for simple switching there seems to be something going on when rapid switching is required, for example a lighting circuit, where it starts to trip over itself. The relay starts to cycle erratically when changing light shows. I will sometimes end with 3 different colors out of my 7 lights. And during the relay cycling they sound like they are galloping at irregular intervals.

## Pitch

This stackable 3 relay HAT could be used instead of 3 individual high load relays for various circuits reducing the room required in the cabinet and the complexity of the wiring. The relays can each handle 40A and 240V so just about any circuit could be controlled by this.

rstrouse, 2023-05-16:
I'll try to get to this and the other sequent card this weekend.

rstrouse, 2023-05-20:
I posted my first pass to the 3 relay card.  I don't have this card but try it out I think I read the command line code correct.
[image]

subcritical71, 2023-05-20:
I will install tonight and test it in the morning and report back if any issues pop up. Thank you!

subcritical71, 2023-05-20:
Tested it out this morning. Seems to be working good. I do get a spurious out of sync condition on the pool lights once in awhile. I have 6 lights and its usually the same 1 or 2 which wind up out of sync. Picking the same color clears it up. FYI, here is a screenshot of the REM screen which all relays de-energized. I cycled all the relays in various groupings and they all turned off/on as expected. Once again, thank you for doing this.

[image]

rstrouse, 2023-05-20:
Intellibrite, ColorLogic, or any of the switchy goofy lighting schemes for that matter are just plain awful.  Out of sync lights seem to go with the territory.

subcritical71, 2023-05-20:
Yeah, I figured. This method seems about 30 years out of style (technically speaking). You would think for the price they sell these for it would be RS-485 capable.

closed 2023-05-22 by rstrouse