rstrouse/relayEquipmentManager #49: How do you delete a generic device?

chadhobson, 2023-06-11:
Unless I'm just missing it right in front of my face, there's no trash icon and its throwing an error in the console when you click the device at the top of the screen.

[image]

rstrouse, 2023-06-11:
Hard refresh (clear cache) your browser.

rstrouse, 2023-06-12:
Please make sure you are sending the right value.  You need to be sending the in0-10x.value one not the full object.   This is in the docs but I haven't gotten around to moving the other unit based items to the end of the list.  So double check the feed from the MEGA-BAS the send value should be `0-10v in #(x) value`

chadhobson, 2023-06-12:
I am... here's my output...

[image]

But once you're done on this screen, go back to generic devices and click on the temp probe. Does your seem to lose its settings?

[image]

Notice the Input Type and Bitness. Shouldn't that be resistance and kohms? And either way, shouldn't there be some value for temperature showing?

Side question: How do you dev on this app when you can't plug a MEGA-BAS into your Macbook? Are you replaying captured data or have you emulated it? I don't see a test suite here either.

chadhobson, 2023-06-12:
Hmmmm... something I just noticed. Your screenshot has the word "value" in both red circles. Mine doesn't have it in either. Am I running the same version as you? Mine says its 8.0.0.

rstrouse, 2023-06-13:
Scroll down in the dropdown and you will see a second set of options for the feed that include value.  Also, if you run REM on a computer without an I2c bus it will create Mock devices.

chadhobson, 2023-06-13:
Now its showing temperature in REM. And the device is also editable/deletable. So the error above is likely that it was trying to run replace() on an object rather than a string. Makes sense.

The only part still not working is the display of this temperature in the Pool Control Dashboard. It says "Pool Temp" but there's no value and its greyed out. I assume the REM config is pulling all the data, correct? I don't need to create a generic device feed in REM to push to the dashboard?

Sorry for all the questions but the wiki kind of ends at this point, unless I'm missing something.

Thanks for all your help!

chadhobson, 2023-06-13:
Excellent. Now its working perfectly. Closing and moving on... thanks @rstrouse!

closed 2023-06-13 by chadhobson