rstrouse/relayEquipmentManager #50: [FEATURE REQUEST] Add MEGA-BAS Onewire Device

Crewski, 2023-06-14:
It would be nice to be able to use the one-wire interface from the MEGA-BAS in REM.  My original setup has some DS18B20 sensors mounted in my pipes.  I have moved to a MEGA-BAS with some 10K thermistors (once I install them), but it would be nice to still monitor the DS18B20's I have.  I wrote a simple python script to send the values to njs-PC, but it would be nice if it was already in REM.  Thanks.

If anyone is interested, my code is:
```

#!/usr/bin/env python3

import megabas
import requests
import time

def convert_to_f(t_c):
        t_f = round((t_c * 9/5) + 32,1)
        return t_f

def process_sensors():
        c_heater = megabas.owbGetTemp(0,1)
        f_heater = convert_to_f(c_heater)

        c_pool = megabas.owbGetTemp(0,2)
        f_pool = convert_to_f(c_pool)

        c_air = megabas.owbGetTemp(0,3)
        f_air = convert_to_f(c_air)

        data = {"airSensor": f_air, "waterSensor1": f_pool, "waterSensor2": f_heater}
        response = requests.put('http://127.0.0.1:4200/state/temps', json=data)

while True:
        process_sensors()
        time.sleep(10)
```

rstrouse, 2023-06-14:
Actually, all you should need to do is enable the-one wire bus on the BAS and then it should be recognized in the One-wire section of the software.

Crewski, 2023-06-14:
Do you know how to enable it?  I'm looking through the literature and can't find it.  From what I'm experiencing, the BAS sees the sensors and can read them, but the one-wire interface for the pi has no idea they are there.  They don's show up when I run `ls /sys/bus/w1/devices/`.  But when I run `megabas 0 owbidrd 1` I can get the address.

rstrouse, 2023-06-14:
Did you enable the 1-wire bus on the pi?

Crewski, 2023-06-14:
Yeah, I have it enabled.  I've sent an email to sequent asking them about it, I'll let you know what they say.

I'm assuming since its "one-wire over I2C" it won't show up on the pi bus since it isn't connected to a gpio pin.

Crewski, 2023-06-14:
Perfect timing from them, here is their response:

Hi David,
Unfortunately you can not access the sensors without the library. Just for my curiosity, what will be the benefit for that?

Best regards,

rstrouse, 2023-06-14:
It had to be just a feature that existed on the MCU they popped a connector on the board.  I'm sure it could be extracted from the I2c registers.  However, I am not sure that would be the way to go.

lymanepp, 2024-02-23:
https://github.com/SequentMicrosystems/megabas-rpi/blob/master/python/megabas/__init__.py

rstrouse, 2024-02-23:
Why would one use I2c to access a 1 wire device on a raspberry pi?

tagyoureit, 2024-02-23:
@Crewski Are you looking for the [1-wire REM Wiki](https://github.com/rstrouse/relayEquipmentManager/wiki/1-Wire-Devices)?

Crewski, 2024-02-23:
> @Crewski Are you looking for the [1-wire REM Wiki](https://github.com/rstrouse/relayEquipmentManager/wiki/1-Wire-Devices)?

No, it was just for convenience of my setup.  You can disregard this.  I'll close it.

closed 2024-02-23 by Crewski