rstrouse/relayEquipmentManager #51: ADS1115 Read Error

dougcfl, 2023-07-17:
Hello.  Great job with nodejs pool controller and thank you.  I'm setting up a new system and have a few questions/issues.
I'm getting a read error (Remote I/O) when reading temp voltage from an ADC1115 board.  I have an Amazon knock-off ADS1115, so I suspect I'm dealing with a non TI chip.  The error happens when reading "channel" 0x10 and 0x11.  It looks like these are hard-coded values connected to the lowThreshold and highThreshold registers.  I suspect my clone chip is using different registers, or just doesn't support the low/high threshold overflow registers.  I commented out the two lines that read those registers the temps are working as expected and I can see the values in nodejs-poolController.  Is there something else I should try?  I'm okay not dealing with the low/high thresholds as I think they are really just edge cases that will likely never happen.
Thanks.

rstrouse, 2023-07-18:
Which ADC do you have (a link would be great)?  Is it possible that you have one of the boards with an NTC on it.  If you do you need to disable that as the chip will tie channels 0 and 3 into a differential read then oddly enough read the 0 register single ended.

rstrouse, 2023-07-18:
I removed the reading of these registers as we are not using the comparator block.  I suspect the pointer register is probably write only on your chip and cannot be read.

dougcfl, 2023-07-19:
I'm using this board. 
https://www.amazon.com/gp/product/B01DLHKMO2/ref=ask_ql_qh_dp_hza
 It's working just fine, even with the read error.  I'll do a pull to get your changes.  Thanks.

dougcfl, 2023-07-19:
fixed it!

closed 2023-07-19 by dougcfl