rstrouse/relayEquipmentManager #52: REM seems to have lost its config

triorbonk, 2023-08-20:
Not sure what caused it, but today several components on my Nixe pool controller stopped working.  I have narrowed it down to the REM and it does not seem to be configured any longer.  I have some backups of the directories, but not sure what director the configuration was in.

triorbonk, 2023-08-23:
> The entire setup is in the data directory. All configuration is located in the controllerConfig.json file.

I restored the file from my backup.  It all seems to be working now.  Do you know what would cause this file to be reset back to initial install?

Eric

rstrouse, 2023-08-23:
I have seen this once before.  The cause for this was related to power.

nkm8, 2025-01-30:
I wanted to add another data point that I have experienced this same issue - total data/config loss as a result of power loss - on 2 separate occasions. Now that I am aware of the issue (and have backups in place) - recovering is relatively painless, but is still a pain nonetheless.

davthomaspilot, 2025-04-04:
It has happened to me several times.

tagyoureit, 2025-05-30:
Added https://github.com/rstrouse/relayEquipmentManager/commit/9a4130dd289eb0d913343f5fb64c5de5dd9d7708 to increase the write timer to 60s and use buffered writes.  Hope this helps.

tagyoureit, 2025-06-13:
Great to hear!  Thanks for testing.

closed 2025-06-13 by tagyoureit