rstrouse/relayEquipmentManager #53: Multi-relay board error in logs

seaverd, 2023-10-05:
rstrouse,

I could use your assistance.  I had a working REM, Nodejs Pool Controller and Dashboard running for the past year or so due to your help when my intellicenter board died.  I had an SD card failure on the pi, so I am starting with a clean install.  I am having one issue and that is I cannont get the Sequent Microsystems multi-relay board to function.  After adding the board in REM and enabling the relays...I cannot get them to toggle.  I set my debug to silly and have attached a log from a clean boot and then attempted to toggle each of the 8 relays a couple  of times.  The log shows an error of undefined as part of each relay toggle press.  I have also included a copy of my REM controllerConfig.json file.

Any assistance you can provide is greatly appreciated!
Dan

[consoleLog(2023-10-05_14-18-35).log](https://github.com/rstrouse/relayEquipmentManager/files/12821821/consoleLog.2023-10-05_14-18-35.log)
[controllerConfig.txt](https://github.com/rstrouse/relayEquipmentManager/files/12821897/controllerConfig.txt)

seaverd, 2023-10-06:
@rstrouse 

As a follow up I can confirm that the relays can be toggled from command line using the 8relind-rpi tool.  

I have also done npm rebuild and npm I and restarted and get the same behavior as original post.

Thanks,
Dan

rstrouse, 2023-10-06:
The 8 relay board does not respond to the REM detection so can you verify that the board is located at address 0x2d.

seaverd, 2023-10-06:
@rstrouse

Yes the relay board is located at 0x2d.  See attached screenshot.

Dan

rstrouse, 2023-10-06:
I didn’t get the screenshot.  I mean to check it with i2cdetect to see if it shows up there.

seaverd, 2023-10-06:
Yes, i2cdetect saw all 3 of my devices.   Megabas, multi relay board and smart fan.  

Here is previous screenshot and results of i2cdetect -y 1
[image]

[image]

rstrouse, 2023-10-06:
Actually you have the SmartFan and the Relay 8 addresses reversed.  That is why it is able to read the registers at that address and not report errors.

seaverd, 2023-10-06:
OMG, such a foolish mistake.  Reversed them and all works as expected.
THANK YOU!

closed 2023-10-06 by seaverd