rstrouse/relayEquipmentManager #54: Sequent 16 relay card

brydenver2, 2024-01-25:
Hello @rstrouse,  I have been working on the Sequent 16 Relay Hat in the I2cRelay.ts but using the 8 relay as a template I can’t seem to get the address’s correct.  Excuse me as I am a newbie at GitHub, how can I submit the code for you to take a look? Do I just create a branch?

rstrouse, 2024-02-20:
Clone the repository, then make your local changes, then make a pull request.  Sorry, I missed your issue.

phaelium, 2024-05-06:
Hey guys, just seeing if this board has been added yet?  I can't seem to get it to work with any of the settings available under multi-relay board but not sure if I'm doing something wrong.  I can activate it from the command line with the Sequent utility.  This is the board - https://sequentmicrosystems.com/products/sixteen-relays-8-layer-stackable-hat-for-raspberry-pi

tagyoureit, 2024-05-06:
This hasn't been added yet.

rstrouse, 2024-05-07:
The only problem with that board is that it only has NO contacts.  You cannot power a valve.

celestinjr, 2024-05-22:
I too would be interested in this 16 relay board being implemented.  I have quite a few on/off devices I may end up controlling from this box (several but not all via 3 HP relays). I’ll likely be buying a second relay card, and getting 16 NO relays vs 8 more NO/NC seems a bit more future proof of a purchase.  I can’t see needing more than 8 NO/NC relays. 

I considered purchasing the 16 relay board and contributing the code to add it, but my experience is mostly in python. I did some C/C++/Java way back when, but no nodejs to date.   Perhaps it’s time to learn!

How intensive is adding a new board?  I’ll try to poke around the code to get an idea as well.

rwagoner, 2025-05-10:
I'm also interested in using the 16 relay solid state board. With it being solid state this would provide increased reliability for controlling DIN mounted relays for larger loads.

https://sequentmicrosystems.com/products/solid-state-relays-8-layer-stackable-hat-for-raspberry-pi

tagyoureit, 2025-05-12:
Short answer is no.  As stated above it has NO contacts which typically aren't ideal for pool equipment.