rstrouse/relayEquipmentManager #55: Mega-BAS V4 update?

sjpbailey, 2024-02-07:
Hello,
  I am having an issue with my Sequent stacks or Hats. 
  https://sequentmicrosystems.com/collections/all-io-cards/products/building-automation-8-layer-stackable-hat-v4-for-raspberry-pi
  I have a Mega-BAS V4 and it has 8 universal inputs with dip switches that work fine so far for 10k2 sensors. 
  It has ttl rs-485 with the four analog outputs and "four Triac outputs."
  No Triac's on the configuration?
Is this a need to be added or am i missing an update running 8.0.0?

Thank You for Your time!
 
[image]

[image]

sjpbailey, 2024-02-07:
I also use these Contemporary Controls BASpi controllers that are BacNet and Wifi. You all have a control system as you could add Devices PoolController, IrrigationController, HVACController. 
https://www.ccontrols.com/basautomation/baspi.htm
These controllers sit on a RPi and are also BacNet compatible. Have them on the UDI Plugin store in Python and have a module here. These are nice you configure them in a webpage and they have a free programming tool.
https://pypi.org/project/bascontrolns/

If you could have REM add them as a new device it would be great. Then you wouldn't need a home automation interface especially with a Modbus interface then Rheem and SolarEdge.
Thank You for Your Time,
Steve

rstrouse, 2024-02-20:
Sorry, I never found a use for the 24v triacs.  Keep in mind if you are powering using 24vdc these will not work anyway.

sjpbailey, 2024-02-20:
Thank You for all your work here and Yes you're right! However a 24vac transformer and the four Triac firing relays would be just right for my application. One valve open close, a light and a boiler.
 I get it I wouldn't want to revamp something that already works as intended. 
 Please Forgive my ignorance.

 I also have the Home Automation V4 and if you enable relays  the wrong ones toggle and will not turn off again before they are ever triggered or feed anywhere.
Also Analog inputs show voltage when they are evidently Universal inputs capable of 10k2 sensors?
What am i doing wrong because if you tell it, it is a relay card instead and pick a four relay device the first four relays toggle just fine however no response if you assign them to a pool light.
So has an update from Sequent on hardware affected your software?
Or am I supposed to update and load the card somehow differently?
I develop Python C++ so i am an idiot when it comes to Java or nodejs otherwise i would do a pull request to help instead of seemingly complaining.
This is an extremely cool endeavor on everyone's parts!
 Thank You for Your Valuable Time!
Steve
[image]

rstrouse, 2024-02-20:
If you are referring to a 24vac valve actuator then the triac output is not a good choice for this.  The valve circuitry always has one and only one leg of its circuit energized.  That is why we use a DT relay for such tasks.   

For the analog inputs what you are seeing is what the I2c register happens to be reading.  Different registers are used for different purposes so you need to set the dip switches to the proper setting for the input.  Then when you enable the input you select type of reading that will be taken.  This tells the software which registers need to be read for the input.   In the case of a 10k thermistor this is reading the resistance from the divider circuit on the card so there will be voltage on that divider.

The card in your picture is not a Home Automation card.  It is a relay 8.  You need to use a multi-relay device type.  However, I am not sure what you mean that it doesn't switch off.   There is a very annoying thing that Sequent tends to do.  They cannot seem to help themselves and fiddle with the bit values for the relay outputs between versions if the issue is not simply that you selected the wrong device.   Yes in the sequent world each card has its own masking for the bit triggers.

sjpbailey, 2024-02-20:
Thank You for clearing it up!

sjpbailey, 2024-02-21:
Unfortunately this is a Home Automation V4 card.
No jumpers for the inputs on it.
So I can't buy the old card i guess so is there any plans on updating your REM? 
https://sequentmicrosystems.com/products/raspberry-pi-home-automation-card

[image]
[image]

rstrouse, 2024-02-21:
But in your pictures you have a MEGA-BAS which is perfect for the inputs you need.  No voltage dividers required.  Honestly that home automation card isn't good for much because the relays are all ST and not capable of switching your valve.  Get a relay 8 and pair it with the BAS you have above and all your needs are covered.

sjpbailey, 2024-02-22:
Thank You!I will get the relay card!

sjpbailey, 2024-02-22:
Before i close i am using a usb ttl rs-485 for one VSF intelliflo pump. It works with feedback from the pump gpm, rpm, watts...When i use the Mega-Bas card rs-485  the pump functions however there is no status feedback from the pump?Is this normal?Using rs485 pass through ttyS0.Using USB just curious as to why.Thank You For Your Time!

tagyoureit, 2024-02-22:
You should get the same results no matter which RS-485 port you are using.  The status requests come from njsPC (or the Pentair systems); the pumps don't send them out on their own.

sjpbailey, 2024-02-22:
Thank You only feedback from USB dongle nothing while iUsing the RS-485 input on the Mega-BAS.Ordered the relay 8 card and will give it another go on the 24th.Great frontend for a pool would make a nice frontend for overall Home Automation.

sjpbailey, 2024-02-29:
Very Impressive! Negating the RS-485 port and feedback the Sequent hats function as advertised! Thank YOU for the push! So impressive that i have rebuilt the first beta Plugin formally a Node Server with Universal Devices.I just started to code my Nixie Single Body and plan on the other three listed Shared, Dual and Nobody.Currently it is a local unreleased plugin that adds Circuits automatically. Circuits are override able via UDI's admin console. I need to add schedule status and heating set point to this first basic single body Plugin.I have backed up Images that will make it easier to share.Speaking of which i am one version behind, need to look up how to upgrade njs-poolcontroller.Phone app works and a shot of the controller with temperatures.

rstrouse, 2024-02-29:
Now you are starting to have fun!

sjpbailey, 2024-02-29:
Yes sir great Project here!Thank You!

sjpbailey, 2024-03-03:
Hello, I know it is normal for your VSF intelliflo to show no display however i have completely lost my LCD.Is there a fix for this or do i replace my LCD only to have it die again?Have a new universal devices node server now plug-in close to release however if this take out pump LCD's i cannot release it.Thank You for your time,Steve

sjpbailey, 2024-03-03:
Sorry to have bugged you.I took the display apart. It was dry inside. The LCD is a separate module that has small gold connectors.I just took it apart and reassembled it.Now it is working.I will need to seal it as i have ruined its outside gasket.The assembly has a plastic box that squeezes the LCD onto the PC board. The pump has been fine for a few years i was wondering how many had this issue.Thank You for your time.

tagyoureit, 2024-03-03:
There is no way for any of the master components (njsPC, Pentair, etc) to completely turn off the display.  It should always have something on it.  Glad you got it working again, but keep an eye on it.  It's definitely a hardware and not a software issue.

sjpbailey, 2024-03-03:
Thank you as i see that now as the pump continues to work.Sorry for the interruption and the assumption that made an 🫏 outta me.

celestinjr, 2024-05-22:
> Sorry, I never found a use for the 24v triacs. Keep in mind if you are powering using 24vdc these will not work anyway.

Would the triac be a good option for the 24VAC fireman switch, provided the BAS is powered with 24VAC?  Seems like it would save me a relay.

rstrouse, 2024-05-22:
Honestly, triacs leak current so I wouldn't use it.

celestinjr, 2024-05-22:
> Honestly, triacs leak current so I wouldn't use it.

Understood. I’ll scratch that off my list as an option. Thanks!