rstrouse/relayEquipmentManager #60: ORP, PH pump and sensor faults

davthomaspilot, 2024-05-01:
[image]

I've had a problem for a couple of weeks with PH not dosing.  pH pump would turn on/off as expected, but ph wouldn't change.  So, I've been dosing acid manually.

I replaced the pump tube and one way injector valve, and still it seemed no acid was making it in the pool (and actual acid tank level wasn't changing).

Today, I was going to disconnect the tube at the check valve to see if the acid actually made it that far.  But, I could not find a way  to manually dose.  Then I noticed a "no flow" message.

Flow is actually good, so I was heading to get a DVM to see if the sensor was actually indicating flow and rebooted both the REM pi and the njsPC pi.  When it came back up, I saw probe and pump faults for both ph and ORP.

What could cause this?   

Note the ph and ORP readings are still updating and look believable.  

Also, the tank levels I just set didn't seem to survive the reboot.

davthomaspilot, 2024-05-01:
I used pigpio (pigs) to check state of the flow switch GPIO.  It changes as expected when I turn the pool pump on and off.

Are there any diagnostics I can do in the web interface to REM?  For some reason,  the state of that GPIO is seen by the rpi (pigpio) but not REM.

Probably same root cause is causing the faults.

tagyoureit, 2024-05-02:
It could be some corrupt settings on REM.  Try going into the individual sensors and save them again.