rstrouse/relayEquipmentManager #62: Relay board acting strange

flyboyusa, 2024-05-25:
I have Sequent 8 multi relay board V5 that is acting strange.
The relays all cycle fine, except on a few of the relays, when I cycle them from ON to OFF, all of the other relays cycle OFF then ON momentarily.  This is cycling my heater OFF and ON whenever my chlorine pump kicks off.
Any idea why a few relays affect other relays?

[image]

rstrouse, 2024-05-27:
Can I get a copy of your controllerConfig.json file in your data directory?   This does however seem a bit like a power issue since I am not able to replicate a scenario where the bitmasks get switched.

flyboyusa, 2024-05-27:
[config.json](https://github.com/rstrouse/relayEquipmentManager/files/15458921/config.json)

rstrouse, 2024-06-06:
Actually, I am looking for the controllerConfig.json file in your data directory.

flyboyusa, 2024-06-06:
[controllerConfig.json](https://github.com/user-attachments/files/15687648/controllerConfig.json)

flyboyusa, 2024-09-02:
Were you able to figure anything out on this?  It's still acting this way.  Maybe a bad board?  Whenever any of the relays switch off, all of the other relays cycle off for a second and then back on.