rstrouse/relayEquipmentManager #64: REM not starting / Error: EINVAL: invalid argument, write

dakowals, 2024-05-29:
Up and running just fine.   Updated my pi yesterday.  Today I notice REM is not running.  This is the error I get:

```
npm start

> relay-equipment-manager@8.0.0 start
> npm run build && node dist/app.js

> relay-equipment-manager@8.0.0 build
> tsc

linux
[29/05/2024, 19:18:14] info: Starting up SSDP server
[29/05/2024, 19:18:14] info: Initializing GPIO Pins 1
[29/05/2024, 19:18:14] info: Configuring Pin #32 Gpio #12:low on Header 1 Edge: none. {"activeLow":false,"reconfigureDirection":false}
[29/05/2024, 19:18:15] info: Server is now listening on 0.0.0.0:8080
Error: EINVAL: invalid argument, write
    at Object.writeSync (node:fs:929:3)
    at Object.writeFileSync (node:fs:2328:26)
    at exportGpio (/home/pebblepool/relayEquipmentManager/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/pebblepool/relayEquipmentManager/node_modules/onoff/onoff.js:172:36)
    at GpioController.initPin (/home/pebblepool/relayEquipmentManager/gpio/Gpio-Controller.ts:103:36)
    at GpioController.initPins (/home/pebblepool/relayEquipmentManager/gpio/Gpio-Controller.ts:142:28)
    at GpioController.init (/home/pebblepool/relayEquipmentManager/gpio/Gpio-Controller.ts:22:14)
    at /home/pebblepool/relayEquipmentManager/app.ts:22:38
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
pebblepool@raspberrypi:~/relayEquipmentManager $ 
```
I am not proficient in programming.  Any help would be appreciated.

lymanepp, 2024-05-29:
Did you happen to upgrade Raspberry Pi OS? The latest kernel changed how GPIO works. I reverted back to 6.1.x kernel and it's working again.

dakowals, 2024-05-29:
How do I revert back from 6.6.28 to 6.1.x?

lymanepp, 2024-05-30:
Here's what I found: https://forums.raspberrypi.com/viewtopic.php?t=365677

And the commands that I used to get back to 6.1.x:
`
sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f
sudo reboot
`

dakowals, 2024-05-30:
> Here's what I found: https://forums.raspberrypi.com/viewtopic.php?t=365677
> 
> And the commands that I used to get back to 6.1.x:
> 
> ```
> sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f
> sudo reboot
> ```

You Rock.... this fixed it!

sberkovitz, 2024-05-31:
To get around this on Bookworm, see https://github.com/rstrouse/relayEquipmentManager/pull/65.  Specifically, changes to boards/Constants.ts and pinouts/raspi-4b-bookworm.json.  After you have these, update your controllerConfig.json with

` "controllerType": "raspi-4b-bookworm"`

You might need to remove everything below in "pins": [...]  to get it to start again.

brynw, 2024-05-31:
The 6.6 Kernel re-mapped the GPIO pins - effectively adding 512 to each GPIO value - gpio-0 (ID_SDA) got remapped to gpio-512 (ID_SDA), gpio-1 to gpio-513 etc...

I ended up editing pinouts/raspi.json and remapping the gpioIds in the config file:

[raspi.json](https://github.com/user-attachments/files/15518540/raspi.json)

`
       {
          "id": 3,
          "name": "BCM 2",
          "type": "gpio",
          "gpioId": 514
        },
        {
          "id": 4,
          "name": "5v Power",
          "type": "power5"
        },
        {
          "id": 5,
          "name": "BCM 3",
          "type": "gpio",
          "gpioId": 515
        },
</snippet>
`

bayfacalendar, 2024-06-03:
> To get around this on Bookworm, see #65. Specifically, changes to boards/Constants.ts and pinouts/raspi-4b-bookworm.json. After you have these, update your controllerConfig.json with
>
> ` [...]
I updated my boards/Constants.ts and  added pinouts/raspi-4b-bookworm.json, and then updated my controllerConfig.json to
{
  "controllerType": "raspi-4b-bookworm",
  "gpio": { "pins": [] },
  "spi0": {
    "isActive": false,
    "busNumber": 0,
    "channels": []
  }
}

And I still get the same error.
Error: EINVAL: invalid argument, write
0|REM  |     at Object.writeSync (node:fs:929:3)
0|REM  |     at Object.writeFileSync (node:fs:2328:26)
0|REM  |     at exportGpio (/home/Admin/relayEquipmentManager/node_modules/onoff/onoff.js:18:8)
0|REM  |     at new Gpio (/home/Admin/relayEquipmentManager/node_modules/onoff/onoff.js:172:36)
0|REM  |     at GpioController.initPin (/home/Admin/relayEquipmentManager/gpio/Gpio-Controller.ts:103:36)
0|REM  |     at GpioController.initPins (/home/Admin/relayEquipmentManager/gpio/Gpio-Controller.ts:142:28)
0|REM  |     at GpioController.init (/home/Admin/relayEquipmentManager/gpio/Gpio-Controller.ts:22:14)
0|REM  |     at /home/Admin/relayEquipmentManager/app.ts:22:38
0|REM  |     at processTicksAndRejections (node:internal/process/task_queues:95:5)

What am I missing?

bayfacalendar, 2024-06-03:
> And I still get the same error. Error: EINVAL: invalid argument, write 0|REM | at Object.writeSync (node:fs:929:3) 0|REM | at Object.writeFileSync (node:fs:2328:26) 0|REM | at exportGpio [...]

I'm an idiot, updated the wrong config file. up and working 
thank you!

rstrouse, 2024-06-15:
Unfortunately, the rpi foundation really messed this up.  There is not currently any libgpiod capabilities for node or python projects that include interrupt processing.  I could write my own lib but why... don't they know that a huge amount of software for Raspberry Pi interacts with the GPIO?  They really are making Espressif look like the adults in the room.  They even assigned a different chip id to the pi5.  Not a problem but they left no clear way of determining what hardware you are on that I can find anywhere.

The fix described here is simply including the chip id in the pin id which will work fine until a Pi5 rolls around. </end rant>

sberkovitz, 2024-06-15:
If we leave the onoff library as is, I think we just need a new hardware .json file for each device type -- I assume all of the Pi5 will run Bookworm and not previous builds?

CodeSlinger69, 2024-08-25:
I was bitten by this today. I upgraded to Bookworm last week, and today I received a Smart Fan Hat that I ordered on Amazon. I received v5 of this Fan Hat.

In controllerConfig.json, I changed "controllerType": "raspi-4b-bookworm" as instructed in this thread. I also selected "Sequent Smart Fan v4 and v5" as the device. While this worked, my fan was set at 100% even though REM was trying to set it to a lower percentage.

I removed the device, and reconfigured it as a v6+ fan, and REM was then able to change the fan speed as expected. 

I might suggest that the dropdown be adjusted as follows:

- Sequent Smart Fan v4
- Sequent Smart Fan v5+