rstrouse/relayEquipmentManager #66: Losing configuration

celestinjr, 2024-06-02:
I am using one of the Mega BAS card's 0-10V outputs to send 10V to the AquaRite SWG for RS485, which is working fine.  I can "sudo reboot now" and everything comes back online as it should (Output channel #1 enabled with voltage output = 10V).

However, it seems that if I completely kill power to the Pi (even after "sudo shutdown -h now"), the voltage output reverts to 0V (channel remains enabled).

Configuration attached.
[controllerConfig.json](https://github.com/user-attachments/files/15526162/controllerConfig.json)

Appreciate any help!

rstrouse, 2024-06-05:
I'll bet that the BAS is not retaining any registers when it loses power so when we read it back it is 0v on startup.  How are you setting it to 10v?

celestinjr, 2024-06-06:
I’ve been logging into the REM web interface -> I2C -> MegaBAS -> Outputs -> 0-10v -> Channel 1 gear icon -> set output = 10V. 

FWIW, I also had a separate issue where I kept losing RS485 communication with my AquaRite after ~12+ hrs  despite retaining the 10v output. I believe it may have been a result of having all packets logged to file. Since disabling log to file, I haven’t lost communication with the SWG, but I’m also not able to power cycle the system for a few days. 

I’m not convinced the issues are related, but I figured I’d mention it.

tagyoureit, 2024-06-06:
What you're doing is setting the value manually every time you reboot.  But that won't make it stick between power cycles.  The `-h` is a hardware reset so the Mega-bas will lose it's settings while the regular `sudo shutdown` is just a software reset.  (I'm sure there is a more technical explanation here...)

What you want is something that will feed the value to the output.  An easy way is like in the picture below... set a feed from another device that changes (like cpu temp) to the output.  The sampling will determine how often the value is set (ie the combination of 50+only when changed could actually be a long time; if you set it to a very small number it would be extra processing).  But something like this would ensure the value is set to 10 and then is repeatedly set to 10 on a somewhat regular basis.  This doesn't feel like the most elegant solution, but is the one that comes to mind at the moment.

[image]

tagyoureit, 2024-06-06:
Oops, didn't mean to hit close.  Feel free to keep the conversation going.

celestinjr, 2024-06-06:
Very much appreciated.  I didn't realize the way I was setting the value would not persist.

Is there a way for me to push the current output value to MQTT and set it via a trigger from MQTT?  I could then use NodeRed to monitor and set the value if needed, although I'm not crazy about the SWG being dependent on NodeRed being up to function properly.  I ask more to understand the functionality available.

rstrouse, 2024-06-06:
Actually, I will add persistence to the state so that when it initializes the output it will set it to the stored value.

tagyoureit, 2024-06-06:
👏  Your persistence, @rstrouse, is really the highlight of this app.  😸

rstrouse, 2024-06-06:
I'll post an update soon with this all working.

celestinjr, 2024-06-07:
@rstrouse and @tagyoureit, you guys are top notch.  Appreciate your efforts with these projects.  I'll post a summary of my setup once I'm back in town and work through a few more kinks.

rstrouse, 2024-06-09:
If you update REM it should now persist the outputs.

celestinjr, 2024-06-13:
Tested 14b2b6a90951887187705abb892a946b14619875 this morning, and it seems to have retained the 10V through complete power loss and restart.  Much appreciated!