rstrouse/relayEquipmentManager #67: Sequent Home Automation Card v4

derickcassidy, 2024-06-09:
Hello

I am using a [Sequent Home Automation v4](https://sequentmicrosystems.com/products/raspberry-pi-home-automation-card) card to do some simple pool automation (RS485 for the chlorinator, fireman switch for the heater, and simple relay for the variable speed pump (not rs485).  

I see the definition of the card in the SequentIO file, and a section specifically for the v4 card with a device class of SequentHomeAuto.  However, it appears that many of the registers are mapped differently in this card.  For example temp is listed as 114 (decimal) but it's actually at 62 (decimal) 

Additionally, the analog inputs are all showing as AIN type, and not allowing a setting to make them T10k.  There are no DIP switches on this board to flag the analog input types, so wondering how these might be mapped (rather than defaulting all to T10k  instead of AIN) 

Would a new card be the appropriate way to do this?  Ie:  a Sequent Home Automation v4.0fw1.33. (As I'm guessing the latest firmware is what changed the mappings...) ?   Would a simple mapping of the ioplus.h file where the registers are all enumerated do the job? 

No really wanting to spring for a full MegaBAS + relays hat when I have no valves, no lighting, and no additional features to enabled.  I just want to read the salt level, and turn on the heater  :) 

Cheers!

rstrouse, 2024-06-10:
Actually the current fw is 1.3.6.  It looks like they changed some components on the board with total disregard to any software that might be using it including their own CLI.

The right way to do this might be to read the registers out of the json file.  Just decoding the register mapping out of the ridiculous enum in the CLI is frustrating.

derickcassidy, 2024-06-11:
Agreed decoding the enums was super frustrating.  I just wrote a script to printf() them all and mapped them into the SequentIO file.   More than 50% of the registers changed from one version to the next. 

I asked Sequent for a register map for the specific card and version.  They pointed to the python library as the reference - but the CLI was more up to date.

Makes sense to put them in a JSON file happy to create a pull request if you'd like to point me to where you want me to put them.

On a tangent, I've noticed an error during startup with this board that cannot init the config because of a buffer size.  I'll open another issue specific to that so as to not confuse the two issues.

closed 2024-06-12 by derickcassidy