rstrouse/relayEquipmentManager #68: REM Error

spifftek70, 2024-06-10:
I need some help.  I had to reconfigure my router thanks to my ISP.  this caused some issues, long story short: I tried to restore my NjsPC and REM and it has been acting funny ever since, so I did a wipe and started from scratch.  Now I'm getting the following error from my REM:

```[10/06/2024, 06:50:09] info: Connected to autopool : http://192.168.68.55:4200
[ 39, 72 ]
[10/06/2024, 06:50:09] info: Found I2C device Multi-Relay Board at address: 39 - (0x27)
[10/06/2024, 06:50:09] info: Found I2C device Sequent MEGA-BAS at address: 72 - (0x48)
[10/06/2024, 06:50:09] info: Device Multi-Relay Board initialized for i2c bus #1 address 39
[10/06/2024, 06:50:09] info: Device Sequent MEGA-BAS initialized for i2c bus #1 address 72
[10/06/2024, 06:50:09] info: i2c Bus #1 Initialized
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/connections/Bindings.js:181
    }
    ^

RangeError: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/boards/Controller.js:1067
                await this.server.send({
                                  ^

RangeError: Maximum call stack size exceeded
[10/06/2024, 06:50:16] error: Error sending device feed: Maximum call stack size exceeded
[10/06/2024, 06:50:16] error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded
[10/06/2024, 06:50:18] error: Error sending on internal connection generic:5:3: Maximum call stack size exceeded
[10/06/2024, 06:50:20] error: Error sending on internal connection generic:5:1: Maximum call stack size exceeded
[10/06/2024, 06:50:21] error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/connections/Bindings.js:181
    }
    ^

RangeError: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/boards/Controller.js:1067
                await this.server.send({
                                  ^

RangeError: Maximum call stack size exceeded
[10/06/2024, 06:50:23] error: Error sending device feed: Maximum call stack size exceeded
[10/06/2024, 06:50:23] info: New socket client connected H4zvjWaYUi-Skhu1AAAB -- 192.168.68.55
[10/06/2024, 06:50:24] error: Error sending on internal connection generic:5:1: Maximum call stack size exceeded
[10/06/2024, 06:50:26] error: Error sending on internal connection generic:5:2: Maximum call stack size exceeded
[10/06/2024, 06:50:28] error: Error sending on internal connection generic:5:3: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/connections/Bindings.js:181
    }
    ^

RangeError: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/boards/Controller.js:1067
                await this.server.send({
                                  ^

RangeError: Maximum call stack size exceeded
```

What the heck is this and how do I fix it?

spifftek70, 2024-06-10:
Update from log file:

```[10/06/2024, 07:04:19] info: Starting up SSDP server
[10/06/2024, 07:04:19] info: Initializing GPIO Pins 0
[10/06/2024, 07:04:19] info: Initializing SPI Bus #0
[10/06/2024, 07:04:19] info: SPI Bus #0 Initialized
[10/06/2024, 07:04:19] info: Initializing SPI Bus #1
[10/06/2024, 07:04:19] info: SPI Bus #1 Initialized
[10/06/2024, 07:04:19] info: Initializing i2c Interface
[10/06/2024, 07:04:19] info: Detecting i2c Buses
[10/06/2024, 07:04:19] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
[10/06/2024, 07:04:19] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-20
[10/06/2024, 07:04:19] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-21
[10/06/2024, 07:04:19] info: Initializing 1-Wire Interface
[10/06/2024, 07:04:19] info: Detecting 1-Wire Buses
[10/06/2024, 07:04:19] info: oneWireInterface: w1_gpio                16384  0
wire                   49152  1 w1_gpio

[10/06/2024, 07:04:19] info: Detecting 1-Wire Buses at /sys/bus/w1/devices
[10/06/2024, 07:04:19] info: Initializing Generic Devices
[10/06/2024, 07:04:20] info: Initializing i2c Bus #1
[10/06/2024, 07:04:20] info: Device 10k Temperature Probe initialized for generic device 1 - Water
[10/06/2024, 07:04:20] info: Device 10k Temperature Probe initialized for generic device 2 - Air
[10/06/2024, 07:04:20] info: Device 10k Temperature Probe initialized for generic device 3 - Solar
[10/06/2024, 07:04:20] info: Generic Devices Initialized
[10/06/2024, 07:04:20] info: Server is now listening on 0.0.0.0:8080
[10/06/2024, 07:04:20] error: Error connecting to autopool http://192.168.68.55:4200: xhr poll error
[10/06/2024, 07:04:20] info: Scanning i2c Bus #1
[10/06/2024, 07:04:20] info: Found I2C device Multi-Relay Board at address: 39 - (0x27)
[10/06/2024, 07:04:20] info: Found I2C device Sequent MEGA-BAS at address: 72 - (0x48)
[10/06/2024, 07:04:20] info: Device Multi-Relay Board initialized for i2c bus #1 address 39
[10/06/2024, 07:04:20] info: Device Sequent MEGA-BAS initialized for i2c bus #1 address 72
[10/06/2024, 07:04:20] info: i2c Bus #1 Initialized
[10/06/2024, 07:04:21] error: Error connecting to autopool http://192.168.68.55:4200: xhr poll error
[10/06/2024, 07:04:24] error: Error connecting to autopool http://192.168.68.55:4200: xhr poll error
```

tagyoureit, 2024-06-10:
> error: Error connecting to autopool http://192.168.68.55:4200/: xhr poll error

Start with this error.  Click on the connections edit icon for autopool and either use the `find server` button or enter the ip of your njsPC server manually.

spifftek70, 2024-06-10:
> > error: Error connecting to autopool http://192.168.68.55:4200/: xhr poll error
> 
> Start with this error. Click on the connections edit icon for autopool and either use the `find server` button or enter the ip of your njsPC server manually.

I think I figured that one out.  It had the same connection in there twice, so I removed one, restarted and that error no longer appears.

When I start REM now I get this:

```
$ npm start

> relay-equipment-manager@8.0.0 start
> npm run build && node dist/app.js

> relay-equipment-manager@8.0.0 build
> tsc

linux
[10/06/2024, 08:16:11] info: Starting up SSDP server
[10/06/2024, 08:16:11] info: Initializing GPIO Pins 0
[10/06/2024, 08:16:11] info: Initializing SPI Bus #0
[10/06/2024, 08:16:11] info: SPI Bus #0 Initialized
[10/06/2024, 08:16:11] info: Initializing SPI Bus #1
[10/06/2024, 08:16:11] info: SPI Bus #1 Initialized
[10/06/2024, 08:16:11] info: Initializing i2c Interface
[10/06/2024, 08:16:11] info: Detecting i2c Buses
[10/06/2024, 08:16:11] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
[10/06/2024, 08:16:11] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-20
[10/06/2024, 08:16:11] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-21
[10/06/2024, 08:16:11] info: Initializing 1-Wire Interface
[10/06/2024, 08:16:11] info: Detecting 1-Wire Buses
[10/06/2024, 08:16:11] info: oneWireInterface: w1_gpio                16384  0
wire                   49152  1 w1_gpio

[10/06/2024, 08:16:11] info: Detecting 1-Wire Buses at /sys/bus/w1/devices
[10/06/2024, 08:16:11] info: Initializing Generic Devices
[10/06/2024, 08:16:11] info: Initializing i2c Bus #1
[10/06/2024, 08:16:11] info: Initializing 1-Wire Bus #1
[10/06/2024, 08:16:11] info: Scanning 1-Wire Bus #1
[10/06/2024, 08:16:11] info: Device 10k Temperature Probe initialized for generic device 1 - Water
[10/06/2024, 08:16:11] info: Device 10k Temperature Probe initialized for generic device 2 - Air
[10/06/2024, 08:16:11] info: Device 10k Temperature Probe initialized for generic device 3 - Solar
[10/06/2024, 08:16:11] info: Generic Devices Initialized
[10/06/2024, 08:16:11] info: Server is now listening on 0.0.0.0:8080
[10/06/2024, 08:16:11] info: Scanning i2c Bus #1
[ '00-9a0000000000', '00-5a0000000000', '00-da0000000000' ]
[10/06/2024, 08:16:11] info: Connected to autopool : http://192.168.68.55:4200
[10/06/2024, 08:16:11] info: 1-Wire Bus #1 Initialized
[ 39, 72 ]
[10/06/2024, 08:16:12] info: Found I2C device Multi-Relay Board at address: 39 - (0x27)
[10/06/2024, 08:16:12] info: Found I2C device Sequent MEGA-BAS at address: 72 - (0x48)
[10/06/2024, 08:16:12] info: Device Multi-Relay Board initialized for i2c bus #1 address 39
[10/06/2024, 08:16:12] info: Device Sequent MEGA-BAS initialized for i2c bus #1 address 72
[10/06/2024, 08:16:12] info: i2c Bus #1 Initialized
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/connections/Bindings.js:181
    }
    ^

RangeError: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/relayEquipmentManager/dist/boards/Controller.js:1067
                await this.server.send({
                                  ^

RangeError: Maximum call stack size exceeded
```

tagyoureit, 2024-06-10:
Try going into the triggers or feeds for each of your i2c devices and reconfigure or at least resave them.  Worst case, try deleting each i2c device, restarting the server, and then adding it again.

spifftek70, 2024-06-10:
> Try going into the triggers or feeds for each of your i2c devices and reconfigure or at least resave them. Worst case, try deleting each i2c device, restarting the server, and then adding it again.

The only feeds/triggers I have configured are the temp sensor feeds. I'll try deleting and see what happens.

spifftek70, 2024-06-10:
Okay... Almost there.  Only issue now is that the temp sensors seem to be about 20° too high

spifftek70, 2024-06-10:
Nevermind.  It's all good now.

closed 2024-06-10 by spifftek70