rstrouse/relayEquipmentManager #69: Thermistor Calibration - Steinhart-Hart coefficients?

dkossman, 2024-06-11:
I bought a 10k thermistor (Pentair 520272 non-oem) on Amazon, (https://www.amazon.com/dp/B07Q8RHK7D?psc=1&ref=ppx_yo2ov_dt_b_product_details) and have this hooked up to an ADS1115 on a Pi, and did a quick test using a C program which uses the Steinhart-Hart model.  I measured Vs and Rs with a pretty code multimeter.  However the temperature measurement is off by almost 2 degrees (F).

I found a calculator online at https://www.thinksrs.com/downloads/programs/Therm%20Calc/NTCCalibrator/NTCcalculator.htm which calculates the coefficients for you, based on 3 temperature/resistance readings.  I was planning to use this to adjust the coefficients, but:

Does REM use Steinhart-Hart, if so is there a way to adjust the coefficients, and where are these held?

thanks!

`
#define Rs 9930  // series resistor
#define mx 32767 // adc range max
#define Vr 2.048 // adc voltage measurement range
#define Vs 3.28  // adc supply voltage

/* steinhart-hart coefficents for 10K Ohm resistor, -55 C to 150 C */
#define sA 1.129241E-03
#define sB 2.341077E-04
#define sC 8.775468E-08

float temperature(float r)
{
	/* if  r is the thermistor resistance we can use the Steinhart-Hart 
	 equation to calculate the temperature t in degrees kelvin:
	 t = 1 / (sA + sB * ln(r) + sC * ln(r) * ln(r) * ln(r)) 
	 */
	float k, c;
	k = 1 / (sA + sB * log(r) + sC * log(r) * log(r) * log(r));
	c = k - 273.15;  // convert to Celcius
	return(c * 9 / 5 + 32); // and then return Fahrenheit
}`

tagyoureit, 2024-06-11:
REM includes a calibration offset.

[image]

dkossman, 2024-06-11:
thanks for pointing this out - hopefully the offset will do the trick.  I am waiting for my Pi 4b to arrive and then will breadboard this to check it out.  Currently testing on an old Pi Zero W which i don't think has the horsepower to run NJSPC, REM, Nixie and DashPanel...

If i get really ambitious and you think it would be helpful, I may look at contributing a Steinhart-Hart option to REM for thermistors, but my javascript skills are C- at best.

tagyoureit, 2024-06-11:
You can.  But I'd just keep in mind that precision is never really a goal with pool management.  IE if you change your diverters to use the floor drains instead of the skimmers then your temperature may drop 7+ degrees.  So, philosophically, what is "the temperature" of your pool?  If you can get the Steinhart-Hart to be 50% more accurate and the "true" readings are calibrated to be only 1 degree off (or even perfectly accurate) instead of 2 degrees, will that really be a huge gain over just setting an offset in the UI?  Just food for thought to keep everything in perspective.

rstrouse, 2024-06-15:
If you pull REM it will now allow you to set the calculation type to Steinhart-Hart.  It uses the coefficients for 0 - 50 - 100C.  I believe the table was also generated from these same values,

closed 2024-06-16 by dkossman