rstrouse/relayEquipmentManager #70: Support for Pimoroni Automation HAT

marthoc, 2024-06-11:
I see in the wiki that the ADCs on [this board](https://shop.pimoroni.com/products/automation-hat?variant=30712316554) are supported for e.g. temperature probes. Are the rest of the inputs/outputs on that HAT supported also (e.g. relays, 24v inputs/outputs)? I have a simple pool setup (single body, one VS pump, one heater, one filter, Aquarite LS SWG) and if I can accomplish most of what I want to do with that relatively inexpensive HAT and an RS485 usb, that would be ideal.

Thanks for all of your work on this solution!

marthoc, 2024-06-11:
I think I have answered my own question - it appears that the inputs/outputs and relays are tied to GPIO pins: 
[image]
So, given that REM can control the RPi status of the relay PINs, I would assume that it can control these?

rstrouse, 2024-06-11:
That is correct.  The ADC is actually an ADS 1x15 and there is a Pimoroni setting in there for that.

marthoc, 2024-06-11:
@rstrouse Thanks for confirming!

closed 2024-06-11 by marthoc