rstrouse/relayEquipmentManager #71: Generic Devices loses Save and Delete button

katodude, 2024-06-14:
I am helping someone set up a pool controller. 

We are trying to set up Generic devices for Temp. We can create the generic device. But when we go back in to edit it the save and delete buttons are gone. 
[image]

rstrouse, 2024-06-14:
I’ll have a look.

rstrouse, 2024-06-14:
In the browser console are you seeing any errors?   It appears as if it is throwing an error and not rendering the buttons.

rstrouse, 2024-06-14:
Ok I figured out what you did.  You have a feed coming from the MEGA-BAS that is the input pin and not the value.  You want to set the send value on the feed to the value property and not the entire ADC output.

## BELOW IS WRONG
[image]

## THIS IS CORRECT
While this is documented in the setup I think that we probably need to make this more apparent or parse the data out of the feed anyway.  Notice the selection for the `Send Value`.
[image]

lymanepp, 2024-06-14:
I had the same problem on initial setup. I wasted quite a bit of time before I figured it out.

rstrouse, 2024-06-15:
I made it so that it can digest the full output reference.  At least it won't cause the feed to fail.

rstrouse, 2024-06-15:
@katodude if you are still having issues please upload your `data/controllerConfig.json`.   Also, have updated REM to allow for unexpected values stored on the configuration.  Update REM to install these changes.

```
cd ~/relayEquipmentManager
git pull
npm i
```

dkossman, 2024-06-16:
I'm having somewhat similar problems.  I can't connect my ADS1115 channel to my 10k thermistor device.
When i run my own server code, i have no problems reading the ADS1115 and computing the temperature.  (my server is shut down)

Not sure if I'm doing something wrong, I had this working a few days ago but now can't get it to work.

I've cloned the latest REM code into a new local repo and run "rpm i"
- i set up a generic 10k thermistor with input type Raw Value, Bitness 16, calculation thermistor table
- i set up the ADS1115 as a 16-bit device
- I enabled channel 1 which has the thermistor
- i set up a data feed to send channel 1 as an ADC value to the 10k device, input adcValue
- i can see the correct voltage and bit value of the channel in the ADS panel
- i can see the same bit value in the Thermistor device panel, but the temperature and resistance are blank
- the ADS1115 and Thermistor devices are set to active

```
pi@njspcPi:~/relayEquipmentManager/data $ cat controllerConfig.json
{
  "controllerType": "raspi",
  "gpio": {
    "pins": [],
    "exported": []
  },
  "spi0": {
    "isActive": false,
    "busNumber": 0,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "spi1": {
    "isActive": false,
    "busNumber": 1,
    "channels": [],
    "referenceVoltage": 3.3,
    "spiClock": 1000
  },
  "i2c": {
    "isActive": false,
    "buses": [
      {
        "id": 1,
        "isActive": true,
        "devices": [
          {
            "id": 1,
            "values": {
              "channels": [
                {
                  "id": 1,
                  "value": 24011,
                  "voltage": 1.5007332987456894,
                  "maxValue": 65536,
                  "rawBytes": [
                    93,
                    203
                  ]
                },
                {
                  "id": 2
                },
                {
                  "id": 3
                },
                {
                  "id": 4
                }
              ]
            },
            "options": {
              "adcType": "ads1115",
[... 161 lines cut ...]
          "calibration": 0
        },
        "info": {},
        "triggers": [],
        "feeds": [],
        "typeId": 5,
        "isActive": true,
        "name": "Return Temperature"
      }
    ]
  },
  "oneWire": {
    "isActive": false,
    "buses": [],
    "detected": [
      {
        "path": "/sys/devices/w1_bus_master1",
        "name": "w1_bus_master1",
        "busNumber": 1
      }
    ]
  },
  "lastUpdated": "6/16/2024, 11:47:59 AM",
  "configVersion": 1
```

rstrouse, 2024-06-17:
Please stop pm2 and recompile the Typescript

```
cd ~/relayEquipmentManager
pm2 stop REM
npm start
```

katodude, 2024-06-17:
Done, and that worked. REM is functioning as expected. Thank you. 

Now, just struggling to get the temp registering in the Nixie. I have the feed going into watersensor1. 

This is hardest install I have ever done.

rstrouse, 2024-06-17:
You are sending a feed from the Generic Temp sensor... correct?  Remember the feed does not register until the body is turned on.

katodude, 2024-06-17:
Yes, the feed from the I2C goes into the generic temp sensor. Then the feed from the generic temp sensor goes out. And yes the pool is active. 

[image]
[image]
[image]

rstrouse, 2024-06-17:
Change the send value to All.  We added the other options later so folks could digest the straight temperature values in MQTT.

katodude, 2024-06-17:
Still does not pick it up in the Nixie

[image]

rstrouse, 2024-06-20:
@katodude check the REM log to see if there is an error in there sending on the socket.

katodude, 2024-06-20:
Here is an excerpt from the REM logs. 

20/06/2024, 11:01:53] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:53] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:53] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:53] error: Error connecting to Node JS Pool http://192.168.1.72:4200: xhr poll error
[20/06/2024, 11:01:53] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:53] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:53] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:53] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:54] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:54] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:54] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:54] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:54] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:54] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:54] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:54] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:55] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:55] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:55] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:55] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:55] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:55] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:55] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:55] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:56] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:56] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:56] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:56] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:56] error: Error connecting to Node JS Pool http://192.168.1.72:4200: xhr poll error
[20/06/2024, 11:01:56] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:56] error: 39 3,0: EIO: i/o error, write
[20/06/2024, 11:01:56] error: 39 1,0: EIO: i/o error, write
[20/06/2024, 11:01:56] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:57] error: 39 - Multi-Relay Board Bus #1 Read Command: , Remote I/O error
[20/06/2024, 11:01:57] error: 39 3,0: EIO: i/o error, write

tagyoureit, 2024-06-22:
You may need to physically remove/kill power and restart.  These can be hardware related issues.