rstrouse/relayEquipmentManager #72: Error trying to save config of generic 10k temperature probe

dkossman, 2024-06-15:
Note: this error seems to have started since i pulled REM this morning, after the Steinhard-Hart option was added (so this is arguably my own fault :-)).

I haven't yet fired up NJSPC or Dashpanel, just trying to get REM working.

In REM

added an I2C relay board and an ADS1115
added 3 1-wire DS18B20 probes
added a generic 10k temp probe device.
configured the probe for voltage and 16 bits
expected: it works and probe config is saved
actual:
   the configuration is not saved and reverts to the defaults (Raw value, 10 bits).
  errors in the log:
```
[15/06/2024, 13:41:22] error: Error creating generic device factoryCreate: mod[sName] is not a constructor
[15/06/2024, 13:41:22] error: Factory error creating device for undefined
```

This log starts at resetting the server, i attempted to set up the probe using both Steinhart-Hart and the lookup table, same issue. This makes it impossible to feed the ADS1115 voltage to the 10k probe device

```
[15/06/2024, 13:40:41] info: [1:40:41 PM] 10.0.1.36 GET /config/options/general?null {}
[15/06/2024, 13:40:44] info: [1:40:44 PM] 10.0.1.36 PUT /config/reset {"controllerType":"raspi","app":{"level":"info","logToFile":false},"spi0":{"isActive":false},"spi1":{"isActive":false},"busNumber":"1","detected":{"path":"/sys/devices/w1_bus_master1"}}
[15/06/2024, 13:40:44] info: Stopping GPIO Controller
[15/06/2024, 13:40:44] info: Initializing GPIO Pins 0
[15/06/2024, 13:40:44] info: Initializing SPI Bus #0
[15/06/2024, 13:40:44] info: SPI Bus #0 Initialized
[15/06/2024, 13:40:44] info: Initializing SPI Bus #1
[15/06/2024, 13:40:44] info: SPI Bus #1 Initialized
[15/06/2024, 13:40:44] info: Closing 2 i2c devices.
[15/06/2024, 13:40:44] info: Stopped I2c Multi-Relay Board
[15/06/2024, 13:40:44] info: Stopped I2c ADS1115
[15/06/2024, 13:40:44] info: Closed i2c Bus #1
[15/06/2024, 13:40:44] info: Initializing i2c Interface
[15/06/2024, 13:40:44] info: Detecting i2c Buses
[15/06/2024, 13:40:44] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-1
[15/06/2024, 13:40:44] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-20
[15/06/2024, 13:40:44] info: Detecting i2c Buses /sys/class/i2c-dev/i2c-21
[15/06/2024, 13:40:44] info: Initializing i2c Bus #1
[15/06/2024, 13:40:44] info: Scanning i2c Bus #1
[ 32, 72 ]
[15/06/2024, 13:40:44] info: Found I2C device Multi-Relay Board at address: 32 - (0x20)
[15/06/2024, 13:40:44] info: Found I2C device ADS1115 at address: 72 - (0x48)
[15/06/2024, 13:40:44] info: Device Multi-Relay Board initialized for i2c bus #1 address 32
[15/06/2024, 13:40:44] info: Device ADS1x15 ADC initialized for i2c bus #1 address 72
[15/06/2024, 13:40:44] info: i2c Bus #1 Initialized
[15/06/2024, 13:40:46] info: REM Devices Reset
[15/06/2024, 13:40:53] info: [1:40:53 PM] 10.0.1.36 GET /config/options/genericDevices?null {}
[15/06/2024, 13:40:57] info: [1:40:57 PM] 10.0.1.36 GET /config/options/genericDevices?null {}
[15/06/2024, 13:41:22] info: [1:41:22 PM] 10.0.1.36 PUT /config/genericDevices/device {"options":{"name":"Pump Return Temp","units":"F","inputType":"volt","vccRef":3.3,"inputBitness":10,"inputResistanceUnits":1,"calcType":"interpolate","calibration":0},"values":{"adcValue":null,"inputUnits":"","resistance":null,"temperature":null,"units":""},"id":2}
[15/06/2024, 13:41:22] info: received {"options":{"name":"Pump Return Temp","units":"F","inputType":"volt","vccRef":3.3,"inputBitness":10,"inputResistanceUnits":1,"calcType":"interpolate","calibration":0},"values":{"adcValue":null,"inputUnits":"","resistance":null,"temperature":null,"units":""},"id":2}
[15/06/2024, 13:41:22] error: Error creating generic device factoryCreate: mod[sName] is not a constructor
[15/06/2024, 13:41:22] error: Factory error creating device for undefined
[15/06/2024, 13:41:29] info: [1:41:29 PM] 10.0.1.36 PUT /config/genericDevices/device {"options":{"name":"Pump Return Temp","units":"F","inputType":"volt","vccRef":3.3,"inputBitness":10,"inputResistanceUnits":1,"calcType":"shart","calibration":0},"values":{"adcValue":null,"inputUnits":"","resistance":null,"temperature":null,"units":""},"id":2}
[15/06/2024, 13:41:29] info: received {"options":{"name":"Pump Return Temp","units":"F","inputType":"volt","vccRef":3.3,"inputBitness":10,"inputResistanceUnits":1,"calcType":"shart","calibration":0},"values":{"adcValue":null,"inputUnits":"","resistance":null,"temperature":null,"units":""},"id":2}
[15/06/2024, 13:41:29] error: Error creating generic device factoryCreate: mod[sName] is not a constructor
[15/06/2024, 13:41:29] error: Factory error creating device for undefined
```

rstrouse, 2024-06-16:
This seems like the temperature.json file is out of date. Pull again and verify that it pulled successfully or is already up to date.

dkossman, 2024-06-16:
```
pi@njspcPi:~/relayEquipmentManager $ git pull
Already up to date.
pi@njspcPi:~/relayEquipmentManager $ 
```

closed 2024-06-16 by dkossman